AltBeacon/android-beacon-library #540: didExitRegion called but beacon is right next to phone

yoyo-san, 2017-07-12:
### Expected behavior
didExitRegion only called when beacon is out of range
### Actual behavior
didExitRegion is sometimes called even though beacon is right next to phone. This happens seemingly randomly, sometimes every few seconds, sometimes minutes go by.
### Steps to reproduce this behavior
Using a bit [modified reference app](https://github.com/yoyo-san/android-beacon-library-reference). 
It has hardcoded beacon UUID which we use, feel free to change to whatever UUID you use.
### Mobile device model and OS version
Motorola Moto 4G (peregrine) - Android 6.1
LG G4-H815 - Android 6.0
### Android Beacon Library version
We tried both 2.9 and 2.11 happens in both.

### Additional info
Our use case is to detect when user approaches beacon and invoke certain action (which must be finished before user is right next to beacon). This means we should detect beacon in approximately 30 seconds when it enters the region (exit can be longer).

We tested our app with several different phones (the two above and also Nexus 5 with Android 6) and several different beacons (different manufacturers). This happens with all of them. Reference library was modified to look like our app. 
We tested reference library linked above with two phones mentioned above and two different beacons.
Bluetooth and location permissions are enabled. Beacon is on the same table as phone and doesn't move. We don't use any power saving apps. We also tried to remove `BackgroundPowerSaver` and extend `MaxTrackingAge` to 60000. 

Beacon specs:
Measured power: -59dBm
Broadcasting Interval: 900ms

Logs:
[Motorola](https://gist.github.com/yoyo-san/31b96ede90bbfc873f0a450cf1775e80)
[LG](https://gist.github.com/yoyo-san/fa1f93ecf10581157bfe29de672ebb06)

Not sure if it matters, but system_process shows this error every few seconds:
`system_process E/AlarmManager: RTC Alarm: 2 null org.altbeacon.beaconreference org.altbeacon.beaconreference/org.altbeacon.beacon.startup.StartupBroadcastReceiver`

If there is anything else we missed or we can help in any way, please let us know.

davidgyoung, 2017-07-12:
This problem is typically caused by the phone not detecting any advertising packets from the beacon in 10 seconds.     Based on your description of the beacons advertising every 900 ms, I would expect the symptoms you report would almost never happen with default scan periods (1.1 secs in the foreground with 0 seconds between scans, 10 secs in the background with 5 minutes between scans).  Are you *sure* the beacon is advertising that frequently?  

I can see from the modified reference app that you have customized the scan intervals:

```
beaconManager.setBackgroundScanPeriod(10000);
beaconManager.setBackgroundBetweenScanPeriod(10000);
beaconManager.setBackgroundMode(true);
```

Note, however, that setting the background mode to true will not stick so long as you have the `BackgroundPowerSaver` enabled, so you will revert to foreground scan intervals when the app enters the foreground.  And only about 90% of beacon packets are received.  so what may be happening is you get an exit on transition to foreground mode in a scenario like this, where a single packet is missed.  The following would happen about 10% of the time:

```
t+00.0s Beacon packet detected in background
t+00.0s Library ends its background scan cycle
t+09.5s user brings app to the foreground
t+09.5s Library starts foreground scan
t+10.0s beacon transmits a single packet during this scan, but it is not received due to radio noise
t+10.6s Library stops foreground scan
t+10.6s Library fires didExitRegion event, as beacon has not been seen in 10.6 seconds
```

But this should only happen 10% of the time when transitioning from the background to the foreground.  So this probably does not explain it.

It may help if you capture a log excerpt of such an exit (at least 10 seconds) with `beaconManager.setDebug(true);`

yoyo-san, 2017-07-12:
Thank you for your prompt reply.
I had tried to use with and without `BackgroundPowerSaver`. App was in the background the entire time. 
I am not absolutely sure how often do those beacons advertise, but as mentioned we use two different beacons from two different manufacturers. We ordered the second one for the exact reason to allow beacon to advertise as often as we need (which manufacturer claims is every 900ms). 

Maybe I am misunderstanding something, but doesn't longer scan period increase chance of discovering beacon? For example, if beacon is advertising every second and scan period is 10 seconds, then beacon should be detected (in ideal world) 10 times? Also, I read that from Android L on beacon should be detected in a matter of seconds even if pause between scans is set to 5 minutes, but that didn't work for me (although I am not 100% sure and will try again with default settings).

I also `setMaxTrackingAge` to 60000, which (again, if I understood correctly) should wait at least 60 seconds before "exiting" beacon and it didn't work either.

I will provide debug log later or tomorrow.

davidgyoung, 2017-07-12:
Yes, longer scan periods increase the chance of discovering the beacon *in that period*, but what matters is the absolute time it has been since the beacon was seen at the end of the scan period..     At the end of each scan period there is an evaluation of how long it has been since the beacon was last seen, and if it is > 10 seconds, the exit event fires.   

That number is customizable with `BeconManager.setRegionExitPeriod(20000l);`, which would change the time from 10 secs to 20.  The maxTrackingAge is used to determine how long a beacon is tracked for ranging purposes (not monitoring).  It affects when a beacon will drop off calls to  the list in `didRangeBeaconsInRegion()` callbacks.  I suspect that if you lengthen `BeconManager.setRegionExitPeriod();` it will solve your problem, but we may not know the root cause if you go this route.  

Please do update this issue with what you find.

yoyo-san, 2017-07-13:
Thanks for explanation.

As you said, increasing exit period to 60s worked great (will gradually try to lower it). We will use this workaround in our app, but I can still help to debug if you think this is library issue.

Here are the logs:
(without `BackgroundPowerSaver`)
https://gist.github.com/yoyo-san/f4be547629f0071fe959d2e252718762
(with `BackgroundPowerSaver`)
https://gist.github.com/yoyo-san/eaa28355299d7b7d818777b9e91b41d5

I have also tried to use default settings (10 seconds scan / 5 minutes pause), but it didn't detect beacon leaving (which means it didn't detect entering either).

Please let me know if there is anything else I can do. Otherwise, as far as I am concerned I think our issue is resolved by extending exit period, which is good enough for us.

Thank you for your help!

davidgyoung, 2017-07-13:
From the first log, I see that the library is in background mode and the backgroundBetweenScanPeriod is 10 secs.  So this would cause the spurious exits whenever the beacon is not detected in the first scan cycle after the betweenScanPeriod ends and the exit period is the default.

Bottom line: I think this is expected.  But if extending the exit period works for you, that is a fine solution!

closed 2017-07-13 by davidgyoung