AltBeacon/android-beacon-library #542: How to extract one of the beacons?

JasonLin1126, 2017-07-13:
How to collection beacons and extract one of the beacons?  I use to 
> public void didRangeBeaconsInRegion(Collection<Beacon> beacons, Region region) 
collection all the beacons, but I don't know beacons of the array. If I want to extract one of the beacons data ( example: beacon[1] ), how can I do?

wanjuanjuanv, 2017-07-13:
This is its javadoc documentation：
http://altbeacon.github.io/android-beacon-library/javadoc/index.html

Iterator iterator = beacons.iterator();
                    while (iterator.hasNext()){
                        Beacon beacon = (Beacon) iterator.next();
                        String uuid = beacon.getId1().toUuid().toString();
                        int major = beacon.getId2().toInt();
                        int minor = beacon.getId3().toInt();
                        double distance = beacon.getDistance();
                    }

JasonLin1126, 2017-07-13:
@wanjuanjuanv 
Thank you. I will try it.
But I just extract on of the beacons. For examples: I have 3 beacons, I would extract second beacons, I should call what variable as beacon array ( beacon[1] or others? )

davidgyoung, 2017-08-04:
Hi, @wanjuanjuanv, these kinds of questions about how to use the library should be asked on StackOverflow.com, as this forum is reserved for bug reports or feature requests with the library itself.  If you need further help, feel free to ask there and I will help.

closed 2017-08-04 by davidgyoung