AltBeacon/android-beacon-library #543: From the startRangingBeaconsInRegion method to didRangeBeaconsInRegion always takes more than 1 second

wanjuanjuanv, 2017-07-13:
### From the startRangingBeaconsInRegion method to didRangeBeaconsInRegion can shorten the time

### From the startRangingBeaconsInRegion method to didRangeBeaconsInRegion always takes more than 1 second

### 
**See code excerpt and sample LogCat output below.**

beaconManager.startRangingBeaconsInRegion(mRegion);
  Log.i("wjj" , "startRangingBeaconsInRegion");

@Override
                public void didRangeBeaconsInRegion(Collection<Beacon> beacons, Region region) {
                    Log.i("wjj" , "didRangeBeaconsInRegion");
}

**Logcat output:**
[image]

Then I added the settings for the front polling interval
`beaconManager.setForegroundScanPeriod(200);`
**Logcat output:**

[image]

**As shown in the above two figures, the interval between didRangeBeaconsInRegion does become 200ms for me, but why is the interval between startRangingBeaconsInRegion and didRangeBeaconsInRegion or more than a second. Is there any way to shorten the time between the two? The time interval between opening the page and requesting data is too long.**

### All models

### compile 'org.altbeacon:android-beacon-library:2+'    The latest version

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2017-08-04:
I suspect the problem here has to do with when the timing of when `beaconManager.setForegroundScanPeriod(200);` is applied.  It looks like the default 1100 ms scan cycle is still active for the first callback.  I suggest you move the code that sets `beaconManager.setForegroundScanPeriod(200);` earlier in the program (before you bind() to the BeaconManager or before you set up RegionBootstrap) so it will be set up before you start ranging.

Please report back if this solves the problem.

wanjuanjuanv, 2017-08-09:
I move the code that sets beaconManager.setForegroundScanPeriod(200)  earlier in the program (before I bind() to the BeaconManager).
Unfortunately, it is ineffective and does not work.
I do not know why the first callback requires 1100 ms.

piotr-pawlowski, 2017-12-30:
Please look at this line: https://github.com/AltBeacon/android-beacon-library/blob/c6e34c53132dfbc2366f80ec534b4395112846cc/src/main/java/org/altbeacon/beacon/service/ScanHelper.java#L119

`BeaconManager.DEFAULT_FOREGROUND_SCAN_PERIOD` equals 1100L so 1.1 second. Maybe that is an issue? Maybe you should pass value from `beaconManager.setForegroundScanPeriod` method. 

It is just a hint, not a fix.

mikhailmelnik, 2018-02-27:
I'm having exactly the same issue. App logs both `didEnterRegion` and `didRangeBeaconsInRegion` invocations and it takes about 3 second to get the latter. 

Here are my settings:
```
beaconManager.setForegroundScanPeriod(500);
beaconManager.setForegroundBetweenScanPeriod(1000);
```
and the library version is 2.12.4

davidgyoung, 2018-02-27:
@mikhailmelnik, do you start ranging at the same time as you start monitoring?  Or do you wait for a monitoring callback before starting ranging?  If you do the latter, it will take two cycles to get a ranging callback.

mikhailmelnik, 2018-02-27:
@davidgyoung here is shortened version of code I call in applications' onCreate
```
        ...
        beaconManager = BeaconManager.getInstanceForApplication(context);
        beaconManager.setRegionStatePersistenceEnabled(false);
        beaconManager.addRangeNotifier((collection, region) -> {
            ...
        });
        beaconManager.setForegroundScanPeriod(500);
        beaconManager.setForegroundBetweenScanPeriod(1000);
        new RegionBootstrap(new BootstrapNotifier() {
            ...
            @Override
            public void didEnterRegion(Region region) {
                beaconManager.startRangingBeaconsInRegion(region);
            }
            ...
        }, SOME_REGIONS);
        ...
```
Is it the case leading to two cycles? What could help me in this case, starting ranging explicitly without using region bootstrap? What are the downsides of this?

davidgyoung, 2018-02-27:
Yes, it takes two cycles because it take one cycle to scan beacons and get the didEnterRegion, then a second one to get a ranging callback.

Try starting ranging in didDetermineStateForRegion.  This callback isade before the first scan cycle.

mikhailmelnik, 2018-03-01:
@davidgyoung I'm still having strange results. 

So I have moved code from `didEnterRegion` and `didExitRegion` (they are empty now) to the `didDetermineStateForRegion` so it looks like this:
```
@Override
public void didDetermineStateForRegion(int state, Region region) {
    if (state == BootstrapNotifier.INSIDE) {
        Log.d(MyApp.LOG_TAG, "Region entered");
        beaconManager.startRangingBeaconsInRegion(region);
    } else {
        Log.d(MyApp.LOG_TAG, "Region left");
        beaconManager.stopRangingBeaconsInRegion(region);
    }
}
```
and here is what I have in logs:
```
03-01 08:47:43.466 18520-18775/my.app D/APP: Region entered
03-01 08:47:43.478 18520-18520/my.app I/BeaconService: start ranging received
03-01 08:47:44.150 18520-18795/my.app D/APP: didRangeBeaconsInRegion is called with 0 beacons in collection
03-01 08:47:44.663 18520-18809/my.app D/APP: didRangeBeaconsInRegion is called with 1 beacons in collection
```
and observed delay between the first and the last records varies from 1 to 5 seconds.

davidgyoung, 2018-03-01:
For fastest ranging start time, I would change your code to look like this:

    @Override
    public void didDetermineStateForRegion(int state, Region region) {
        if (state == BootstrapNotifier.INSIDE) {
            Log.d(MyApp.LOG_TAG, "Region entered");
        } else {
            Log.d(MyApp.LOG_TAG, "Region left");
        }
        beaconManager.startRangingBeaconsInRegion(region);
    }

From a performance perspective, there is really not a reason to ever stop ranging if you are monitoring anyway -- both have the same costs.   By setting it up like above, you'll get a ranging callback at the end of the first scan cycle instead of using the first scan cycle to enter the region, then getting your first ranging callback on the second scan cycle.

On a related note, if you are finding that you are getting ranging callbacks with 0 beacons detected even though you know the beacon is present, this typically means that the advertising rate of the beacon is too low, and that no packets are detected in one scan cycle.  If you increase the advertising rate of your beacon, you can increase the probability that you detect it on each scan cycle.  This effectively gives you faster average detection times.

mikhailmelnik, 2018-03-01:
Thank you @davidgyoung! 

* Is it safe to call `startRangingBeaconsInRegion` multiple times without corresponding `stopRangingBeaconsInRegion`?
* What is the trade off  of not having `stopRangingBeaconsInRegion` called, would it affect battery consumption? 
* Can I just call `startRangingBeaconsInRegion` straight after this line 
```
 beaconManager = BeaconManager.getInstanceForApplication(context);
```
skipping `RegionBootstrap` usage completely? Or in this case it won't be able to resume my app after the sudden death/crash?

davidgyoung, 2018-03-01:
Yes, it is safe to call `startRangingBeaconsInRegion` multiple times.  If the region is already ranged, it simply updates the ranged region.

Not having `stopRangingBeaconsInRegion` will have no effect on battery consumption when beacons are not around (which is the only time you are calling it.)  You will still get calls to `didRangeBeaconsInRegion` with an empty list.  So long as you do not do heavy processing in this mehthod when it is called with an empty list, you are OK.

You *cannot* call `startRangingBeaconsInRegion` just after making the beaconManager, because the service is not yet bound.  You may either use `RegionBootstrap` or call `bind` manually.  `RegionBootstrap` is largely a convenience class.

mikhailmelnik, 2018-03-01:
@davidgyoung although I'm doing exactly what you've recommended I still notice a delay.

Could you please check the following log and think of what might be a problem?
```
03-01 18:46:11.789 18405-18405/my.app D/BeaconParser: Parsing beacon layout: <LAYOUT_1>
03-01 18:46:11.791 18405-18405/my.app D/BeaconParser: Parsing beacon layout: <LAYOUT_1_EXTRA>
03-01 18:46:11.816 18405-18405/my.app I/BeaconManager: BeaconManager started up on pid 18405 named 'my.app' for application package 'my.app'.  isMainProcess=true
03-01 18:46:11.817 18405-18405/my.app D/BeaconParser: Parsing beacon layout: <?DEFAULT_LAYOUT?>
03-01 18:46:12.962 18405-18405/my.app I/BeaconService: beaconService version 2.12.4 is starting up on the main process
03-01 18:46:12.992 18405-18405/my.app I/BeaconService: starting with intent Intent { cmp=my.app/org.altbeacon.beacon.service.BeaconService }
03-01 18:46:12.995 18405-18405/my.app I/BeaconService: binding
03-01 18:46:13.030 18405-18405/my.app D/APP: Region left: RegionA
03-01 18:46:13.031 18405-18405/my.app D/APP: Region left: RegionB
03-01 18:46:13.698 18405-18405/my.app I/BeaconService: start monitoring received
03-01 18:46:13.710 18405-18405/my.app I/BeaconService: start ranging received
03-01 18:46:13.723 18405-18405/my.app I/BeaconService: set scan intervals received
03-01 18:46:13.733 18405-18405/my.app I/BeaconService: start monitoring received
03-01 18:46:13.744 18405-18405/my.app I/BeaconService: start ranging received
03-01 18:46:13.794 18405-18405/my.app I/BeaconService: set scan intervals received
03-01 18:46:13.956 18405-18570/my.app D/APP: Region entered: RegionB
03-01 18:46:14.044 18405-18405/my.app I/BeaconService: start ranging received
03-01 18:46:14.045 18405-18405/my.app I/BeaconService: Already ranging that region -- will replace existing region.
03-01 18:46:14.755 18405-18578/my.app D/APP: didRangeBeaconsInRegion is called with 0 beacons in collection
03-01 18:46:14.759 18405-18578/my.app D/APP: didRangeBeaconsInRegion is called with 0 beacons in collection
03-01 18:46:16.363 18405-18597/my.app D/APP: didRangeBeaconsInRegion is called with 0 beacons in collection
03-01 18:46:16.370 18405-18597/my.app D/APP: didRangeBeaconsInRegion is called with 1 beacons in collection
```

As you can see it takes about 3 sec between `Region entered: RegionB` and `didRangeBeaconsInRegion is called with 1 beacons in collection` messages.

If it makes any difference I have two regions registered and two layouts: iBeacon as primary frame and sBeacon as an extra frame (although logs show one more layout `m:2-3=beac,...`).

davidgyoung, 2018-03-01:
If you are getting a call to `didRangeBeaconsInRegion` with 0 beacons detected, this means:

1. Ranging is started, and scanning is on.
2. No Bluetooth LE advertisements were detected by android that match the beacon definition for the first three cycles.

So the delay is not in ranging.  It is in detection, which is a different problem.

I suspect that your detection is slow because of a combination of your beacon's advertising rate and your scan settings:

        beaconManager.setForegroundScanPeriod(500);
        beaconManager.setForegroundBetweenScanPeriod(1000);

This means that you will scan for 500ms and then stop scanning for 1000ms.  Unless your beacon is advertising at 3 Hz or more, it is unlikely that you will get a  detection in just 500ms.    This is because not 100% of advertising packets get received by a bluetooth scanner due to radio noise and other factors.  A single packet generally has a 80-90% chance of being received at close range *if* the scanner is on. 

I would recommend you change your scan cycle back to the default on the library:

        beaconManager.setForegroundScanPeriod(1100);
        beaconManager.setForegroundBetweenScanPeriod(0);

If you can't or don't want to do this, then you really need to increase the advertising rate of your beacon to be at least 10Hz to make detections reliably quick.  If it is advertising at only 1 Hz, you have about a 40% chance of detecting it in a single callback to didRangeBeaconsInRegion with the scan period settings shown.

mikhailmelnik, 2018-03-01:
Thanks a lot @davidgyoung! 

I had a suspicion that delay is caused by specific settings as (1000+500) * 2 = 3000 but somehow was pretty sure that if region entry is detected then beacon details should be available immediately as entry event was caused by beacon advertisement. 

If ranging and detecting are two different processes which rely on different data (so single packet can't trigger region entry and then be detected straight away) then I probably have no other options than return back to default options or just give up and let delay happen.

Just to reconfirm - if I switch back to the default settings would it mean app will consume 30% more energy (1700ms vs 2200ms of scanning within 2200ms timeframe)?

davidgyoung, 2018-03-01:
Understand that ranging and monitoring are handled by the same process, but results are reported at the end of a scan cycle.  So if you only start ranging after monitoring detects a beacon, you need one scan cycle to get the monitoring callback and then another scan cycle to get the first ranging callback.  This is what caused the delays you saw.

If you change the scan settings to the defaults, then yes, it will use more battery.  Actually about 67% more battery because the duty cycle you have shown is 500ms on, 1000ms off (scanning on only 33% of the time) whereas the default foreground scan rate is 100% on.

That's why the recommended way to use the library is with the `BackgroundPowerSaver` so when the app is in the background the scan duty cycle is greatly reduced to save battery.

mikhailmelnik, 2018-03-01:
@davidgyoung got it, thank you so much for the great explanation and your time!

closed 2018-03-01 by davidgyoung