AltBeacon/android-beacon-library #545: What is a type of beacon?

JasonChungLin, 2017-07-17:
What is a type of beacon?
for(Beacon **beacon** : beacons){
    ...
    ...
    ...
}
What's default type for the beacon? I would extract beacon data, how can I do?

adri1992, 2017-07-17:
What kind of data do you want to extract from it? Have you added a BeaconLayout for your BeaconParser?

Probably you have this layout: "m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25", that means that all beacons are about AltBeacon specification.

JasonChungLin, 2017-07-17:
@adri1992 
Assume I have three data of beacon. it may be like
1. UUID, major, minor, distance, RSSI
2. UUID, major, minor, distance, RSSI
3. UUID, major, minor, distance, RSSI
How to extract the data that may be required? because I don't know what type of beacon, If I print it.
It will show currently detect Beacon.(Like it is UUID, major, minor ), how can record every piece of data through the array?

davidgyoung, 2017-07-18:
@JasonChungLin, questions on how to use this library are best suited for StackOverflow.com, as this forum is reserved for bug report and feature requests with the library itself.  Please post your question there and I am happy to help on that forum.

closed 2017-07-18 by davidgyoung