AltBeacon/android-beacon-library #546: RSSI 127 filter

iProxDev, 2017-07-17:
In 2.3 of the library a filter was put in place to ignore RSSI values of 127.

I would like to request the addition of a method so that this filter can be set on/off.

In my testing I see RSSI values of 127 when the beacon is next to the phone but the didRange never gets called due to the filter. I would like to be able to optionally ignore such values in my own code.

Thanks

davidgyoung, 2017-07-17:
I see.  The problem is that *all* of your measurements have a RSSI value of 127, so filtering them out means you get no detections.    Perhaps a solution would be to move the filter to take place after all the collections are taken.  If you get any measurements that are not 127, then it removes all the 127 and averages them.  If you get only measurements of 127, it simply returns 127.

iProxDev, 2017-07-18:
Hi David,

Thanks for the reply,  I am new to this. ☺

Yes your solution sounds like it would work well.

Are you also able to point me to the documentation that explains the difference between the deprecated setRangeNotifier and addRangeNotifier ?
Do I still need a background service to handle background scanning?

Regards

Rohan