AltBeacon/android-beacon-library #551: java.lang.NoSuchFieldError: No static field PREVIEW_SDK_INT on ScanJob start

shliama, 2017-07-26:
### Expected behavior
No crash

### Actual behavior
Crash happens when ScanJob is started

### Steps to reproduce this behavior
Just start the scan. PREVIEW_SDK_INT was added in API 23, so there is no such field prior that.

### Mobile device model and OS version
Samsung S5, OS 5.0

### Android Beacon Library version
2.12-beta3

Stacktrace:
```
Fatal Exception: java.lang.NoSuchFieldError: No static field PREVIEW_SDK_INT of type I in class Landroid/os/Build$VERSION; or its superclasses (declaration of 'android.os.Build$VERSION' appears in /system/framework/framework.jar:classes2.dex)
       at org.altbeacon.beacon.service.ScanJob.startPassiveScanIfNeeded(ScanJob.java:118)
       at org.altbeacon.beacon.service.ScanJob.onStopJob(ScanJob.java:134)
       at android.app.job.JobService$JobHandler.handleMessage(JobService.java:135)
       at android.os.Handler.dispatchMessage(Handler.java:102)
       at android.os.Looper.loop(Looper.java:145)
       at android.app.ActivityThread.main(ActivityThread.java:5832)
       at java.lang.reflect.Method.invoke(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:372)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1399)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1194)
```

davidgyoung, 2017-07-26:
Fixed in 0dac329b4cf507036b37b9339a7bc49a4d009172

closed 2017-07-26 by davidgyoung

shliama, 2017-07-26:
@davidgyoung got it, thanks! 
The library truly is a lifesaver :octocat: