AltBeacon/android-beacon-library #553: ConcurrentModificationException Android

DejanRistic, 2017-08-01:
We just sent an update to our app live and have been seeing this crash pretty consistently through fabric/crashlytics. StackTrace and device info can be seen here: http://crashes.to/s/44f7887ced0

### Expected behavior
No exception. 
### Actual behavior
ConcurrentModificationException gets hit. 
### Steps to reproduce this behavior
http://crashes.to/s/44f7887ced0
### Mobile device model and OS version
http://crashes.to/s/44f7887ced0
### Android Beacon Library version
2.11

davidgyoung, 2017-08-01:
Thanks for this report.   

The call to :

    objectOutputStream.writeObject(getRegionsStateMap());

in MonitoringStatus.java must by synchronized because it iterates through the contents on the map.

TonyTangAndroid, 2017-09-01:
I have experienced another ConcurrentModificationException. But it is not related to this issue. https://github.com/AltBeacon/android-beacon-library/issues/577