AltBeacon/android-beacon-library #555: How to implement a Custom Parser

ElieMakhoul, 2017-08-01:
### Expected behavior
Use setBeaconLayout to configure parsing of custom advertisement.

### Actual behavior
Our BLE device doesn't use UUID and implements a custom data structure in its adv. packet.
When applying our data structure to the BeaconLayout, we're not able to see any of our devices.

Here's a sample of the error when debug is set to true;

This is not a matching Beacon advertisement. (Was expecting 02 01.  The bytes I see are: 0201061aff750200d6ff05013506ff0ee8401fdf6e85601a60e8a0bea0510000000000000000000000000000000000000000000000000000000000000000

Clarification: in the above packet, there's no UUID and here's the layout set;
"m:1-2=0201,i:3-3,i:4-4,i:5-5,i:6-7,i:8-8,p:9-9,d:10-13,d:14-29"

### Steps to reproduce this behavior
Should we be implementing our own custom parser or is this an issue in setting the right layout?

### Mobile device model and OS version

Samsung Note 5 running Android V7.1

### Android Beacon Library version
V2.11

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2017-08-02:
From the look of the custom layout shown, I don't think it is going to work for a number of reasons.  To explain better, let's take an example of the AltBeacon layout and see how it works.

`public static final String ALTBEACON_LAYOUT = "m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25";`

The debug log lines below show what you see when it is matched by the library:

```
04-21 15:41:26.095 16692-17437/com.cu.fimobile D/BeaconParser: Processing pdu type FF: 0201061bff1801beac2f234454cf6d4a0fadf2f4911ba9ffa600010001be0000000000000000000000000000000000000000000000000000000000000000 with startIndex: 5, endIndex: 30
04-21 15:41:26.095 16692-17437/com.cu.fimobile D/BeaconParser: This is a recognized beacon advertisement -- be ac seen
04-21 15:41:26.105 16692-17437/com.cu.fimobile D/BeaconService: beacon detected : id1: 2f234454-cf6d-4a0f-adf2-f4911ba9ffa6 id2: 1 id3: 1
```

The first line in the log shows the actual bytes of the advertisement read by Android.  These include multiple Bluetooth LE PDUs, only one of which is the PDU that contains the manufacturer data.   The library parses each PDU, then applies the layout only to the manufacturer data PDU (PDU type 0xff).  Its data fields are: 1801beac2f234454cf6d4a0fadf2f4911ba9ffa600010001be00

The beacon layout byte offsets are relative to this PDU.  So the AltBeacon layout's m:2-3=beac matches bytes 2 as 0xbe and 3 as 0xac.  And that's exactly what you see above.

The beacon layout shown in this issue won't work against the bytes shown for a number of reasons. 
 To understand why, we need to break it down into PDUs:

PDU #1: 020106
PDU #2: 1aff750200d6ff05013506ff0ee8401fdf6e85601a60e8a0bea05100

The first PDU is a two bytes long and of type 0x01 = flags.   The flags value is 0x06.  This PDU type is ignored by the library.

The second PDU is 26 bytes long (0x1a)  and is of type 0xff = manufacturer data.  This PDU type *is* matched by the library, so the layout is applied to this PDU.

The following layout won't work for a few reasons:

  "m:1-2=0201,i:3-3,i:4-4,i:5-5,i:6-7,i:8-8,p:9-9,d:10-13,d:14-29"

1. The matching term m:1-2=0201 doesn't match anything in manufacturer data PDU.  It only matches bytes in the flags PDU (which is ignored by the library).  You might fix this by changing it to m:1-2=7502, *if* those bytes are static in your transmission.  You will need some static bytes to recognize the transmission as your own. 

2. The maximum length of a data field is 8 bytes, as they are returned as Java long values which have this maximum length.  The term d14-29 is 16 bytes long which is twice as long as allowed.  It will still match with this value, but it won't parse it properly.  You might try splitting it into two data fields.

ekryvoshey, 2017-08-02:
Hello, David. From the look into the library's code, I can tell that this custom layout won`t work with current alt-beacon protocol layout and PDU. But what if we can add new PDU and Advertisment for a custom layout? PDU that will ignore first 2 bytes as length and type of the message, that wouldn't demant ServiceUUID? Is it even worth trying to implement something like that?

davidgyoung, 2017-08-02:
It would certainly be possible to augment the library to allow custom beacon layouts that ignore PDU parsing, and and apply the layout against the raw scan data returned by Android regardless of PDU boundary. 

In fact, the library used to work this way in version 1.x, but it was change to parse PDUs because it was much more reliable way to parse beacon advertisements for standard formats where the flags PDU might be of variable length, or there may be multiple flags PDUs.

davidgyoung, 2017-08-04:
Closing this issue, as I worked with @ElieMakhoul offline to come up with a solution.

closed 2017-08-04 by davidgyoung