AltBeacon/android-beacon-library #556: Long time in front desk detection Beacon BUG

lujianzhao, 2017-08-02:
### Expected behavior
Long time in front desk detection Beacon

### Actual behavior
Scan the result is 0, restart the Bluetooth after the normal

### Steps to reproduce this behavior
The front desk runs for a long time

### Mobile device model and OS version
android 7.1.2 (25)，google:Nexus 6P:arm64-v8a

### Android Beacon Library version
Jul 8, 2017 Commits code

davidgyoung, 2017-08-02:
@lujianzhao, sorry, I do not fully understand.  A few questions:

* By "the front desk runs for a long time" do you mean "The app runs for a long time in the foreground" ?

* How long is "a long time"?  5 minutes?  5 hours?  5 days?  

* Can you attach a LogCat excerpt (including system level messages so I can look for Bluetooth Stack errors) for a few minutes before this problem occurs?

davidgyoung, 2017-08-04:
@lujianzhao, happy to help here, but this sounds like a better question for StackOverflow.com, where we help users get started using the library.  Please open a question there for assistance.

closed 2017-08-04 by davidgyoung