AltBeacon/android-beacon-library #560: RuntimeException Android O

DejanRistic, 2017-08-10:
We've noticed a ton of crashes coming from alt beacons when it tries to start a services intent when the app is in the background. Looks like its mostly on the newer android O (8.0) which isn't fully out yet but a bunch of our users have adapted the beta already.

```
java.lang.IllegalStateException: Not allowed to start service Intent { cmp=com.roadtrippers/org.altbeacon.beacon.service.BeaconService }: app is in background uid UidRecord{d853383 u0a4 RCVR idle procs:1 seq(0,0,0)}
```

full crash report: 
http://crashes.to/s/11bc7a4b416

davidgyoung, 2017-08-10:
Thanks for this report.  

We just released a 2.12 version of the library to address Android O compatibility issues, so upgrading to that library version should resolve this crash.  I highly recommend upgrading to 2.12 and re-releasing your app soon, as this problem will become more common on Pixel phones when they get the upgrade soon after Android 8 goes live.

You can read more about this library release here:

http://www.davidgyoungtech.com/2017/08/07/beacon-detection-with-android-8

DejanRistic, 2017-08-10:
Ah okay i missed the release. Thanks! ill give it a shot in our next update.

davidgyoung, 2017-08-11:
closing this for now.  Please reopen if you find the 2.12 release does not resolve the issue.

closed 2017-08-11 by davidgyoung

r2DoesInc, 2017-11-06:
Im still seeing this on version     compile 'org.altbeacon:android-beacon-library:2.12.3@aar'

`Fatal Exception: java.lang.RuntimeException: Unable to create application com.futurehax.msz.UberApplication: java.lang.IllegalStateException: Not allowed to start service Intent { cmp=com.futurehax.msz/.services.BluetoothWatcher }: app is in background uid UidRecord{809d5d7 u0a254 TRNB idle procs:1 seq(0,0,0)}
       at android.app.ActivityThread.handleBindApplication(ActivityThread.java:5810)
       at android.app.ActivityThread.-wrap1(Unknown Source)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1664)
       at android.os.Handler.dispatchMessage(Handler.java:105)
       at android.os.Looper.loop(Looper.java:164)
       at android.app.ActivityThread.main(ActivityThread.java:6600)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:772)
Caused by java.lang.IllegalStateException: Not allowed to start service Intent { cmp=com.futurehax.msz/.services.BluetoothWatcher }: app is in background uid UidRecord{809d5d7 u0a254 TRNB idle procs:1 seq(0,0,0)}
       at android.app.ContextImpl.startServiceCommon(ContextImpl.java:1524)
       at android.app.ContextImpl.startService(ContextImpl.java:1480)
       at android.content.ContextWrapper.startService(ContextWrapper.java:649)
       at com.futurehax.msz.UberApplication.setupBluetooth(UberApplication.java:44)
       at com.futurehax.msz.UberApplication.onCreate(UberApplication.java:37)
       at android.app.Instrumentation.callApplicationOnCreate(Instrumentation.java:1118)
       at android.app.ActivityThread.handleBindApplication(ActivityThread.java:5807)
       at android.app.ActivityThread.-wrap1(Unknown Source)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1664)
       at android.os.Handler.dispatchMessage(Handler.java:105)
       at android.os.Looper.loop(Looper.java:164)
       at android.app.ActivityThread.main(ActivityThread.java:6600)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:772)
`

stephenruda, 2017-11-06:
I'm also having the same issue on 2.12.2

`Fatal Exception: java.lang.RuntimeException: Unable to create application *retracted*: java.lang.IllegalStateException: Not allowed to start service Intent { cmp=*retracted*/org.altbeacon.beacon.service.BeaconService }: app is in background uid UidRecord{e14db10 u0a107 TRNB idle procs:1 seq(0,0,0)}
       at android.app.ActivityThread.handleBindApplication(ActivityThread.java:5794)
       at android.app.ActivityThread.-wrap1(Unknown Source)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1661)
       at android.os.Handler.dispatchMessage(Handler.java:105)
       at android.os.Looper.loop(Looper.java:164)
       at android.app.ActivityThread.main(ActivityThread.java:6541)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)
Caused by java.lang.IllegalStateException: Not allowed to start service Intent { cmp=*retracted*/org.altbeacon.beacon.service.BeaconService }: app is in background uid UidRecord{e14db10 u0a107 TRNB idle procs:1 seq(0,0,0)}
       at android.app.ContextImpl.startServiceCommon(ContextImpl.java:1505)
       at android.app.ContextImpl.startService(ContextImpl.java:1461)
       at android.content.ContextWrapper.startService(ContextWrapper.java:644)
       at org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer.bindService(RegionBootstrap.java:171)
       at org.altbeacon.beacon.BeaconManager.bind(BeaconManager.java:423)
       at org.altbeacon.beacon.startup.RegionBootstrap.<init>(RegionBootstrap.java:77)
       at *retracted*.BeaconHelper.<init>(BeaconHelper.java:46)
       at *retracted*.MyApplication.onCreate(MyApplication.java:71)
       at android.app.Instrumentation.callApplicationOnCreate(Instrumentation.java:1118)
       at android.app.ActivityThread.handleBindApplication(ActivityThread.java:5791)
       at android.app.ActivityThread.-wrap1(Unknown Source)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1661)
       at android.os.Handler.dispatchMessage(Handler.java:105)
       at android.os.Looper.loop(Looper.java:164)
       at android.app.ActivityThread.main(ActivityThread.java:6541)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)`

r2DoesInc, 2017-11-06:
Nevermind, see https://developer.android.com/about/versions/oreo/background.html, its an issue in our apps.

davidgyoung, 2017-11-06:
@stephenruda, the library version 2.12+ detects Android 8, and if it finds it, it is designed to not start up an IntentService as shown in the stack trace, instead using a JobSceduler.  Earlier versions of the library, however, would have this problem you show.  

Are you certain the stack trace is from an app running 2.12+?

stephenruda, 2017-11-06:
Hi @davidgyoung, thanks for that information.  Yes I am on 2.12.2 but I am using `setEnableScheduledScanJobs(false);`.  I was attempting to scans more frequently than 15 minutes as discussed [here](https://github.com/AltBeacon/android-beacon-library/issues/586).  Because the JobScheduler sets up the periodic scans at 15 minutes I was disabling it.

I will have to take a new approach.  Let's say I have a JobScheduler within my application that fires independent non-periodic jobs every 1 minute.  Is there a method I can call from the android-beacon-library that will perform a scan at that exact time?  Currently, to trigger a scan I have just been significantly lowering the scanInterval to trigger a scan and then raising the scanInterval after the scan completes.  Is there an easier way?

davidgyoung, 2017-11-06:
@stephenruda, I am happy to discuss options to circumvent Android 8's background ScheduledJob and Service limits.  But this is a bit off topic for this issue which is about a RuntimeException.  If you open a new issue with this as the topic I can offer some suggestions.

davidgyoung, 2017-11-06:
Or if you wish, just add as a comment to your isdue #586, which is basically on the same topic you suggest.

priteshyugtia, 2018-11-21:
I got crash when kill app in android oreo 8.1.0 and i am using org.altbeacon:android-beacon-library:2.15.2 version
Caused by: java.lang.IllegalStateException: Not allowed to start service Intent { cmp=org.altbeacon.beaconreference/org.altbeacon.beacon.service.BeaconService }: app is in background uid UidRecord{6b25193 u0a144 RCVR idle change:uncached procs:1 seq(0,0,0)}