AltBeacon/android-beacon-library #564: BeaconTransmitter initialized when Bluetooth OFF causes issues

parthrparekh93, 2017-08-16:
When I initialize BeaconTransmitter with Bluetooth OFF, I am unable to start advertising which is correct. However, when I switch Bluetooth ON later, it still throws NullPointerException. I realized that this is because BeaconTranmsitter has BluetoothLeAdvertiser as an instance variable and it is null because it was initialized when Bluetooth was OFF.

### Expected behavior
When I switch my Bluetooth ON later, it should change the BluetoothLeAdvertiser instance so that advertising can begin

### Actual behavior
When I switch my Bluetooth ON later, BluetoothLeAdvertiser instance is null and cannot start advertising. This is what I think. If there is a workaround or a good reason to keep it as it is, let me know too :)

### Steps to reproduce this behavior
Initialize BeaconTransmitter when Bluettooth is OFF. Start advertising. It will throw exceptions. Switch Bluetooth ON. It will still throw exception 

Cannot start advertising due to exception                                                                                                                 java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.le.BluetoothLeAdvertiser.startAdvertising(android.bluetooth.le.AdvertiseSettings, android.bluetooth.le.AdvertiseData, android.bluetooth.le.AdvertiseCallback)' on a null object reference

### Mobile device model and OS version

### Android Beacon Library version
2.11
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

steff107, 2020-08-20:
Same happens in my case :/
How can I check wether advertiser is null without checking for bluetooth while initializing?

davidgyoung, 2020-08-20:
You can call:

If (BeaconTransmitter.checkTransmissionSupported(context)) ==
BeaconTransmitter.SUPPORTED) {
BeaconTransmitter = new BeaconTransmitter(context);
}

If Bluetooth is off the check will return
NOT_SUPPORTED_CANNOT_GET_ADVERTISER

DarckBlezzer, 2025-03-17:
And what was the solution for this problem?