AltBeacon/android-beacon-library #567: ScanRecord keeps working and logging a lot in Android 7

aloz77, 2017-08-20:
### Expected behavior

AltBeacon stops anything after 

`08-20 15:27:57.229 10022-10022/de.my.app I/BeaconService: onDestroy called.  stopping scanning`

### Actual behavior

Beacons can be detected, but after stopping BeaconService ScanRecord continues running and logging lots of:

```
08-20 15:21:46.741 28307-28319/de.my.app D/ScanRecord: first manudata for manu ID
08-20 15:21:46.743 28307-28320/de.my.app D/ScanRecord: parseFromBytes
```
Even after exiting the main activity. No way to stop it?

### Steps to reproduce this behavior

```
beaconManager.stopRangingBeaconsInRegion(myRegion);
beaconManager.removeAllRangeNotifiers();
beaconManager.unbind(myActivity);
```

### Mobile device model and OS version

Samsung S7, Android 7.0

### Android Beacon Library version

2.11

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2017-08-22:
Thanks for this report.  I believe the problem is that if scanning is on when the unbind() call happens, the scanner is left on.  This is certainly a bug that needs to be fixed.

davidgyoung, 2017-08-27:
Fixed in #569

closed 2017-08-27 by davidgyoung