AltBeacon/android-beacon-library #568: java.io.FileNotFoundException: /data/user/0/my.app/files/android-beacon-library-scan-state (No such file or directory)

ncapdevi, 2017-08-24:
### Expected behavior
Error logged on startup
### Actual behavior

### Steps to reproduce this behavior

### Mobile device model and OS version
Nexus 6P Android 8
### Android Beacon Library version
2.12.1
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

      java.io.FileNotFoundException: /data/user/0/com.roqbot.client.beta/files/android-beacon-library-scan-state (No such file or directory)
                                        at java.io.FileInputStream.open0(Native Method)
                                        at java.io.FileInputStream.open(FileInputStream.java:200)
                                        at java.io.FileInputStream.<init>(FileInputStream.java:150)
                                        at android.app.ContextImpl.openFileInput(ContextImpl.java:514)
                                        at android.content.ContextWrapper.openFileInput(ContextWrapper.java:189)
                                        at org.altbeacon.beacon.service.ScanState.restore(ScanState.java:142)
                                        at org.altbeacon.beacon.service.ScanJobScheduler.applySettingsToScheduledJob(ScanJobScheduler.java:91)
                                        at org.altbeacon.beacon.BeaconManager.applyChangesToServices(BeaconManager.java:950)
                                        at org.altbeacon.beacon.BeaconManager.startMonitoringBeaconsInRegion(BeaconManager.java:890)
                                        at org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer.onBeaconServiceConnect(RegionBootstrap.java:155)
                                        at org.altbeacon.beacon.BeaconManager.bind(BeaconManager.java:410)
                                        at org.altbeacon.beacon.startup.RegionBootstrap.<init>(RegionBootstrap.java:58)
                                        at com.my.app.setupBeacon(MyApp.java:121)
                                        at com.my.app.onCreate(MyApp.java:51)
                                        at android.app.Instrumentation.callApplicationOnCreate(Instrumentation.java:1118)
                                        at android.app.ActivityThread.handleBindApplication(ActivityThread.java:5791)
                                        at android.app.ActivityThread.-wrap1(Unknown Source:0)
                                        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1661)
                                        at android.os.Handler.dispatchMessage(Handler.java:105)
                                        at android.os.Looper.loop(Looper.java:164)
                                        at android.app.ActivityThread.main(ActivityThread.java:6541)
                                        at java.lang.reflect.Method.invoke(Native Method)
                                        at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
                                        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)

davidgyoung, 2017-08-24:
Thanks for this report.  While this does not indicate any problem (it is expected that at first run there will be no file), an error should not be logged.

davidgyoung, 2017-08-27:
Fixed in #570

PR #570 (2017-08-29): Suppress file not found on scan state deserialize
Fixes #568 showing an ugly stack trace during normal operations on first run of an app when the serialization file does not exist yet.
closed 2017-08-27 by davidgyoung