AltBeacon/android-beacon-library #571: Android O StackOverflowError in updateMonitoringStatusTime during ScanJob

ryot, 2017-08-25:
### Expected behavior
No crash in ScanJob

### Actual behavior
`Fatal Exception: java.lang.StackOverflowError: stack size 8MB
       at java.io.File.exists(File.java:807)
       at android.app.ContextImpl.ensurePrivateDirExists(ContextImpl.java:559)
       at android.app.ContextImpl.ensurePrivateDirExists(ContextImpl.java:550)
       at android.app.ContextImpl.getFilesDir(ContextImpl.java:595)
       at android.app.ContextImpl.getFileStreamPath(ContextImpl.java:693)
       at android.content.ContextWrapper.getFileStreamPath(ContextWrapper.java:205)
       at org.altbeacon.beacon.service.MonitoringStatus.updateMonitoringStatusTime(MonitoringStatus.java:204)
       at org.altbeacon.beacon.service.MonitoringStatus.updateNewlyOutside(MonitoringStatus.java:112)
       at org.altbeacon.beacon.service.ScanHelper$1.onCycleEnd(ScanHelper.java:215)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:373)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:363)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:323)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:422)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:363)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:323)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:422)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:363)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:323)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:422)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:363)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:323)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:422)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:363)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:323)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:422)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:363)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:323)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:422)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:363)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:323)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:422)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:363)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:323)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:422)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:363)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:323)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:422)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:363)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:323)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:422)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:363)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:323)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.start(CycledLeScanner.java:215)
       at org.altbeacon.beacon.service.ScanJob.restartScanning(ScanJob.java:167)
       at org.altbeacon.beacon.service.ScanJob.startScanning(ScanJob.java:190)
       at org.altbeacon.beacon.service.ScanJob.onStartJob(ScanJob.java:77)
       at android.app.job.JobService$1.onStartJob(JobService.java:71)
       at android.app.job.JobServiceEngine$JobHandler.handleMessage(JobServiceEngine.java:108)
       at android.os.Handler.dispatchMessage(Handler.java:105)
       at android.os.Looper.loop(Looper.java:164)
       at android.app.ActivityThread.main(ActivityThread.java:6541)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)
`

### Steps to reproduce this behavior
Above stack trace is from the Crashlytics logs of a number of users who are early adopters of Android O. So far I've been unable to reproduce this on my Nexus 5X running the latest Android O, perhaps it doesn't happen consistently.

### Mobile device model and OS version
Android O (8.0)
Google Pixel
Google Pixel XL
Nexus 6P
Nexus 5X

### Android Beacon Library version
2.12.1

davidgyoung, 2017-08-26:
Thanks for this report.  I will investigate.

davidgyoung, 2017-08-26:
Update:  Still having trouble reproducing this issue.

I cannot happen if you set up monitoring in the onCreate() method of a custom Application class, because this will always recreate the serialization file.  I am trying to reproduce by removing that code from the reference app and instead starting monitoring in an Activity, then killing the app and deleting its serialization file.  (This would simulate such an app that does not initialize scanning in Application#onCreate and is later upgraded to library version 2.12+)

davidgyoung, 2017-08-26:
Could not cause it on my first attempt.  But I think I need to have a monitoring state file but no scan state file, then it might happen.

```
08-26 14:23:33.645 18422 18422 E ScanState: Deserialization exception
08-26 14:23:33.651 18422 18422 E ScanState: error: 
08-26 14:23:33.651 18422 18422 E ScanState: java.io.FileNotFoundException: /data/user/0/org.altbeacon.beaconreference/files/android-beacon-library-scan-state: open failed: ENOENT (No such file or directory)
08-26 14:23:33.651 18422 18422 E ScanState: 	at libcore.io.IoBridge.open(IoBridge.java:452)
08-26 14:23:33.651 18422 18422 E ScanState: 	at java.io.FileInputStream.<init>(FileInputStream.java:76)
08-26 14:23:33.651 18422 18422 E ScanState: 	at android.app.ContextImpl.openFileInput(ContextImpl.java:384)
08-26 14:23:33.651 18422 18422 E ScanState: 	at android.content.ContextWrapper.openFileInput(ContextWrapper.java:177)
08-26 14:23:33.651 18422 18422 E ScanState: 	at org.altbeacon.beacon.service.ScanState.restore(ScanState.java:142)
08-26 14:23:33.651 18422 18422 E ScanState: 	at org.altbeacon.beacon.service.ScanJob.restartScanning(ScanJob.java:150)
08-26 14:23:33.651 18422 18422 E ScanState: 	at org.altbeacon.beacon.service.ScanJob.startScanning(ScanJob.java:199)
08-26 14:23:33.651 18422 18422 E ScanState: 	at org.altbeacon.beacon.service.ScanJob.onStartJob(ScanJob.java:77)
08-26 14:23:33.651 18422 18422 E ScanState: 	at android.app.job.JobService$JobHandler.handleMessage(JobService.java:126)
08-26 14:23:33.651 18422 18422 E ScanState: 	at android.os.Handler.dispatchMessage(Handler.java:102)
08-26 14:23:33.651 18422 18422 E ScanState: 	at android.os.Looper.loop(Looper.java:148)
08-26 14:23:33.651 18422 18422 E ScanState: 	at android.app.ActivityThread.main(ActivityThread.java:5417)
08-26 14:23:33.651 18422 18422 E ScanState: 	at java.lang.reflect.Method.invoke(Native Method)
08-26 14:23:33.651 18422 18422 E ScanState: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
08-26 14:23:33.651 18422 18422 E ScanState: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
08-26 14:23:33.651 18422 18422 E ScanState: Caused by: android.system.ErrnoException: open failed: ENOENT (No such file or directory)
08-26 14:23:33.651 18422 18422 E ScanState: 	at libcore.io.Posix.open(Native Method)
08-26 14:23:33.651 18422 18422 E ScanState: 	at libcore.io.BlockGuardOs.open(BlockGuardOs.java:186)
08-26 14:23:33.651 18422 18422 E ScanState: 	at libcore.io.IoBridge.open(IoBridge.java:438)
08-26 14:23:33.651 18422 18422 E ScanState: 	... 14 more
08-26 14:23:33.652 18422 18422 D MonitoringStatus: Not restoring monitoring state because it was recorded too many milliseconds ago: 1503750213652
08-26 14:23:33.652 18422 18422 D ScanState: Scan state restore regions: monitored=0 ranged=0
08-26 14:23:33.652 18422 18422 I CycledLeScanner: This is Android 5.0.  We are using new scanning APIs
08-26 14:23:33.657 18422 18422 D CycledLeScanner: Set scan periods called with 0, 0 Background mode must have changed.
08-26 14:23:33.657 18422 18422 D CycledLeScanner: We are not in the background.  Cancelling wakeup alarm
08-26 14:23:33.657 18422 18422 D CycledLeScanner: cancel wakeup alarm: null
08-26 14:23:33.659 18422 18422 D CycledLeScanner: Set a wakeup alarm to go off in 9223372036852938110 ms: PendingIntent{1b4ca6a: android.os.BinderProxy@7a2ab5b}
08-26 14:23:33.659 18422 18422 I ScanJob : Warning! starting scan with scan period of zero.
08-26 14:23:33.660 18422 18422 D CycledLeScanner: stop called
08-26 14:23:33.660 18422 18422 D CycledLeScanner: scanning already stopped
08-26 14:23:33.660 18422 18422 I ScanJob : No monitored or ranged regions. Scan job complete.
```

davidgyoung, 2017-08-27:
Fixed in #572

closed 2017-08-27 by davidgyoung

DejanRistic, 2017-08-28:
@davidgyoung curious if there is going to be a release with this bug fix soon, high volume of our users are already on O and we're are getting slammed with this. 

thanks!

davidgyoung, 2017-08-28:
I'm planning a release later this week.  If you need something sooner, I think it would be safe to build a binary yourself and bundle with your app.  The fixes are pretty straightforward.

If you are worried about reports of crashes, they probably seem worse than they are.  In the typical app setup, you will get one crash at each phone reboot until the  first time the app is launched manually by the user after the 8.0 upgrade.  A user may not even notice this crash, but beacon detections will not work until the app is manually launched once or until you upgrade the app with this fix.

DejanRistic, 2017-08-28:
This sounds good, thanks for the quick response.