AltBeacon/android-beacon-library #573: Beacon sent out cannot be detected by ios.

Hubertzzz, 2017-08-29:
###Expected behavior
The beacon signals sent out by android also can be detected by ios using locate.
### Actual behavior
Only the android version locate can detect the signal sent out by android, ios cannot.
### Steps to reproduce this behavior
Using the beacon layout with ibeacon "m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24", Manufacturer with estimote one "0x015D", others same as the sample code. Then send out but cannot be detected by ios, although using the ibeacon layouut.
### Mobile device model and OS version
Oneplus 3T, andorid 7.1.1
### Android Beacon Library version
2.9.2
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2017-08-29:
For detection on Apple devices the company code must be 0x004c

Hubertzzz, 2017-08-30:
I have already changed to 0x004c, but also cannot be detected. idk why.

Hubertzzz, 2017-08-30:
And here is my code.
[image]

davidgyoung, 2017-08-30:
Code looks OK.  Two thoughts:

1. Is the UUID (Id1) you are using configured in Locate iOS?  As you may know iOS must be pre-configured for the beacon UUIDs to be detected or it will not be able to do so.

2. Are you sure the Android device supports transmission?  Not all do. Use the callback interface to make sure it starts OK:

```
     beaconTransmitter.startAdvertising(beacon, new AdvertiseCallback() {

            @Override
            public void onStartFailure(int errorCode) {
                Log.e(TAG, "Advertisement start failed with code: "+errorCode);
            }

            @Override
            public void onStartSuccess(AdvertiseSettings settingsInEffect) {
                Log.i(TAG, "Advertisement start succeeded.");
            }
        });
```

Hubertzzz, 2017-08-30:
Thanks for your reply.
I forgot to add the UUID into the Locate iOS. After I added, it can be detected.
Thank you so much.

davidgyoung, 2017-08-30:
Glad to hear that fixed it!

closed 2017-08-30 by davidgyoung