AltBeacon/android-beacon-library #574: regionBootstrap notifier never called on Android O

esogt, 2017-08-29:
### Expected behavior
BootstrapNotifier interface methods are called while running monitor region.

### Actual behavior
BootstrapNotifier interface methods never called.
But it works fine on Android N, also M.

### Steps to reproduce this behavior
Implement region monitor with regionBootstrap and run app.

### Mobile device model and OS version
Nexus 5X, Android 8.0 (bullhead-opr6.170623.013)

### Android Beacon Library version
2.12.1

davidgyoung, 2017-08-29:
I have tested that this works with the reference app so perhaps there is something different about your test setup.  A few questions:

1. Do you see this problem on clean app install, or only after upgrade of an existing installation to Android 8?
2. Which background condition are you testing (a) app launched manually then put to the background (b) app killed by task switcher (c) reboot phone
3. How long do you wait for a detection?

Depending on the answers to the above it may be useful to set `beaconManager.setDebug(true)` then capture a long LogCat excerpt (25 min) after the beacon is turned on in range but there is no detection.

esogt, 2017-08-30:
Thanks for investigation.

My answer:
    1. I had tested with the clean install app after upgrade of Android 8.
    2. It seems to repro in the foreground condition.
    3. A few minutes (5 min or so).

I have checked for `beaconManager.setDebug(true)` . It seems that the beacons has been detected and parsed the advertised packets fine but BootstrapNotifier was not fired.

davidgyoung, 2017-08-30:
I probably need to see your setup code and log excerpt showing the detection and call back failure to help more, as I cannot reproduce with the reference app.

setheclark, 2017-09-20:
I too am having this issue.  Originally I had this issue in me app that doesn't utilize the RegionBootstrap class, but I'd see if I could reproduce my issues in the reference app and I was able to.

Like I mentioned, I'm using the reference app with only modifications added to target API 26 and to add my particular beacon and beacon pattern.

With the app in the foreground the beacon is detected as expected, but if I've backed out of the application (leaving it in recents) and then turn on the beacon the app is never notified.

My test device is a Nexus 6P with Android 8.0.0.  I have tested the same code on a Nexus 10 with Android 7.1.1 and beacons are recognized with the app in the background as expected.

Here is the log output from my 6P during my test.
[beacon_log.txt](https://github.com/AltBeacon/android-beacon-library/files/1318937/beacon_log.txt)

davidgyoung, 2017-09-21:
Thanks for reproducing with the reference app, @setheclark.  Can you please explain exactly what you see during the test shown in the log?

I do see a didEnterRegion get received here:

```
09-20 13:58:53.081  1273  1376 D BeaconParser: This is a recognized beacon advertisement -- 4c 00 02 15 seen
09-20 13:58:53.082  1273  1376 D BeaconParser: Bytes are: 0201061aff4c0002152f234454cf6d4a0fadf2f4911ba9ffa600051388c50000000000000000000000000000000000000000000000000000000000000000
09-20 13:58:53.085  1273  1376 D ScanHelper: Beacon packet detected for: id1: 2f234454-cf6d-4a0f-adf2-f4911ba9ffa6 id2: 5 id3: 5000 with rssi -68
09-20 13:58:53.088  1273  1376 D ScanHelper: beacon detected : id1: 2f234454-cf6d-4a0f-adf2-f4911ba9ffa6 id2: 5 id3: 5000
09-20 13:58:53.089  1273  1376 D Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.monitor_notification
09-20 13:58:53.094  1273  1273 D IntentHandler: got monitoring data
09-20 13:58:53.095  1273  1273 D IntentHandler: Calling monitoring notifier: org.altbeacon.beaconreference.BeaconReferenceApplication@97d1a78
09-20 13:58:53.095  1273  1273 D BeaconReferenceApp: did enter region.
09-20 13:58:53.095  1273  1273 D BeaconReferenceApp: auto launching MainActivity
```

and it looks like it continually gets detected during background scans in the log file, so it stays inside the defined Region.  The last detection time being here:

```
09-20 14:12:11.762  1273  1893 D ScanHelper: beacon detected : id1: 2f234454-cf6d-4a0f-adf2-f4911ba9ffa6 id2: 5 id3: 5000
```

What do you see on the device that is different than expected.

setheclark, 2017-09-21:
I see nothing.  On pre-O devices the device will display a notification, but on O it does nothing.  This is similar to what I was seeing in my own app.  I see that the library detects the beacon, but it never notifies the application code.

davidgyoung, 2017-09-21:
Hmmm... I suspect this may be a notification issue not a beacon detection
issue.  Note that the reference app is designed to launch an Activity on
first detection and send a notification on subsequent detections.  So I
think this may be expected behavior, yes?

setheclark, 2017-09-21:
It is a notification thing.  I forgot to add notification channel support to the reference app which means the notification will not be shown by the OS when on O.  Let me update that and get back.  Its still concerning that my own app also isn't showing notifications on beacon detection (I do have channels set up) when the app is in the background, but it might not be that I can reproduce the same issue in the reference app.