AltBeacon/android-beacon-library #577: ConcurrentModificationException on Google Pixel Android O 8.0 at startPassiveScanIfNeeded(ScanJob.java:106)

TonyTangAndroid, 2017-09-01:
### Expected behavior
No exception.
### Actual behavior
Throw the following exception:

> java.lang.RuntimeException: java.util.ConcurrentModificationException
                                         at android.app.job.JobServiceEngine$JobHandler.handleMessage(JobServiceEngine.java:121)
                                         at android.os.Handler.dispatchMessage(Handler.java:105)
                                         at android.os.Looper.loop(Looper.java:164)
                                         at android.app.ActivityThread.main(ActivityThread.java:6541)
                                         at java.lang.reflect.Method.invoke(Native Method)
                                         at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
                                         at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)
                                      Caused by: java.util.ConcurrentModificationException
                                         at java.util.HashMap$HashIterator.nextNode(HashMap.java:1441)
                                         at java.util.HashMap$KeyIterator.next(HashMap.java:1465)
                                         at org.altbeacon.beacon.service.ScanJob.startPassiveScanIfNeeded(ScanJob.java:106)
                                         at org.altbeacon.beacon.service.ScanJob.onStopJob(ScanJob.java:137)
                                         at android.app.job.JobService$1.onStopJob(JobService.java:76)
                                         at android.app.job.JobServiceEngine$JobHandler.handleMessage(JobServiceEngine.java:117)
                                         at android.os.Handler.dispatchMessage(Handler.java:105) 
                                         at android.os.Looper.loop(Looper.java:164) 
                                         at android.app.ActivityThread.main(ActivityThread.java:6541) 
                                         at java.lang.reflect.Method.invoke(Native Method) 
                                         at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240) 
                                         at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767) 

### Steps to reproduce this behavior
I could not stably reproduce this issue. But whenever it is reproduced, it is under real slow network.
### Mobile device model and OS version
Google Pixel Android O 8.0
### Android Beacon Library version
2.12.1

davidgyoung, 2017-09-02:
Thanks for this report.  I will investigate.

davidgyoung, 2017-09-02:
Fixed in #578

closed 2017-09-02 by davidgyoung

TonyTangAndroid, 2017-09-11:
Hi @davidgyoung I had run into this crash again in Nexus 5 Android 6.0.1 model. And the crash is quite often and could be reproduced easily.  I tested 2 Nexus 5 devices and this issue has been reproduced on both of them. Is there any plan to release another hotfix build in the near future?

davidgyoung, 2017-09-12:
I'd like to get a release together that covers all three CMEs.  I'm still working through one issue.  I'll let you know when I have a test build.

davidgyoung, 2017-09-14:
@TonyTangAndroid, please see a beta release here with instructions on how to link it with your app:

https://github.com/AltBeacon/android-beacon-library/releases/tag/2.12.3-beta2

Can you please give this a try and let me know if it solves these issues?

TonyTangAndroid, 2017-09-14:
@davidgyoung Sure. I am to do right away. Really appreciate you making effort to resolve this issue so promptly.

TonyTangAndroid, 2017-09-21:
@davidgyoung Just want you to know that after I upgrade to 2.12.3-beta2, I have not notice any crashes. Thank you again.

davidgyoung, 2017-09-21:
Thanks for the update! I have merged the changes.