AltBeacon/android-beacon-library #580: Any Option to opt out Job Schedule on Android O

TonyTangAndroid, 2017-09-05:
In the release 2.12.2,  a lot of effort has been made to support background scanning in Android O by scheduling a job around every 15 minutes (depending on system).  While I do understand that this is necessary to keep backwards compatibility for this library, is there any possible to opt out Job Schedule on Android O? There has been a reason that Google enforces the background service limitation. And in this library, by scheduling periodic job to keep background scanning sounds like not a legitimacy idea. 

There is another case that might cause confusion for user because of periodic job schedule. When a user does not want the app keeping scanning the beacon, what they would normally do is to swiping the app from recent history to "Kill" it. But in reality, we all know that it might not be killed if there is ongoing foreground service running. And even it has been killed,  the app will start again because of the scheduled periodic job, which is not something an ordinary user want.  

Any thoughts on how to address this issue?

strooooke, 2017-09-06:
If you have this problem right now, you can implement your own logic to stop monitoring/ranging when your app exits the foreground (see https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/java/org/altbeacon/beacon/powersave/BackgroundPowerSaver.java for inspiration) and resume monitoring/ranging when it enters the foreground again.

Note that altbeacon relies on you as the app developer anyway to tell it about foreground/background state; `BackgroundPowerSaver` is just one canned solution for that which does not take e.g. a foregroundservice you might have running into account. Instead of switching to background mode, stop the scan altogether (no need to unbind from the service).

davidgyoung, 2017-09-07:
I agree with @TonyTangAndroid  that it is reasonable to be able to disable the ScanJob on Android 8 devices because it does represent a change from the way pre-2.12 library versions worked, at least for Android 8.  Previously, if you started ranging or monitoring and your app was killed, it wasn't restarted automatically unless you used RegionBootstrap in an Application class or otherwise started monitoring/ranging in an Application onCreate or with another other startup hook.  As of 2.12, Android 8 devices will restart scanning even if you don't explicitly do this.  For some apps, that might not be desired behavior.

@strooooke  is correct that you can build your own solution, but I agree with @TonyTangAndroid that it is a good idea for the library to provide a built in way to do this.

There is a very simple way right now -- simply call `beaconManager.setEnableScheduledScanJobs(false);`  But if you make this call, background scanning will be stopped after about 15 minutes per the new rules of Android 8.  

This solution doesn't fully address @TonyTangAndroid's concerns in my mind, because it doesn't allow a way to do use ScanJobs after the app is launched but then disable them if the app is killed from the task switcher or the phone is rebooted.  I'm not sure such a change is even possible since you can't detect an app being killed by the task switcher anyway.

TonyTangAndroid, 2017-09-08:
Thank @strooooke @davidgyoung for your constructive thoughts and suggestion. I would choose to disable the Android Oreo Background Scanning first for the coming release of our application.

closed 2018-10-10 by davidgyoung