AltBeacon/android-beacon-library #581: Opportunistic scan on Smart Watches (Android 7.1.1)

Masaaki-Sato, 2017-09-06:
### Expected behavior
  Never move to opportunistic scan mode in long running test
### Actual behavior
  Moved to opportunistic scan mode
### Steps to reproduce this behavior
  (described in this topic)
### Mobile device model and OS version
   ASUS   ZenWatch3     (Android 7.1.1/Android Wear 2.0)
   Polar   M600               (Android 7.1.1/Android Wear 2.0)
### Android Beacon Library version
   2.12.2

 Hi, all
 
 Thank you very much for great library and support!
 
 We are trying to create beacon scan application for smart watch with this library (2.12.2)
 and found different behavior in moving to opportunistic scan mode.
  
 As it is described in this library we added following line to avoid
 opportunistic scan mode, and also changed "ScanPeriod" for our application.
 
     <meta-data android:name="longScanForcingEnabled"              ( AndroidManifest.xml (lib) )
                         android:value="true" />  
    
     beaconManager.setForegroundBetweenScanPeriod(0);               ( Activity )
     beaconManager.setForegroundScanPeriod(100);
     beaconManager.setBackgroundBetweenScanPeriod(0);
     beaconManager.setBackgroundScanPeriod(100);
     beaconManager.setDebug(true);
 
 And then I did long running test and found that watches still moved to opportunistic scan mode 
 when they are unplugged.
 Here are log data captured from Polar M600.

     09-04 14:32:13.481  3359  3359 D CycledLeScanner: Scan started
     09-04 14:32:13.487  3359 29734 D IntentHandler: got ranging data
     09-04 14:32:13.582  3359  3359 D CycledLeScanner: Done with scan cycle
     09-04 14:32:13.583  3359  3359 D ScanHelper: Calling ranging callback
     09-04 14:32:13.593  3359  3359 D Callback: attempting callback via global broadcast intent: ComponentInfo{com.wearable.watchsample/org.altbeacon.beacon.BeaconIntentProcessor}
     09-04 14:32:13.601  3359  3359 D CycledLeScanner: Not stopping scanning.  Device capable of multiple indistinct detections per scan.
     09-04 14:32:13.601  3359  3359 D CycledLeScanner: starting a new scan cycle
     09-04 14:32:13.601  3359  3359 D CycledLeScanner: We are already scanning and have been for 963161 millis
     09-04 14:32:13.602  3359  3359 D CycledLeScanner: Waiting to stop scan cycle for another 100 milliseconds
     09-04 14:32:13.603   490   621 W BtGatt.ScanManager: Moving scan client to opportunistic (clientIf 4)
     09-04 14:32:13.603   490   621 D BtGatt.ScanManager: configureRegularScanParams() - queue=1
     09-04 14:32:13.603   490   621 D BtGatt.ScanManager: configureRegularScanParams() - ScanSetting Scan mode=-1 mLastConfiguredScanSetting=2
     09-04 14:32:13.604  3359  3359 D CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{dc6130a: android.os.BinderProxy@adf577b}
     09-04 14:32:13.604  3359  3359 D CycledLeScanner: Scan started

As log data reported above, opportunistic mode seems to started at 963161 millis (about 16min)
so I changed the value of ANDROID_N_MAX_SCAN_DURATION_MILLIS to 14min.
Apparently this change works but I'm not sure if this is OK for other wearable devices or not.

Is this a OS/Device specific? 

Thank you very much in advance.

davidgyoung, 2017-09-06:
I suspect this is a device-specific optimization.  . I am not sure there is any way to make this universal unless we make the interval configurable.  If so, you could set it to 14 min in the manifest.

I am open tp.better ideas.

safakadir, 2018-10-23:
Android 7 force converts long running scans to opportunistic scan with an intent of preventing abusive apps. The duration is 30 minutes for a long running scan. This is undocumented, mentioned in this post: https://blog.classycode.com/undocumented-android-7-ble-behavior-changes-d1a9bd87d983