AltBeacon/android-beacon-library #582: Used jobIds should be documented and configurable

strooooke, 2017-09-08:
See https://commonsware.com/blog/2017/06/07/jobscheduler-job-ids-libraries.html for a much better written explanation of why it matters.

It's up to the client app to make sure that their chosen jobIds will then be unique inside their uid, and stay fixed over process restarts, device reboots, updates etc... a use case for statics if I've ever seen one. But they need to be configurable to avoid clashes.

davidgyoung, 2017-09-08:
Fair point.  Seems reasonable to make this configurable in the AndroidManifest.xml

strooooke, 2017-09-15:
Manifest configuration would work - personally, as an app developer, I'd prefer the ability to configure in Java (because I like to keep such constants (like e.g. notification ids) in some constant collection class and reference from there whereever needed; Manifest configuration would force me to keep that collection as int resources - also works, but less comfortable to reference in the majority of the code).

davidgyoung, 2018-02-18:
Resolved in #645

PR #645 (2018-02-23): Configurable scan job ids
The Job Scheduler ScanJobs used in Android 8 has a hard-coded job id of 1 and 2.  Because this job id must be unique per application, this can cause collisions with other app job ids.  This changes the default job ids to be a high integer number, and declares them in AndroidManifest.xml:

```
        <service android:name=".service.ScanJob"
                android:permission="android.permission.BIND_JOB_SERVICE" >
            <meta-data android:name="immmediateScanJobId" android:value="208352939" />
            <meta-data android:name="periodicScanJobId" android:value [... cut]
closed 2018-02-18 by davidgyoung