AltBeacon/android-beacon-library #583: RuntimeException while App in background on Android 8

dsahm, 2017-09-08:
### Expected behavior
App does not crash while in background.

### Actual behavior
App crashes after ~30-60 minutes in background.

java.lang.RuntimeException: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.le.BluetoothLeScanner.stopScan(android.app.PendingIntent)' on a null object reference
                                                                       at android.app.job.JobServiceEngine$JobHandler.handleMessage(JobServiceEngine.java:112)
                                                                       at android.os.Handler.dispatchMessage(Handler.java:105)
                                                                       at android.os.Looper.loop(Looper.java:164)
                                                                       at android.app.ActivityThread.main(ActivityThread.java:6541)
                                                                       at java.lang.reflect.Method.invoke(Native Method)
                                                                       at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
                                                                       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)
                                                                    Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.le.BluetoothLeScanner.stopScan(android.app.PendingIntent)' on a null object reference
                                                                       at org.altbeacon.beacon.service.ScanHelper.stopAndroidOBackgroundScan(ScanHelper.java:185)
                                                                       at org.altbeacon.beacon.service.ScanJob.restartScanning(ScanJob.java:160)
                                                                       at org.altbeacon.beacon.service.ScanJob.startScanning(ScanJob.java:190)
                                                                       at org.altbeacon.beacon.service.ScanJob.onStartJob(ScanJob.java:77)
                                                                       at android.app.job.JobService$1.onStartJob(JobService.java:71)
                                                                       at android.app.job.JobServiceEngine$JobHandler.handleMessage(JobServiceEngine.java:108)
                                                                       at android.os.Handler.dispatchMessage(Handler.java:105) 
                                                                       at android.os.Looper.loop(Looper.java:164) 
                                                                       at android.app.ActivityThread.main(ActivityThread.java:6541) 
                                                                       at java.lang.reflect.Method.invoke(Native Method) 
                                                                       at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240) 
                                                                       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767) 

### Steps to reproduce this behavior
I use both the RegionBootstrap and the ranging capabilities in my Application Object.
This works quite well on all devices, even on the Nexus 5X with Android 8. However, after 30-60 minutes of being in background, the app crashes.  

The code in my application object that start the ranging / monitoring:

```
public void startMonitoringBeacons() {
        try {
            beaconManager = BeaconManager.getInstanceForApplication(this);
            beaconManager.setRangeNotifier(this);
            beaconManager.startRangingBeaconsInRegion(region);
        } catch (Exception ignore) {
        }

        beaconManager.getBeaconParsers().clear();
        beaconManager.getBeaconParsers().add(new BeaconParser().
                setBeaconLayout(BeaconUtil.IBEACON_LAYOUT));
        Beacon.setHardwareEqualityEnforced(true);

        if (RestAdapterUtil.getLogLevel().equals(RestAdapter.LogLevel.FULL)) {
            BeaconManager.setDebug(true);
        }
        beaconManager.setRegionStatePersistenceEnabled(false);

        regionBootstrap = new RegionBootstrap(CoreApplication.this, region);
        backgroundPowerSaver = new BackgroundPowerSaver(this);
    }
```

### Mobile device model and OS version
Nexus 5X with Android 8, API-Level 26. The app targets API-Level 26.

### Android Beacon Library version
2.12

davidgyoung, 2017-09-08:
The problem is caused by this line: 

`bluetoothAdapter.getBluetoothLeScanner().stopScan(getScanCallbackIntent());`

Apparently the `getBluetoothLeScanner()` returns null in some cases.  It's unclear why this would be, but there obviously needs to be a guard on that to fix this issue.

shawnvega, 2017-09-14:
I'm also getting this same issue. seems like a simple null check should fix it.

davidgyoung, 2017-09-14:
Fixed in eceaa9a73a9b1986aa9efdbb1636f2eb2ec76b5c

closed 2017-09-14 by davidgyoung