AltBeacon/android-beacon-library #586: Android O - Scans more frequent than 15 minutes

stephenruda, 2017-09-11:
### Mobile device model and OS version
Google Pixel - Android 8.0

### Android Beacon Library version
2.12.1

Hi,
I am hoping we can find a way for the library to scan for Beacons more frequently than the periodic 15 minute job on Android O.

We are a Radius Networks reseller and currently use the Beacons for our digital IN/OUT board software.  When they enter or exit a Beacon, it will update their status accordingly for other users to see.  If a user is checked out 10-15 minutes late, this will lead to:
1) Inaccurate data (the user is out, but they are shown as in on the board for potentially 15 minutes).
2) Unreliable payroll.  Some of our companies use our IN/OUT board data for payroll purposes.  If a user was getting paid for an extra 15 minutes, there would be a lot of backlash.

Running a constant foreground service / foreground notification just to scan for Beacons isn't really an option.

I know that Android limits periodic jobs to every 15 minutes but I was hoping we could come up with a workaround.  I was thinking - maybe have a periodic job that runs every 15 minutes that simply starts multiple single fire jobs.
For example, if we were to do scans every 3 minutes...
- You schedule a periodic job to run every 15 minutes starting at 12:00.
- At 12:00, when the periodic job fires, it scans for Beacons and it schedules new, independent jobs for 12:03, 12:06, 12:09, 12:12.  These independent jobs scan for Beacons.
- At 12:15, the periodic job will fire again and schedule jobs for 12:18, 12:21, etc.
Alternatively, each job could schedule the next job.  So at 12:03, it scans and then schedules a job for 12:06.  At 12:06, it scans and then schedules a job for 12:09, etc.

I have set up a similar solution for scanning for WIFI networks on Android O and it seems to work well.  I am hoping we could come up with a way to do it for this library.  I'd be happy to help in a pull-request if you are open to this idea.  If you have any other ideas, please let me know!

Thanks

davidgyoung, 2017-09-12:
@stephenruda, I understand the need for a solution like this in your use case and  in general this sounds like a reasonable approach. A few concerns:

1. Periodic jobs are inexact.  A 15 minute job may be separated by 25 minutes or only 10 minutes, so you would have to schedule 25 min worth of scan jobs.

2. To prevent overlaps, you would have to cancel all the jobs that we're previously scheduled at the start of each periodic job.  The periodic job might become just a scheduler job.

3. This technique basically exploits a loophole.  I would not be surprised if Google closes this loophole in a future release.

stephenruda, 2017-09-12:
@davidgyoung Thank you for the input.  Here are some additional thoughts:

1. Given that the periodic jobs are inexact, you could use the periodic job to ensure that the scan job is always scheduled.  For example, you would have one job that scans and then reschedules itself again (say, in 3 minutes).  The periodic job could be in place just to make sure that the scanning job is always scheduled (just in case the scan job for some reason didn't fire, and therefore not reschedule itself).  This would solve problems 1 and 2.
2. Yes, I would agree that it may be a loophole.  I am sure there are plenty of apps that do require two jobs to go off within a few minutes of each other - so I am not sure if the behavior would change.  The 15 minute periodic job has been implemented since Android N and they didn't patch the loophole in Android O.  This would at least be a solution for now.

Eventually they might fix the loophole and I will have to end up using a foreground service.  But until then, I'd like to avoid upsetting any of our users with a 24/7 notification.

If you would be interested in implementing such a thing, let me know and I will work on a PR.  If not, are there any methods to trigger a background scan manually?  If so, I could set up my app to schedule it's own jobs as described above, and then just call a method that would trigger a scan (ignoring whatever values I have for my scan periods, just a method to do a scan at the current moment).

Thanks!

davidgyoung, 2017-09-12:
Sure, If you want to put together a PR, I am open to merging it into the library.  A few implementation thoughts:

* Please try to make the changes as minimal as possible.  There is already a fair amount of complexity in ScanJobs and I am reluctant to add much more.  
* You will need to assign a different job id for this rescheduled scan job.  I have the job ids as constants now, so we can follow that pattern for now, but ultimately I will make these offsets configurable per a base job id in the manifest as describied in #582
* The rescheduled scan job will need to be cancelled if a new scheduled scan job or immediate scan job starts.  Similar logic may already be in ScanJobScheduler.
* The reschedule interval should not be fixed (e.g. not 3 minutes) but should probably be the backgroundBetweenScanPeriod if scheduled at the end of the ScanJob.
* ScanJobs are also used in the foreground where the 15 minute rule does not apply, so this logic must be suppressed in the foreground.

stephenruda, 2017-11-06:
@davidgyoung 
What would be the best way to trigger a scan manually?  Essentially, I have my own Job built into my application that triggers every minute.  The default periodic job only allows 15 minutes between each execution - but through consecutive single fire jobs I have been able to trigger much more frequently.  This is my own Job that I have set up in my app, not something I have implemented in the library as I have other functionality I am putting in this job (such as scanning for WIFI networks).  I would just need a method that executes the same code within the library's ScanJob trigger that I can put in my own application's job.

In summary, what would be the best way to trigger a scan that I can just call anytime?  I don't need any of the library's automated scheduling as my own JobScheduler is doing all of that on its own; I just need a method to call to trigger the scan.

Thanks!

davidgyoung, 2017-11-06:
I believe if you manually set:

beaconManager.setBackgroundMode(false);

You can then call:

beaconManager.syncSettingsToService();

Which will schedule an immediate ScanJob because the app is in foreground mode.

You must make sure that you do not have BackgroundPowerSaver instantiated, as this will change the background mode and reset your foreground setting.  Also, keep in mind that the library's immediate foreground  ScanJob will run for five minutes.  So if you do this every minute it will cancel and restart the same job.  This may work OK, but it would need testing.

As an aside: I would be curious to hear how you are successfully getting a job to run every minute in the background.  How do you start the second job one minute later after the first one completes?  It seems this would be a good feature to add to the library -- not as a default -- but for folks that really need more frequent scans than Android 8 is designed to allow.

stephenruda, 2017-11-06:
@davidgyoung I will give this a try tomorrow and I will let you know how it goes.  Thanks!

I have found that you can schedule a one-time job and then reschedule the job again when it is fired.  For example, I have a job that fires after a minute and then when the job fires it schedules itself to go off again in a minute.  So I have a working 'periodic' job that fires every minute.  I know this is kind of a cheat and may be patched some day but it is working for now.  [I wrote a small library that shows this off.](https://github.com/simplymadeapps/QuickPeriodicJobScheduler)

I was originally intending to add this behavior to the library and add a pull-request, but I figured it could be used in more use-cases (for example, I also have to scan for WIFI networks in my application every minute).

davidgyoung, 2017-11-06:
Nice work! Any idea happens in Doze mode?

stephenruda, 2017-11-06:
My app is whitelisted and ignores battery optimizations when they have these features enabled (scanning for Beacons, Geofences, and WIFI scanning).

With that said, I have only tested the library using [adb commands](https://developer.android.com/training/monitoring-device-state/doze-standby.html#testing_doze)...
Jobs will not run with:
`adb shell dumpsys deviceidle force-idle`

Jobs will run with:
```
$ adb shell dumpsys battery unplug
$ adb shell am set-inactive <packageName> true
```

stephenruda, 2017-11-06:
@davidgyoung I have implemented your suggestion and it seems to be working well.

I noticed that `beaconManager.syncSettingsToService();` was protected so I wasn't able to call it, but I could get the same result by calling `ScanJobScheduler.getInstance().applySettingsToScheduledJob(context, beaconManager);`

When my job fires every minute I run the following code:
```
beaconManager.setBackgroundMode(false);
ScanJobScheduler.getInstance().applySettingsToScheduledJob(context, beaconManager);

// Turn it off after 8 seconds
final Handler handler = new Handler();
handler.postDelayed(new Runnable() {
    @Override
    public void run() {
        beaconManager.setBackgroundMode(true);
    }
}, 8000);
```

This appears to working pretty well.  I am stopping the ScanJob by calling `setBackgroundMode(true)` after 8 seconds.  As you noted, the ScanJob will scan continuously for 5 minutes and therefore my application was always scanning for beacons.  Stopping the ScanJob by calling `setBackgroundMode(true)` will cause my application to scan every 1 minute (my job interval) for a period of 8 seconds.

Do you see a problem with doing it this way?

I have two additional questions:

1. Because I essentially have my own scan intervals set up, what would be the best way to make the library stop doing it's own scans?  With my code, it scans for 8 seconds every minute, but then it will still scan again depending on what my `setBackgroundBetweenScanPeriod()` is set to.  Is there any way to disable these scans or would it be best to just set `setBackgroundBetweenScanPeriod()` to a very large number (for example 24 hours) so it only scans when my own job fires?
2.  I have noticed that I am seeing `didExitRegion()` and `didEnterRegion()` every couple of minutes despite being right next to the beacon.  I have set `setRegionExitPeriod(180000l)` so I wouldn't expect it to EXIT/ENTER me so frequently when I am right next to the beacon.  Could this be a side effect of how I am doing things?  Perhaps the region exit period isn't being obeyed?  I recently upgraded from Android 8.0 to Android 8.1 and that might also be the cause of the issue.

davidgyoung, 2017-11-06:
Glad to hear this is working for you.  The whole thing is a bit of a Kludge, but given that it goes against the use goes against the way the library is designed to be used there really isn't a better way.  I agree that calling `setBackgroundBetweenScanPeriod()` with a value like Long.MAX_VALUE is probably the simplest way to solve your question 1.

As for question 2, I suspect unexpected didExitRegion are caused by this unusual setup, although it isn't immediately obvious why.  Have you customized the foregroundScanPeriod and foregroundBetweenScanPeriod?

stephenruda, 2017-11-07:
@davidgyoung 
Okay I will continue testing with Long.MAX_VALUE.

I think the random didExitRegion events might be related to the Android 8.1 developer preview.  I went back to doing things the regular way (just using the library's default scan jobs) and I am noticing the same behavior.  I know it used to work correctly previously.  It will be a problem with Android 8.1 or with my beacon, not with library.  I will continue testing and I will update you when I know more.  I am currently in the process of wiping the device and going back to Android 8.0 to see if that really is the problem.

Thanks!

stephenruda, 2017-11-07:
@davidgyoung 
It seems like everything is working well.  I am pleased with how it performs - I appreciate the help!  I will continue testing it for a couple of days and then submit these changes to production.

If you would like to close this issue, feel free.  I'm not sure if you are wanting to implement the 'workaround' I am using for scheduling jobs more frequently than 15 minutes into the library.  In my opinion, it doesn't really belong in the library as a user can set it up within their own application if they need to (just like what I have done).  It is probably out of the scope of the library.

Also, everything seems to be working normally again on Android 8.0.  I am thinking that Android 8.1 was causing the issues.  Have you tested on Android 8.1 yet?  It is still not released but it is available on the beta channel.  This is an entirely separate issue so when I have time I will reinstall Android 8.1 and see if I can reproduce the problem (and potentially start a new, seprate issue here on GitHub).

Thanks for the help!

tekinalper, 2018-02-08:
@stephenruda @davidgyoung 
It might be a good (optional to use) feature within this library for the people like me who need more frequent scans than 15 minutes in the background. I will also follow the issue and try to find the best way. Thanks for your efforts.

davidgyoung, 2018-03-05:
closing as a workaround has been confirmed in the last comment by  @stephenruda

closed 2018-03-05 by davidgyoung