AltBeacon/android-beacon-library #589: Android O 8.0  java.lang.OutOfMemoryError:

MrD0llaro, 2017-09-12:
### Expected behavior
Same behavior or other devices
### Actual behavior
I see an error and a store ANR.
### Steps to reproduce this behavior
use a 
Google Pixel (sailfish), 4096MB RAM, Android 8.0
or 
Sony Xperia XZ1 (G8341), 4096MB RAM, Android 8.0

### Mobile device model and OS version
use a 
Google Pixel (sailfish), 4096MB RAM, Android 8.0
or 
Sony Xperia XZ1 (G8341), 4096MB RAM, Android 8.0

### Android Beacon Library version
2.12.2
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

java.lang.OutOfMemoryError: 
  at java.lang.Thread.nativeCreate (Thread.java)
  at java.lang.Thread.start (Thread.java:733)
  at java.util.concurrent.ThreadPoolExecutor.addWorker (ThreadPoolExecutor.java:970)
  at java.util.concurrent.ThreadPoolExecutor.execute (ThreadPoolExecutor.java:1377)
  at android.os.AsyncTask.executeOnExecutor (AsyncTask.java:651)
  at org.altbeacon.beacon.service.ScanHelper.processScanResult (ScanHelper.java:125)
  at org.altbeacon.beacon.service.ScanHelper$1.onLeScan (ScanHelper.java:207)
  at org.altbeacon.beacon.service.scanner.CycledLeScannerForLollipop$4.onScanResult (CycledLeScannerForLollipop.java:300)
  at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run (BluetoothLeScanner.java:464)
  at android.os.Handler.handleCallback (Handler.java:789)
  at android.os.Handler.dispatchMessage (Handler.java:98)
  at android.os.Looper.loop (Looper.java:164)
  at android.app.ActivityThread.main (ActivityThread.java:6541)
  at java.lang.reflect.Method.invoke (Method.java)
  at com.android.internal.os.Zygote$MethodAndArgsCaller.run (Zygote.java:240)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:767)

hustoj2, 2017-11-05:
I've been seeing similar OutOfMemoryErrors in the background on a Pixel 2 (sailfish) running Android 8.0. It appears to be happening about once a day or slightly more often. At least one of the times, the device would have been locked and stationary for hours and would not have been in range of an active beacon that matches the UUID I'm scanning for. It would have likely been in range of one or more Tile beacons.

I'm on AltBeacon 2.12.2

I'm just getting into my investigation. My next steps are to look through the logs of some other devices that have had this running for a few days, and also update to 2.12.3

Below is the stacktrace from when I wouldn't have expected any beacons nearby that matched my sought UUID.

10-27 04:30:25.884 10192 20356 20356 E AndroidRuntime: java.lang.OutOfMemoryError: pthread_create (1040KB stack) failed: Try again
10-27 04:30:25.884 10192 20356 20356 E AndroidRuntime: 	at java.lang.Thread.nativeCreate(Native Method)
10-27 04:30:25.884 10192 20356 20356 E AndroidRuntime: 	at java.lang.Thread.start(Thread.java:733)
10-27 04:30:25.884 10192 20356 20356 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor.addWorker(ThreadPoolExecutor.java:970)
10-27 04:30:25.884 10192 20356 20356 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor.execute(ThreadPoolExecutor.java:1377)
10-27 04:30:25.884 10192 20356 20356 E AndroidRuntime: 	at android.os.AsyncTask.executeOnExecutor(AsyncTask.java:651)
10-27 04:30:25.884 10192 20356 20356 E AndroidRuntime: 	at org.altbeacon.beacon.service.ScanHelper.processScanResult(ScanHelper.java:125)
10-27 04:30:25.884 10192 20356 20356 E AndroidRuntime: 	at org.altbeacon.beacon.service.ScanHelper$1.onLeScan(ScanHelper.java:207)
10-27 04:30:25.884 10192 20356 20356 E AndroidRuntime: 	at org.altbeacon.beacon.service.scanner.CycledLeScannerForLollipop$4.onScanResult(CycledLeScannerForLollipop.java:300)
10-27 04:30:25.884 10192 20356 20356 E AndroidRuntime: 	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:465)
10-27 04:30:25.884 10192 20356 20356 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:789)
10-27 04:30:25.884 10192 20356 20356 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:98)
10-27 04:30:25.884 10192 20356 20356 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:164)
10-27 04:30:25.884 10192 20356 20356 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:6600)
10-27 04:30:25.884 10192 20356 20356 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
10-27 04:30:25.884 10192 20356 20356 E AndroidRuntime: 	at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
10-27 04:30:25.884 10192 20356 20356 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:772)

davidgyoung, 2017-11-05:
So it is running out of memory trying to start a new thread to process the advertisement detected in a BLE scan.  Even if no beacons are around, the library does this to offload the work of parsing the packet (to see if it is a beacon)  from the main thread.  

It's always possible a device could be low on memory when this happens and cause such a crash.  I cannot think of why this would happen regularly.

hustoj2, 2017-11-05:
Thanks for the quick response. Maybe there's something odd going on with this phone; I'll let you know if I'm seeing similar behavior on any other devices.

davidgyoung, 2017-11-05:
**A little more detail on what is happening here:**

On Android 8.0 (all reports of this bug), an app using this library will wake up and run a scheduled ScanJob every ~ 15 minutes.  When it does, it will start a bluetooth scan that will last about 10 seconds.  If it detects a bluetooth packet, it will start a new thread to parse it.   This is the point where the crash happens because the OS does not have enough memory to allocate the new thread.

**And here is where we don't see the crash:**

The same process of allocating a new thread also happens on beacon detections when the app is in the foreground and for versions of Android prior to 8.0.  Yet the only reports of this  crashcome when Android 8.0 in the background.

**My theory**

Total conjecture here -- but I suspect this has something to do with Android 8 devices in Doze mode.  I'm not sure if this is a bug or some kind of power saving feature, but it seems that thread creation is failing in these cases.   If it were an intentional power saving feature, I would not think it would throw an OutOfMemoryError.  So it kind of sounds like an Android OS bug to me.

**Impact of this bug**

If my conjecture is correct, then the impact of this bug would be minimal for most use cases, other than crashes being logged.  It would just mean that the periodic ScanJob used to look for beacons every 15 minutes would fail while the phone is in Doze mode (a time where finding a new beacon would be unlikely anyway, as for the phone to be in Doze mode it needs to be motionless for some time.)

hustoj2, 2017-11-06:
@davidgyoung I've been very impressed by your responsiveness on this and the other one I commented on. Thank you!

Based on your earlier comment, I tried restarting my phone just to see if that would clear it up, but it's still happening on that device. This morning, the recurring OOM happened while I was in the car and it did show me an "<Your App> has stopped." dialog that I had to dismiss. So far I don't know if it happens on other Oreo devices as well, but for this one device it doesn't seem to be restricted to Doze mode.

I'm using the Take Bug Report feature of Android when this happens, so if there's more context you expect to be contained in there and you'd like to see that, let me know. I don't know what private information may be in that log so my instinct is not to post it here publicly, but I'd be happy to send it to you.

In the meantime, I'm getting a couple other Oreo devices up and scanning to see if I can reproduce this issue on additional devices.

DebaoD, 2018-06-23:
May I know is there any progress on this issue? I can regularly see this issue when my App is in background for about one hour. My device is pixel 2 with Android 8.1. And the issue is reproducible with both beacon library 2.10 and 2.14.

DebaoD, 2018-06-23:
It's very regular. About once every one hour and 20 minutes.

Line 1690: 06-22 17:48:03.810 24646 24646 E AndroidRuntime: java.lang.OutOfMemoryError: pthread_create (1040KB stack) failed: Try again
	Line 1759: 06-22 19:07:44.133 31225 31225 E AndroidRuntime: java.lang.OutOfMemoryError: pthread_create (1040KB stack) failed: Try again
	Line 1797: 06-22 20:27:22.060  5429  5429 E AndroidRuntime: java.lang.OutOfMemoryError: pthread_create (1040KB stack) failed: Try again
	Line 1854: 06-22 21:46:47.449 10706 10706 E AndroidRuntime: java.lang.OutOfMemoryError: pthread_create (1040KB stack) failed: Try again
	Line 1903: 06-22 23:06:13.774 16043 16043 E AndroidRuntime: java.lang.OutOfMemoryError: pthread_create (1040KB stack) failed: Try again
	Line 1938: 06-23 00:25:49.826 21332 21332 E AndroidRuntime: java.lang.OutOfMemoryError: pthread_create (1040KB stack) failed: Try again
	Line 2019: 06-23 01:45:26.835 26124 26124 E AndroidRuntime: java.lang.OutOfMemoryError: pthread_create (1040KB stack) failed: Try again
	Line 2081: 06-23 03:05:06.283 31955 31955 E AndroidRuntime: java.lang.OutOfMemoryError: pthread_create (1040KB stack) failed: Try again
	Line 2144: 06-23 04:24:33.872  5755  5755 E AndroidRuntime: java.lang.OutOfMemoryError: pthread_create (1040KB stack) failed: Try again
	Line 2180: 06-23 05:44:00.597 11187 11187 E AndroidRuntime: java.lang.OutOfMemoryError: pthread_create (1040KB stack) failed: Try again
	Line 2257: 06-23 07:03:37.648 16899 16899 E AndroidRuntime: java.lang.OutOfMemoryError: pthread_create (1040KB stack) failed: Try again
	Line 2332: 06-23 08:23:15.910 22695 22695 E AndroidRuntime: java.lang.OutOfMemoryError: pthread_create (1040KB stack) failed: Try again
	Line 2415: 06-23 09:42:43.728 28247 28247 E AndroidRuntime: java.lang.OutOfMemoryError: pthread_create (1040KB stack) failed: Try again
	Line 2492: 06-23 11:02:19.872  2883  2883 E AndroidRuntime: java.lang.OutOfMemoryError: pthread_create (1040KB stack) failed: Try again
	Line 2532: 06-23 12:21:54.116  8902  8902 E AndroidRuntime: java.lang.OutOfMemoryError: pthread_create (1040KB stack) failed: Try again
	Line 4399: 06-23 13:42:49.201 14397 14397 E AndroidRuntime: java.lang.OutOfMemoryError: pthread_create (1040KB stack) failed: Try again

davidgyoung, 2018-06-23:
We need to find out more information about what is going on with the phone snd app when this happens by getting a system level log capture (not filtered for your application) for a minute or more before the crash and a few seconds after.  

The stack trace indicates that a bluetooth packet easy detected, then the library tried to start a thread to parse it, and the OS ran out of memory allocating this thread.  This should not happen under normal circumstances.  

If we are to figure out the problem and find a fix, we must get that log capture.

DebaoD, 2018-06-25:
@davidgyoung, thanks for comments. I've prepared the logs. May I send to you through your email, tech@davidgyoung.com?

davidgyoung, 2018-06-25:
Sure

davidgyoung, 2018-07-05:
Thanks for this information, @andreybutenko.  I have yet to reproduce this problem with the reference app on the Galaxy S9 with android 8.0.  I see that device model is not on your list. Unfortunately, I do not have any of the device models on your list.

I'm beginning to wonder if this issue is app-specific.  The reference app has a relatively small memory footprint.  Perhaps the issue is that only if an app allocates a large amount of memory at startup, will crash like this.  

The crash happens when the Android JobScheduler launches the app in the background to perform a scan.  But this will cause the Application.onCreate() method to get called before the job starts, so perhaps this is only triggered if there are some significant allocations by the app.

davidgyoung, 2018-07-05:
It's also possible that the exception is thrown not because of a lack of memory, but because a [native thread could not be allocated](https://stackoverflow.com/questions/16789288/java-lang-outofmemoryerror-unable-to-create-new-native-thread).  If the latter is true, it is unclear what conditions would cause this to be the case.

davidgyoung, 2018-07-05:
Folks, if anybody can reproduce this on their phone (and not just see it in crash reports), please help me get the following info:

1. How long does it take the crash to happen after you run the app?
2. How many beacons are transmitting in the vicinity when the crash happens?
3. Is there anything special about the state of the phone when it happens?
4. Do you *ever* get a successful background detection of a beacon without the app crashing?

Please also capture a log like this:

1. Modify your code to set `beaconManager.setDebug(true);`
2. Capture logcat output system wide (not filtered for your app), and write it to a file so that you do not lose old log messages over time.  Like this:  `adb logcat > ~/log.txt`

Ideally, I'd like to get 20 minutes of log capture before the crash happens.

mmd44, 2018-08-13:
Hello,
I am experiencing the same issue.
Most of the crashes are on Android 8 devices after monitoring the app analytics on the dashboard.
I tried reproducing the crash but with no results.
The only major change I can figure out is modifying DEFAULT_BACKGROUND_SCAN_PERIOD and DEFAULT_BACKGROUND_BETWEEN_SCAN_PERIOD...
Hope you can come up with something.
Below is the crash report if it helps.

Thanks!

java.lang.OutOfMemoryError: 
at java.lang.Thread.nativeCreate (Native Method)
at java.lang.Thread.start (Thread.java:733)
at java.util.concurrent.ThreadPoolExecutor.addWorker (ThreadPoolExecutor.java:970)
at java.util.concurrent.ThreadPoolExecutor.execute (ThreadPoolExecutor.java:1377)
at android.os.AsyncTask.executeOnExecutor (AsyncTask.java:651)
at org.altbeacon.beacon.service.ScanHelper.processScanResult (ScanHelper.java:129)
at org.altbeacon.beacon.service.ScanJob.onStartJob (ScanJob.java:70)
at android.app.job.JobService$1.onStartJob (JobService.java:71)
at android.app.job.JobServiceEngine$JobHandler.handleMessage (JobServiceEngine.java:108)
at android.os.Handler.dispatchMessage (Handler.java:105)
at android.os.Looper.loop (Looper.java:164)
at android.app.ActivityThread.main (ActivityThread.java:6938)
at java.lang.reflect.Method.invoke (Native Method)
at com.android.internal.os.Zygote$MethodAndArgsCaller.run (Zygote.java:327)
at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1374)

ryot, 2018-09-04:
One of our clients are also experiencing this OutOfMemoryError on Android 8 devices. I will try to see if they can reproduce this crash on any of their own development devices to help diagnose this issue. Here is a screenshot of their dashboard with details about the 42 devices with the error:

[image]

And the stacktrace:

```
java.lang.OutOfMemoryError:
at java.lang.Thread.nativeCreate (Thread.java)
at java.lang.Thread.start (Thread.java:733)
at java.util.concurrent.ThreadPoolExecutor.addWorker (ThreadPoolExecutor.java:970)
at java.util.concurrent.ThreadPoolExecutor.execute (ThreadPoolExecutor.java:1377)
at android.os.AsyncTask.executeOnExecutor (AsyncTask.java:651)
at org.altbeacon.beacon.service.ScanHelper.processScanResult (ScanHelper.java:129)
at org.altbeacon.beacon.service.ScanHelper$1.onLeScan (ScanHelper.java:214)
at org.altbeacon.beacon.service.scanner.CycledLeScannerForLollipop$4.onScanResult (CycledLeScannerForLollipop.java:313)
at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run (BluetoothLeScanner.java:490)
at android.os.Handler.handleCallback (Handler.java:790)
at android.os.Handler.dispatchMessage (Handler.java:99)
at android.os.Looper.loop (Looper.java:164)
at android.app.ActivityThread.main (ActivityThread.java:6651)
at java.lang.reflect.Method.invoke (Method.java)
at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:438)
at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:810)
```

davidgyoung, 2018-09-05:
I'm still looking for somebody who can reproduce this error and can set `beaconManager.setDebug(true);` then send me a LogCat excerpt with the last 30 seconds before this happens.

Kitsopappas, 2018-10-03:
Hi,

I too have a lot of outOfMemory crashes in my app. A log file that I could obtain from a Samsung Galaxy A5 (Android 8.0).

It appears when the app was in the background for a long time. I can not find any other way to reproduce this only when the app is in the background for a long time and the scan is started.

```log
10-03 11:30:34.601 D/SurfaceFlinger( 2484):   Display[0] configurations (* current):
10-03 11:30:34.601 D/SurfaceFlinger( 2484):     * 0: 1080x1920, xdpi=422.029999, ydpi=424.069000, refresh=16949152
10-03 11:30:34.601 D/SurfaceFlinger( 2484):   numHwLayers=5, flags=00000000
10-03 11:30:34.601 D/SurfaceFlinger( 2484):     type   |  handle  | hint | flag | tr | blnd |   format    |     source crop (l,t,r,b)      |          frame         | name 
10-03 11:30:34.601 D/SurfaceFlinger( 2484): -----------+----------+------+------+----+------+-------------+--------------------------------+------------------------+------
10-03 11:30:34.601 D/SurfaceFlinger( 2484):        HWC | 7b3a2d0fc0 | 0000 | 0020 | 00 | 0100 | RGBA_8888   |    0.0,    0.0, 1080.0, 1857.0 |    0,   63, 1080, 1920 | SurfaceView - gr.terracom.spotypal.com/gr.terracom.spotypal.com.MainActivity@5dad3df@14#0
10-03 11:30:34.601 D/SurfaceFlinger( 2484):        HWC | 7b3a2595c0 | 0000 | 0000 | 00 | 0105 | RGBA_8888   |    0.0,    0.0, 1080.0,   63.0 |    0,    0, 1080,   63 | gr.terracom.spotypal.com/gr.terracom.spotypal.com.MainActivity#0
10-03 11:30:34.601 D/SurfaceFlinger( 2484):       GLES | 00000000 | 0000 | 0200 | 00 | 0105 | ? ffffffff  |   -0.2,   -0.2,   -0.8,   -0.8 |    0,    0, 1080, 1920 | DimLayerController/Stack=0#0
10-03 11:30:34.601 D/SurfaceFlinger( 2484):        HWC | 7b3a258840 | 0000 | 0000 | 00 | 0105 | RGBA_8888   |    
10-03 11:30:34.601 D/SurfaceFlinger( 2484): 0.0,    0.0, 1080.0,   63.0 |    0,    0, 1080,   63 | StatusBar#0
10-03 11:30:34.601 D/SurfaceFlinger( 2484):  FB TARGET | 7b3c3f9080 | 0000 | 0000 | 00 | 0105 | RGBA_8888   |    0.0,    0.0, 1080.0, 1920.0 |    0,    0, 1080, 1920 | HWC_FRAMEBUFFER_TARGET
10-03 11:30:34.616 D/SurfaceFlinger( 2484):   Display[0] configurations (* current):
10-03 11:30:34.616 D/SurfaceFlinger( 2484):     * 0: 1080x1920, xdpi=422.029999, ydpi=424.069000, refresh=16949152
10-03 11:30:34.616 D/SurfaceFlinger( 2484):   numHwLayers=4, flags=00000000
10-03 11:30:34.616 D/SurfaceFlinger( 2484):     type   |  handle  | hint | flag | tr | blnd |   format    |     source crop (l,t,r,b)      |          frame         | name 
10-03 11:30:34.616 D/SurfaceFlinger( 2484): -----------+----------+------+------+----+------+-------------+--------------------------------+------------------------+------
10-03 11:30:34.616 D/SurfaceFlinger( 2484):        HWC | 7b3c140c80 | 0000 | 0020 | 00 | 0100 | RGBA_8888   |    0.0,    0.0, 1080.0, 1857.0 |    0,   63, 1080, 1920 | SurfaceView - gr.terracom.spotypal.com/gr.terracom.spotypal.com.MainActivity@5dad3df@14#0
10-03 11:30:34.616 D/SurfaceFlinger( 2484):        HWC | 7b3a2595c0 | 0000 | 0000 | 00 | 0105 | RGBA_8888   |    0.0,    0.0, 1080.0,   63.0 |    0,    0, 1080,   63 | gr.terracom.spotypal.com/gr.terracom.spotypal.com.MainActivity#0
10-03 11:30:34.616 D/SurfaceFlinger( 2484):        HWC | 7b3a258840 | 0000 | 0000 | 00 | 0105 | RGBA_8888   |    0.0,    0.0, 1080.0,   63.0 |    0,    0, 1080,   63 | StatusBar#0
10-03 11:30:34.616 D/SurfaceFlinger( 2484):  FB TARGET | 7b3c3f9080 | 0000 | 0000 | 00 | 0105 | RGBA_8888   |    0.0,    0.0, 10
10-03 11:30:34.616 D/SurfaceFlinger( 2484): 80.0, 1920.0 |    0,    0, 1080, 1920 | HWC_FRAMEBUFFER_TARGET
10-03 11:30:34.624 I/zygote64(16128): no shared libraies, dex_files: 1
10-03 11:30:34.651 I/SurfaceFlinger( 2484): id=47049 Removed com.google.android.packageinstaller/com.android.packageinstaller.permission.ui.GrantPermissionsActivity#0 (4/5)
10-03 11:30:34.657 I/SurfaceFlinger( 2484): id=47049 Removed com.google.android.packageinstaller/com.android.packageinstaller.permission.ui.GrantPermissionsActivity#0 (-2/5)
10-03 11:30:34.665 I/Layer   ( 2484): id=47049 onRemoved com.google.android.packageinstaller/com.android.packageinstaller.permission.ui.GrantPermissionsActivity#0 
10-03 11:30:34.668 I/zygote64(16128): find the special shared library, skip check
10-03 11:30:34.742 I/PPS_SemPrivateModeManager(16128): SEC_PRODUCT_FEATURE_COMMON_SUPPORT_PRIVATE_MODE is missing
10-03 11:30:34.749 D/MediaProvider(16128): Attached volume: internal
10-03 11:30:34.750 D/MediaProvider(16128): new PersonDataBaseHelper()
10-03 11:30:34.754 I/ContentSyncInterface(16128): Request sync on/off state
10-03 11:30:34.762 I/[SC]CloudProvider(16058): Call: get_cloud_settings_value
10-03 11:30:34.763 I/PPS_SemPrivateModeManager(16128): SEC_PRODUCT_FEATURE_COMMON_SUPPORT_PRIVATE_MODE is missing
10-03 11:30:34.764 I/[SC]CloudProvider(16058): isValidPackage: com.android.providers.media, get_cloud_settings_value
10-03 11:30:34.767 D/VolumeInfo(16128): VolumeInfo from Parcel: id private ,type 1 ,disk null
10-03 11:30:34.772 D/VolumeInfo(16128): VolumeInfo from Parcel: id public:179,33 ,type 0 ,disk DiskInfo{disk:179,32}:
10-03 11:30:34.772 D/VolumeInfo(16128):     flags=SD size=15931539456 label= volumeCount=1 
10-03 11:30:34.772 D/VolumeInfo(16128):     sysPath=/sys//devices/13560000.dwmmc2/mmc_host/mmc2/mmc2:aaaa/block/mmcblk1 
10-03 11:30:34.772 D/VolumeInfo(16128): 
10-03 11:30:34.772 D/VolumeInfo(16128): VolumeInfo from Parcel: id emulated ,type 2 ,disk null
10-03 11:30:34.783 I/[SC]CMHInterface(16058): isEnabled: true
[... cut ...]
10-03 11:30:36.275 D/Notification( 3046): allPendingIntents
10-03 11:30:36.277 D/Notification.Badge( 3359): updateLauncherIconBadges : 1
10-03 11:30:36.302 W/OpenGLRenderer( 3359): Incorrectly called buildLayer on View: WorkspaceCellLayoutChildren, destroying layer...
10-03 11:30:36.302 W/OpenGLRenderer( 3359): Incorrectly called buildLayer on View: WorkspaceCellLayoutChildren, destroying layer...
10-03 11:30:36.433 E/SignalClusterView( 3713): setSlotFocusVisible : subId 2
10-03 11:30:36.433 E/SignalClusterView( 3713): setSlotFocusVisible : subId 2
```

davidgyoung, 2018-10-03:
> It appears when the app was in the background for a long time. I can not find any other way to reproduce this only when the app is in the background for a long time and the scan is started.

@Kitsopappas does this *always* happen if you leave it in the background for a long time, or only rarely?

Kitsopappas, 2018-10-04:
@davidgyoung I have two mobiles at work that have this behaivour really often. Give me the instructions so I can run the tests and capture the error.

kababu007, 2018-10-04:
We are seeing this issue with 2.12.2 version in devices : 
LM-Q710(FGN)
LM-Q710.FGLG Device
moto Device

Unfortunately these crashes are not replicable. These are recorded in newrelic crash list.

We have upgraded it to 2.15 now and waiting to roll out.

davidgyoung, 2018-10-04:
@kababu007, please try the following:

1. Modify your program to set `beaconManager.setDebug(true)`
2. Connect the device to your computer using a USB cable, and run your from Android Studio
3. With the cable still connected, type the following command on your computer
     adb logcat > logfile.txt
4. The above will write all log lines to the logfile.txt file
5. Leave the phone alone until you experience the crash
6. Once you have the crash, look for the stack trace in logfile.txt
7. Share an excerpt of the log file for the 30 seconds prior to the crash

abolfazlahmadi339, 2018-10-04:
Update my android to 8

kababu007, 2018-10-04:
> @kababu007, please try the following:
>
> 1. Modify your program to set `beaconManager.setDebug(true)`
> 2. Connect the device to your computer using a USB cable, and run your from Android Studio
> 3. With [...]

I have 2 test devices but never had a crash before. Let me see if I can replicate leaving them overnight.

davidgyoung, 2018-10-04:
Yeah, I've been meaning to "undepreacate" it.  The alternative syntax is quite verbose and difficult to explain quickly.

davidgyoung, 2018-10-04:
Apologies, @Kitsopappas, I meant to write that note to you about running the test.  See my instructions above.  (@kababu007 you are welcome to try as well if you can.)

kababu007, 2018-10-05:
Sure thank you. I tried overnight with app kept in doze mode but no crashes reproduced.

Kitsopappas, 2018-10-08:
Hi I finally was able to reproduce the error and get a log this morning. In this link you will find the logfile. Hope this helps!!

https://www.dropbox.com/s/g2kxgscgy2ilafa/logfile.txt?dl=0

davidgyoung, 2018-10-08:
Thanks  @Kitsopappas , that is very helpful.

The log shows that the app was woken up in the background by a beacon appearing.  A new Immediate ScanJob was started to process 8 beacon advertisement packets:

```
10-08 10:10:49.041 D/ScanJob (14686): Processing 8 queued scan results
```

The library then spawns one thread to parse and process each of the 8 packets.  The threads get spawned properly for the first two packets, then it crashes with an `OutOfMemoryError` on the third.  The Java VM starts then shutting down, but before it can, the next 5 packets actually do cause threads to get spawned properly and they get processed.

The library uses a ThreadPoolExecutor to spawn the threads that is configured like this: `mExecutor = Executors.newFixedThreadPool(Runtime.getRuntime().availableProcessors() + 1);`

The size of the pool ( based on the number of processors) probably doesn't matter in this case, since we see it crash on the third thread creation attempt.  This means it could not create even three threads, which is pretty surprising.  This suggests that at this time in the background, the Android system is extremely limited in either available threads or stack memory, and doesn't have enough available to allocate even a small number of new threads.  There is no evidence that the library or the app using it is overusing resources.  This might be caused by a bad state on the device (either in the OS, or in another running app.)

The library currently protects against running out of threads by catching a RejectedExecutionException, as seen in the code excerpt below.  However, apparently this only protects agains exhausting the thread pool -- it does not protect against an OutOfMemory exception caused by the inability to allocate a new thread ad the Java VM level inside the thread pool.

```
    @TargetApi(Build.VERSION_CODES.HONEYCOMB)
    void processScanResult(BluetoothDevice device, int rssi, byte[] scanRecord) {
        NonBeaconLeScanCallback nonBeaconLeScanCallback = mBeaconManager.getNonBeaconLeScanCallback();

        try {
            new ScanHelper.ScanProcessor(nonBeaconLeScanCallback).executeOnExecutor(mExecutor,
                    new ScanHelper.ScanData(device, rssi, scanRecord));
        } catch (RejectedExecutionException e) {
            LogManager.w(TAG, "Ignoring scan result because we cannot keep up.");
        } catch (OutOfMemoryError e) {
            LogManager.w(TAG, "Ignoring scan result because we cannot start thread and keep up.");
        }
    }
```

Simply catching this error (like shown below) would prevent the crash, but not fix the underlying problem that Android cannot allocate a new thread to process the bluetooth scanning results.  This means the beacon detections would be ignored.  But this is probably better than crashing the app.

```
        ...

        try {
            new ScanHelper.ScanProcessor(nonBeaconLeScanCallback).executeOnExecutor(mExecutor,
                    new ScanHelper.ScanData(device, rssi, scanRecord));
        } catch (RejectedExecutionException e) {
            LogManager.w(TAG, "Ignoring scan result because we cannot keep up.");
        } catch (OutOfMemoryError e) {
            LogManager.w(TAG, "Ignoring scan result because we cannot start thread and keep up.");
        }
```

davidgyoung, 2018-10-08:
One other important point:  the log excerpt shows this exception happens shortly after the app is launched into the background by the new intent-based bluetooth scanning api introduced in Android 8:

```
bluetoothAdapter.getBluetoothLeScanner().startScan(filters, settings, pendingIntent);
```

Since this crash seems to be related to this relatively new Android API, this explains why it has only shown up in the past year, as prior to that, it did not exist. 

It's unclear why a crash caused by the inability to create a new thread is coming up as a result of the use of this API.

Kitsopappas, 2018-10-09:
Hey, I downloaded the library, made the changes you proposed and generated a jar file to test it.

```java 
 } catch (OutOfMemoryError e) {
   LogManager.w(TAG, "Ignoring scan result because we cannot start thread and keep up.");
}
```

The error seems to be decreased and does not occur as often as it used to, but still crashes!! I am uploading the new logfile I got today.

As you can see the OutOfMemory Exception is handled but eventualy it crashes again.

https://www.dropbox.com/s/7593nawjerbuh0e/logfilespt.txt?dl=0

Kitsopappas, 2018-10-09:
I have made the following changes to the code to test if I could find the new OutOfMemoryError that is thrown at CycledLeScanner and it did not crash! I will continue testing and let you know.

I added try/catch at ScanJob.java 

## Updated code
```java
@Override
    public boolean onStartJob(final JobParameters jobParameters) {
        try {
            initializeScanHelper();
            if (jobParameters.getJobId() == getImmediateScanJobId(this)) {
                LogManager.i(TAG, "Running immediate scan job: instance is "+this);
            }
            else {
                LogManager.i(TAG, "Running periodic scan job: instance is "+this);
            }

            List<ScanResult> queuedScanResults = ScanJobScheduler.getInstance().dumpBackgroundScanResultQueue();
            LogManager.d(TAG, "Processing %d queued scan resuilts", queuedScanResults.size());
            for (ScanResult result : queuedScanResults) {
                ScanRecord scanRecord = result.getScanRecord();
                if (scanRecord != null) {
                    mScanHelper.processScanResult(result.getDevice(), result.getRssi(), scanRecord.getBytes());
                }
            }
            LogManager.d(TAG, "Done processing queued scan resuilts");

            boolean startedScan;
            if (mInitialized) {
                LogManager.d(TAG, "Scanning already started.  Resetting for current parameters");
                startedScan = restartScanning();
            }
            else {
                startedScan = startScanning();
            }
            mStopHandler.removeCallbacksAndMessages(null);

            if (startedScan) {
                LogManager.i(TAG, "Scan job running for "+mScanState.getScanJobRuntimeMillis()+" millis");
                mStopHandler.postDelayed(new Runnable() {
                    @Override
                    public void run() {
                        LogManager.i(TAG, "Scan job runtime expired: " + ScanJob.this);
                        stopScanning();
                        mScanState.save();
                        ScanJob.this.jobFinished(jobParameters , false);

                        // need to execute this after the current block or Android stops this job prematurely
                        mStopHandler.post(new Runnable() {
                            @Override
                            public void run() {
                                scheduleNextScan();
                            }
                        });

                    }
                }, mScanState.getScanJobRuntimeMillis());
            }
            else {
                LogManager.i(TAG, "Scanning not started so Scan job is complete.");
                ScanJob.this.jobFinished(jobParameters , false);
            }
        }catch (OutOfMemoryError e){
            LogManager.e(TAG, "Handle out of memory error");
        }
        return true;
    }
```

kababu007, 2018-10-09:
Hi team. Any plans of publishing the new beacon library soon with this fix?

davidgyoung, 2018-10-09:
I'll review the logs today.  We can push out a build later this week once I
confirm the fix.

Kitsopappas, 2018-10-10:
I did not have any crashes last night with the latest change! Tested on 6 phones.

davidgyoung, 2018-10-10:
After reviewing the log I see that, yes, it is the thread for the Handler used to do scan cycling in the CycledLeSanner that fails to be created due to an OutOfMemroyError.  

I created a PR to fix this in #750.  @Kitsopappas, this is very similar to what you wrote above -- the main difference is that I catch the exception at the very start of the ScanJob.onStart method, and exit out immediately and return false if the thread cannot be created.  Returning false lets the OS know that the scheduled job is complete.

@Kitsopappas, do you want to test with this before I put it in a release just to be sure you get the same results as your last good test?  If so I can make you a binary build, or obviously you seem comfortable building your own library binaries, too.

kababu007, 2018-10-15:
Eagerly Awaiting your testing @Kitsopappas.

Kitsopappas, 2018-10-16:
After 7 days there are no crashes! Tested on 6 phones. The fix worked!!

kababu007, 2018-10-16:
Excellent.

kababu007, 2018-10-17:
> After reviewing the log I see that, yes, it is the thread for the Handler used to do scan cycling in the CycledLeSanner that fails to be created due to an OutOfMemroyError.
> 
> I created a PR to fix this in #750. @Kitsopappas, this is very similar to what you wrote above -- the main difference is that I catch the exception at the very start of the ScanJob.onStart method, and exit out immediately and return false if the thread cannot be created. Returning false lets the OS know that the scheduled job is complete.
> 
> @Kitsopappas, do you want to test with this before I put it in a release just to be sure you get the same results as your last good test? If so I can make you a binary build, or obviously you seem comfortable building your own library binaries, too. If we can confirm this works, I'll roll this into a 2.15.2 release.
***********************************************************
@davidgyoung   - Is it possible to have the 2.15.2 this week?

davidgyoung, 2018-10-17:
2.15.2 is released now.  See here: https://altbeacon.github.io/android-beacon-library/download.html

closed 2018-10-17 by davidgyoung