AltBeacon/android-beacon-library #590: [Support] Advertising Beacon in background (Android O)

mpuchala, 2017-09-12:
Hey,

I'm using AltBeacon library to advertise beacon in background. Right now I have a `Service` with `BeaconTransmitter` which is working in background. After latest changes in Android O it won't work anymore (Foreground Service is not acceptable in that situation). Is there any way to achieve this? I was looking for any solution but just found information about Beacon detecting. If I missed sth out I would be grateful for hint. Also if issue is duplicated just let me know and feel free to close.

Thanks!

android-Israel, 2018-01-06:
need help too

davidgyoung, 2018-01-06:
Unfortunately, Android 8's prohibition on long-running background services means that you can only transmit when a the app is active.  So the app has to be in the foreground, or you get about 10 min of transmission time every 15-25 minutes when a background job runs.

One alternative is to use a foreground service, which can be long-running.  This would still allow transmission in the background.  The only real disadvantage is that the user will see an icon at the top if the screen. 

The limits on transmission serve no power saving purpose as BLE transmissions use negligible power.  However, until Android adds APIs to transmit when your app is not running, the above options are the best there are.

michaellee8, 2020-04-06:
@davidgyoung Can you kindly provide some sample code if I want to do background broadcasting using a Foreground Service? I can only see an example that demonstrates scanning but not advertisement.

davidgyoung, 2020-04-06:
You can enable a foreground service in the same way as described for scanning [here](https://altbeacon.github.io/android-beacon-library/foreground-service.html)

For the foreground service to activate, you will need to start monitoring beacons, even if you don't need to otherwise. If you have no other use for the monitoring callbacks,  you can do so just for a dummy region, and set the scan period and between scan period to minimal intervals to save battery:

```
beaconManager.setForegroundBetweenScanPeriod(99999999);
beaconManager.setBackgroundBetweenScanPeriod(99999999);
Region dummyRegion = new Region("com.example.backgroundRegion",
                Identifier.parse("01020304-0506-0708-090A-0B0C0D0E0F00"), 1, 2);
new RegionBootstrap(this, dummyRegion); // the enclosing class must implement BootstrapNotifier
```

michaellee8, 2020-04-06:
@davidgyoung Do you mean that I by activtating "foreground service" background scanning i can enable "foreground service" advertisement as well? Is it possible for me to do it without requesting for the locations permission since transmitting beacon singal only require bluetooth permissions?

davidgyoung, 2020-04-06:
Yes, you just need to create a foreground service to keep your app alive.
While you can transmit in the background without a foreground service, the
OS will kill your app after ~10min in the background and stop transmission
if you do not have one.

The option I show will activate the library's built in foreground service
designed for scanning purposes. You do not need to request location
permissions from the user or even declare them in the manifest. Without
them no beacons will be detected, but that matters little if you have no
need for detection.

You can also just set up your own dummy foreground service manually.  I am
just showing an easy way to leverage the library's built in one.