AltBeacon/android-beacon-library #592: Android 7 Background scan

MrD0llaro, 2017-09-13:
### Expected behavior
I configure an app correctly, on the method onCreate (Base Application) I configure 2 regions for BootstrapNotifier. I run the app one time for configure it. I close the App. 
I hope the app find a Region/beacon quikly on background .

 
### Actual behavior

on Android device less 5.0 is necessary to wait 5 min. (I read the documentation and is normal behaviour)
on 5 <= devices < 7, ti s action is  fast (10s).   (this use a System scan and are realy good :))
on devices with 7.0 or 7.1.2 is necessary to wait 5 minute. (I was hoping to have the same android performance 5 )

### Steps to reproduce this behavior

try the background scan with Androd 7 devices.
### Mobile device model and OS version
Samsung 
nexus 5x
hawey 6p

### Android Beacon Library version
2.12.2
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

MrD0llaro, 2017-09-13:
the output on debug mode:

E/Zygote: v2
I/libpersona: KNOX_SDCARD checking this for 10008
I/libpersona: KNOX_SDCARD not a persona
E/Zygote: accessInfo : 0
W/SELinux: SELinux selinux_android_compute_policy_index : Policy Index[2],  Con:u:r:zygote:s0 RAM:SEPF_SECMOBILE_7.0_0005, [-1 -1 -1 -1 0 1]
I/SELinux: SELinux: seapp_context_lookup: seinfo=untrusted, level=s0:c512,c768, pkgname=it.test.mobile.test.debug 
I/art: Late-enabling -Xcheck:jni
D/TimaKeyStoreProvider: TimaKeyStore is not enabled: cannot add TimaSignature Service and generateKeyPair Service
D/FirebaseApp: com.google.firebase.auth.FirebaseAuth is not linked. Skipping initialization.
D/FirebaseApp: com.google.firebase.crash.FirebaseCrash is not linked. Skipping initialization.
I/FA: App measurement is starting up, version: 9256
I/FA: To enable debug logging run: adb shell setprop log.tag.FA VERBOSE
D/FA: Debug logging enabled
D/FA: AppMeasurement singleton hash: 14820424
V/FA: Collection enabled
V/FA: App package, google app id: it.test.mobile.test.debug, 1:580950854458:android:077d5bb80f9c876f
V/FA: Registered activity lifecycle callback
I/FirebaseInitProvider: FirebaseApp initialization successful
V/FA: Using measurement service
V/FA: Connecting to remote service
I/BeaconManager: BeaconManager started up on pid 19813 named 'it.test.mobile.test.debug' for application package 'it.test.mobile.test.debug'.  isMainProcess=true
D/BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
D/BeaconParser: Parsing beacon layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
D/BeaconManager: This consumer is not bound.  Binding now: org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer@2bdf662
D/BeaconManager: Binding to service
D/BeaconManager: consumer count is now: 1
D/AppStarter: Waiting for BeaconService connection
D/BluetoothCrashResolver: constructed
W/BluetoothCrashResolver: Can't read macs from BluetoothCrashResolverState.txt
D/BluetoothCrashResolver: Read 0 Bluetooth addresses
D/BluetoothCrashResolver: started listening for BluetoothAdapter events
I/CycledLeScanner: This is Android 5.0.  We are using new scanning APIs
I/BeaconService: beaconService version 2.12.2 is starting up on the main process
D/ModelSpecificDistanceCalculator: Finding best distance calculator for 7.0, NRD90M, SM-G920F, samsung
D/AndroidModel: Score is 0 for XT1115;Moto X Pro;LXG22.67-7.1;5.0.2 compared to samsung;SM-G920F;NRD90M;7.0
D/AndroidModel: Score is 0 for LGE;Nexus 5;LPV79;4.4.2 compared to samsung;SM-G920F;NRD90M;7.0
D/AndroidModel: Score is 0 for LGE;Nexus 4;KOT49H;4.4.2 compared to samsung;SM-G920F;NRD90M;7.0
W/ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
D/ModelSpecificDistanceCalculator: Finding best distance calculator for 7.0, NRD90M, SM-G920F, samsung
D/AndroidModel: Score is 0 for XT1115;Moto X Pro;LXG22.67-7.1;5.0.2 compared to samsung;SM-G920F;NRD90M;7.0
D/AndroidModel: Score is 0 for LGE;Nexus 5;LPV79;4.4.2 compared to samsung;SM-G920F;NRD90M;7.0
D/AndroidModel: Score is 0 for LGE;Nexus 4;KOT49H;4.4.2 compared to samsung;SM-G920F;NRD90M;7.0
W/ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
D/BeaconService: No org.altbeacon.beacon.SimulatedScanData class exists.
I/BeaconService: starting with null intent
V/MultitaskServiceListener: onStarCommand(): 
I/BeaconService: starting with intent Intent { cmp=it.test.mobile.test.debug/org.altbeacon.beacon.service.BeaconService launchParam=MultiScreenLaunchParams { mDisplayId=0 mFlags=0 } }
I/BeaconService: binding
D/BeaconManager: we have a connection to the service now
D/BeaconManager: Not synchronizing settings to service, as it has not started up yet
D/AppStarter: Activating background region monitoring
D/AppStarter: Background region monitoring activated for region id1: 640ac06e-7b2c-4ea0-beb7-eecd836fe80b id2: null id3: null
D/BeaconManager: callback packageName: it.test.mobile.test.debug
D/MonitoringStatus: Restored region monitoring state for 2 regions.
D/MonitoringStatus: Region  id1: 640ac06e-7b2c-4ea0-beb7-eecd836fe80b id2: null id3: null uniqueId: test.primo state: org.altbeacon.beacon.service.RegionMonitoringState@292773f
D/MonitoringStatus: Region  id1: 0c843e85-2b5c-4cee-9af0-6e60a7fad652 id2: null id3: null uniqueId: test.secondo state: org.altbeacon.beacon.service.RegionMonitoringState@26c280c
D/MonitoringStatus: Done restoring monitoring status
D/BeaconApplication: didDetermineStateForRegion: state0
D/BeaconManager: updating background flag to true
D/BeaconManager: updating scan period to 10000, 300000
D/AppStarter: Background region monitoring activated for region id1: 0c843e85-2b5c-4cee-9af0-6e60a7fad652 id2: null id3: null
D/BeaconManager: callback packageName: it.test.mobile.test.debug
D/BeaconApplication: didDetermineStateForRegion: state0
I/BeaconService: start monitoring received
D/FA: Connected to remote service
D/BeaconService: startMonitoring called
D/MonitoringStatus: saveMonitoringStatusIfOn()
V/FA: Processing queued up service tasks: 1
[... 566 lines cut ...]
D/RangedBeacon: calculated new runningAverageRssi: -43.5
D/Callback: attempting callback via global broadcast intent: ComponentInfo{it.test.mobile.test.debug/org.altbeacon.beacon.BeaconIntentProcessor}
D/CurveFittedDistanceCalculator: calculating distance based on mRssi of -43.5 and txPower of -50
D/CurveFittedDistanceCalculator: avg mRssi: -43.5 distance: 0.24842341419143568
[... cut ...]
D/CycledLeScanner: Set scan periods called with 10000, 300000 Background mode must have changed.
D/CycledLeScanner: We are in the background.  Setting wakeup alarm
D/CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{da0ebc3: android.os.BinderProxy@12bfc40}
D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 298844 milliseconds
D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 297840 milliseconds
D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 296838 milliseconds

davidgyoung, 2017-09-13:
Understand that of you kill the app from the task switcher on Android 4-7, it will take 0-5 minutes for the library to resume scanning for beacons, as a 5 minute recurring alarm is used to restart scanning.  After this period, detections will once again be fast (a few secs) on Android 5-8.

This normally is not a problem as fast detections immediately after killing the app are not a common use case.

MrD0llaro, 2017-09-13:
Thank you for the response.
I have a boubt:  is    <meta-data android:name="longScanForcingEnabled" android:value="true" />  necessary for android 7?

MrD0llaro, 2017-09-13:
With nexus 5x , the background scan never start. 
have to look the log:

I/art: Late-enabling -Xcheck:jni
D/FirebaseApp: com.google.firebase.auth.FirebaseAuth is not linked. Skipping initialization.
D/FirebaseApp: com.google.firebase.crash.FirebaseCrash is not linked. Skipping initialization.
I/FA: App measurement is starting up, version: 9256
I/FA: To enable debug logging run: adb shell setprop log.tag.FA VERBOSE
D/FA: Debug logging enabled
D/FA: AppMeasurement singleton hash: 215671731
V/FA: Collection enabled
V/FA: App package, google app id: it.test.mobile.test.debug, 1:580950854458:android:077d5bb80f9c876f
V/FA: Registered activity lifecycle callback
I/FirebaseInitProvider: FirebaseApp initialization successful
V/FA: Using measurement service
V/FA: Connecting to remote service
I/BeaconManager: BeaconManager started up on pid 28867 named 'it.test.mobile.test.debug' for application package 'it.test.mobile.test.debug'.  isMainProcess=true
D/BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
D/BeaconParser: Parsing beacon layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
D/BeaconManager: This consumer is not bound.  Binding now: org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer@4787a3e
D/BeaconManager: Binding to service
D/BeaconManager: consumer count is now: 1
D/AppStarter: Waiting for BeaconService connection
D/BluetoothCrashResolver: constructed
W/BluetoothCrashResolver: Can't read macs from BluetoothCrashResolverState.txt
D/BluetoothCrashResolver: Read 0 Bluetooth addresses
D/BluetoothCrashResolver: started listening for BluetoothAdapter events
I/CycledLeScanner: This is Android 5.0.  We are using new scanning APIs
I/BeaconService: beaconService version 2.12.2 is starting up on the main process
D/ModelSpecificDistanceCalculator: Finding best distance calculator for 7.1.2, N2G48C, Nexus 5X, LGE
D/AndroidModel: Score is 1 for LGE;Nexus 4;KOT49H;4.4.2 compared to LGE;Nexus 5X;N2G48C;7.1.2
D/AndroidModel: Score is 1 for LGE;Nexus 4;KOT49H;4.4.2 compared to LGE;Nexus 5X;N2G48C;7.1.2
D/AndroidModel: Score is 0 for XT1115;Moto X Pro;LXG22.67-7.1;5.0.2 compared to LGE;Nexus 5X;N2G48C;7.1.2
D/AndroidModel: Score is 1 for LGE;Nexus 5;LPV79;4.4.2 compared to LGE;Nexus 5X;N2G48C;7.1.2
D/ModelSpecificDistanceCalculator: found a match with score 1
D/ModelSpecificDistanceCalculator: Finding best distance calculator for 4.4.2, KOT49H, Nexus 4, LGE
D/ModelSpecificDistanceCalculator: Finding best distance calculator for 7.1.2, N2G48C, Nexus 5X, LGE
D/AndroidModel: Score is 1 for LGE;Nexus 4;KOT49H;4.4.2 compared to LGE;Nexus 5X;N2G48C;7.1.2
D/AndroidModel: Score is 1 for LGE;Nexus 4;KOT49H;4.4.2 compared to LGE;Nexus 5X;N2G48C;7.1.2
D/AndroidModel: Score is 0 for XT1115;Moto X Pro;LXG22.67-7.1;5.0.2 compared to LGE;Nexus 5X;N2G48C;7.1.2
D/AndroidModel: Score is 1 for LGE;Nexus 5;LPV79;4.4.2 compared to LGE;Nexus 5X;N2G48C;7.1.2
D/ModelSpecificDistanceCalculator: found a match with score 1
D/ModelSpecificDistanceCalculator: Finding best distance calculator for 4.4.2, KOT49H, Nexus 4, LGE
D/BeaconService: No org.altbeacon.beacon.SimulatedScanData class exists.
I/BeaconService: starting with null intent
V/MultitaskServiceListener: onStarCommand(): 
I/BeaconService: starting with intent Intent { cmp=it.test.mobile.test.debug/org.altbeacon.beacon.service.BeaconService }
I/BeaconService: binding
D/BeaconManager: we have a connection to the service now
D/BeaconManager: Not synchronizing settings to service, as it has not started up yet
D/AppStarter: Activating background region monitoring
D/AppStarter: Background region monitoring activated for region id1: 640ac06e-7b2c-4ea0-beb7-eecd836fe80b id2: null id3: null
D/BeaconManager: callback packageName: it.test.mobile.test.debug
D/MonitoringStatus: Restored region monitoring state for 2 regions.
D/MonitoringStatus: Region  id1: 640ac06e-7b2c-4ea0-beb7-eecd836fe80b id2: null id3: null uniqueId: test.primo state: org.altbeacon.beacon.service.RegionMonitoringState@ef2421c
D/MonitoringStatus: Region  id1: 0c843e85-2b5c-4cee-9af0-6e60a7fad652 id2: null id3: null uniqueId: test.secondo state: org.altbeacon.beacon.service.RegionMonitoringState@af2de25
D/MonitoringStatus: Done restoring monitoring status
D/BeaconApplication: didDetermineStateForRegion: state0
D/BeaconManager: updating background flag to true
D/BeaconManager: updating scan period to 10000, 300000
D/AppStarter: Background region monitoring activated for region id1: 0c843e85-2b5c-4cee-9af0-6e60a7fad652 id2: null id3: null
D/BeaconManager: callback packageName: it.test.mobile.test.debug
D/BeaconApplication: didDetermineStateForRegion: state0
I/BeaconService: start monitoring received
D/BeaconService: startMonitoring called
D/MonitoringStatus: saveMonitoringStatusIfOn()
D/FA: Connected to remote service
V/FA: Processing queued up service tasks: 1
D/BeaconService: Currently monitoring 2 regions.
D/CycledLeScanner: start called
[... 464 lines cut ...]
D/CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{ec54b49: android.os.BinderProxy@a677c4e}
D/CycledLeScanner: Waiting to stop scan cycle for another 1665 milliseconds
D/CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{ec54b49: android.os.BinderProxy@a677c4e}
D/CycledLeScanner: Waiting to stop scan cycle for another 625 milliseconds
D/CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{ec54b49: android.os.BinderProxy@a677c4e}
[... cut ...]
D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 298853 milliseconds
D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 297808 milliseconds
D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 296767 milliseconds
D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 295726 milliseconds
D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 294684 milliseconds
D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 293643 milliseconds

kensykora, 2017-11-06:
I can confirm that I'm also having similar issues to this -- The app I wrote scans just fine on Android Oreo with 2.12.3 (Nexus 6P), but on Lollipop devices (Samsung Galaxy S6, Samsung Galaxy Tab 2) with the same library it fails to find any of my iBeacons, whether in foreground or in background.

I suspect this has something to do with Samsung's ScanFilter implementation -- In past projects I had to throw out ScanFilters because they didn't behave properly on some devices.

davidgyoung, 2017-11-06:
@kensykora, I like the theory but this library only uses a ScanFilter if it is in the background m the backgroundBetweenScanPeriod.  At all other times it has no scan filter active.

I know this is not a universal problem with all Samsung devices as I successfully detect with them.  You might try this app https://play.google.com/store/apps/details?id=com.radiusnetworks.locate based on the library to see if it detects

kensykora, 2017-11-07:
@davidgyoung I'm seeing erratic behavior that I'm going to try and isolate and submit a separate ticket for this -- Even on my Nexus 6P i'm now seeing the beacon show up as a callback to `BeaconManager.NonBeaconLeScanCallback` and implemented `INonBeaconLeScanCallback` interface. Force quitting the app and turning bluetooth off/on on the device seems to resolve the issue, I'm going to try and see if a similar work around works on my Samsung devices for non-Oreo.