AltBeacon/android-beacon-library #599: ¿Is alt-beacon available for Android Version 8.0.0?

JosueTC94, 2017-09-21:
Hi! I am developing an Android application that detects Bluetooth beacons. This app uses alt-beacon and it works well in all tests made taking into account multiple devices characterized by Android versions <= 6.0.0.  However, now I am testing the app in a Pixel Go 8.0.0 (without making any code changes) and beacons aren't detected never. I think that Doze mode could lock scanning process of beacons, but now I am thinking about if AltBeacon can be used with Android version > 6.0.0 without any problem.

I hope that I have explained my problem well.
Regards,
Josué.

davidgyoung, 2017-09-21:
I have tested the latest 2.12.x version of the library with the reference
app successfully on Android 4.3-8.0.

Doze will stop detections on motionless devices in the background after a
period of time with Android 6+

If your experience differs from this, more details would be needed
including steps to reproduce the problem with the reference app in order to
assist.

JosueTC94, 2017-09-22:
Thanks! In this case, have you found any solution to the fact that Doze stops detections on motionless devices in the background with alt-beacon? I have been testing some solutions based on AlarmManager and Wakelocks but I haven't got good results yet.  My application needs to do scans frequently (every 5 seconds) in the way to detect all beacons nearly. I have thought to manage some type of alarm in the way to restart the alt-beacon service,  but AlarmManager is intended to set alarms every minute, hour or weeks instead of every 5 seconds. In fact, some newly developed methods of this library (setAndAllowWhileIdle and setExactAndAllowWhileIdle) can't activate alarms before 9 minutes.

davidgyoung, 2017-09-22:
On Android 6 and 7 I do not believe there is any way to keep a background
service processing scanned beacon detections once Doze starts due to
motionlessness.  Alarms get delayed until after doze exit.

On Android 8, however, I believe this works again, as beacon detections get
delivered by intent, whick should exit doze.  (although I have not
verified.)

closed 2017-09-24 by davidgyoung