AltBeacon/android-beacon-library #601: Disable log on release build.

TonyTangAndroid, 2017-09-22:
### Expected behavior
There should be a way to disable all beacon log when the app is in release mode.
### Actual behavior
Error, warning and Info level log still being printed out in release mode.
I could see a lot of log in the console like this:

> BeaconService: start monitoring received
> MonitoringStatus: Too many regions being monitored.  Will not persist region state

### Steps to reproduce this behavior
Could be easily stably reproduced. 
### Mobile device model and OS version
Google Pixel Android 8
### Android Beacon Library version
2.12.3-beta2

hashbrown, 2017-09-23:
You can control this in your own app.
`LogManager.setLogger(Loggers.empty());`

TonyTangAndroid, 2017-09-23:
Thank you very much for your hint. I had given it a try to read the method in this class but somehow I did not find it.  I will make the update.

closed 2017-09-23 by TonyTangAndroid