AltBeacon/android-beacon-library #603: didRangeBeaconsInRegion is invoked 10 minutes after didEnterRegion on Android O

LudySu, 2017-09-27:
### Expected behavior
`didRangeBeaconsInRegion` shoud be invoked a few seconds after `didEnterRegion`

### Actual behavior
When our app starts for the first time after installation, `didRangeBeaconsInRegion` is invoked 10 minutes after `didEnterRegion` on Android O; but afterward it takes only about 20 seconds which is satisfactory to us. But on Android 7 it only takes 1 second or so.

### Steps to reproduce this behavior
- Install our own app
- Turns on a beacon, wait for notification

Here I attached the logs for the first time detection, an `didEnterRegion` was detected
> 09-27 13:45:52.078   404   404 D BeaconScannerService: Entering beacon region "id1: f0018b9b-xxxxxxxxxxxxxxxxxxxxxxxxxxx id2: null id3: null"

Then 10 minutes later the `didRangeBeaconsInRegion` callback
> 09-27 13:56:02.269   404   404 V BeaconScannerService: ========= ranging id1: f0018b9b-xxxxxxxxxxxxxxxxxxxxxxxxxxx id2: null id3: null

[beacon_ranging_logs.txt](https://github.com/AltBeacon/android-beacon-library/files/1335668/beacon_ranging_logs.txt)

### Mobile device model and OS version
Nexus 5X on Android 8.0

### Android Beacon Library version
2.12.2

LudySu, 2017-10-03:
After 2 days of tracing the code, I found the critical point. 

When the first time the app runs in the background and a `didEnterRegion` is invoked by AltBeacon, the value of `insideAnyRegion` inside `ScanJob.startPassiveScanIfNeeded()` method is `true`. When later the app detects another entry, `insideAnyRegion` becomes `false`, which makes the app receive `didRangeBeaconsInRegion` quickly.

hashbrown, 2017-10-05:
Here's my guess as to what's occurring.  You should also be paying attention to ```public void didExitRegion(Region region)``` to verify.

A scheduled job kicks off a scan, detects the beacon and calls back to any monitor notifiers (like the RegionBootstrap), and you are notified that the region is entered and you now register a RangeNotifier, but now the current scan is done.  
Its common for the scheduled jobs to be run every 10 minutes, so 10 minutes later its scanning again.  At this point the exit period has expired for the region.  By default, if a beacon isn't detected for 10 seconds in the region, its reported as exited.  This time is configurable.
So anyway, the next scan occurs, finds your beacon, this time you already have a RangeNotifer registered and so you get the callback. Since the region isn't considered entered that's also going to start an Immediate scan, which should run for about 5 minutes before it quits.  So by default you likely get 5 minutes of RangeNotifer callbacks.

davidgyoung, 2017-10-05:
Thanks for the analysis, Brad.  If this is the problem, the simple solution
is to start ranging earlier and leave it on.

LudySu, 2017-10-12:
Sorry for late response, we spent some more time into investigating this issue but was distracted by some other tasks.

@hashbrown Thanks for you analysis, we confirmed that there's no unexpected `didExitRegion` in between scan cycles. Apart from that your analysis matches my observation, it looks like when an entry is detected the current scan cycle ends, then ranging can only be started when next scan cycle kicks in 10 minutes later. We still can not figure out why later times the ranging only takes 20 seconds instead of 10 minutes.

@davidgyoung Thanks for your reply. It does resolve the problem if we `startRangingBeaconsInRegion` at the same time as we `startMonitoringBeaconsInRegion`. So my question is will it drain more battery than that we only start ranging when a entry is detected? Our app scans for beacons in the background all the time, and according to our early test result it uses a bit more battery.

Thanks

davidgyoung, 2017-10-13:
@LudySu, I don't think that it will make significant difference in battery usage between starting ranging/monitoring simultanously vs. starting ranging only after a monitoring entry.  The only difference in processing between the two is that empty ranging callbacks would be sent once at the end of each scan cycle.  In the foreground, this will be once a second (so pretty often), but battery usage will be insignificant vs. screen.  In the background, cycles are relatively rare, so this just doesn't happen so often.  

I have not run tests to measure the battery impact of this, but based on my understanding of how it works, I really think it is negligible.

LudySu, 2017-11-05:
Sorry for the delay. We conducted more than a week of testing and confirmed that the battery consumption between start ranging at the beginning and only after an entry is detected are indeed identical.

So with this change it solved this issue, I'm closing this ticket.

Thanks @davidgyoung for your great library and always keeping up to date with latest OS releases. Thanks @hashbrown for your great insight of this great library!

closed 2017-11-05 by LudySu

AishwaryaYML, 2023-02-21:
@davidgyoung For the second time onwards not receiving notification. 3/5 times not receiving notification in Android 12/13 . Any idea what could be going wrong here