AltBeacon/android-beacon-library #604: App Standby Power Saving Mode

jmvines, 2017-10-04:
Hi @davidgyoung,

is there any way for the android-beacon-library to not consume battery in App Standby Power Saving Mode?.

Thanks for your answer.

Regads.

davidgyoung, 2017-10-04:
This is pretty much automatic.  If an app is not used for a week or so, Android 6+ puts it into App Standby mode, and it cannot perform jobs or  network activity unless the phone is plugged in.  This restriction is released next time the user launches the app or it sends a notification to the user.

jmvines, 2017-10-04:
Thanks for your answer, but in standby apps like you say on your blog "App Standby does not specifically affect beacon detection with the Android Beacon Library, because bluetooth scanning service continues to run normally, and the AlarmManager and Handler-based timers continue to work". I need that when my app is in standby mode stop these scanning services in order to ensure that it does not consume battery in standby

Any ideas?.

davidgyoung, 2017-10-04:
I do not know of any way to detect app standby.  If you could, it would be possible to simply stop scanning with the library using it's APIs.  

By the way, that statement you mention refers to Android 6,.  I have not tested on Android 7 and 8 but I suspect the answer may be different, especially on 8.

jmvines, 2017-10-05:
Understood, thanks for your answer and for your effort on your amazing library.