AltBeacon/android-beacon-library #605: Supporting both Scheduled Jobs and the Beacon Service.  Supporting Foreground Services.

hashbrown, 2017-10-05:
### Expected behavior
In my application, I'd like to use the current Scheduled Jobs to run a passive scan while the region is not entered (via RegionBootstrap).  But once the region is entered I run a service that is effectively a RangeNotifier, and constantly monitors beacons in a region.  In Android 7 this would bind to the BeaconService and run in foreground mode, because we need aggressive monitoring while the region is entered,
In Android 8 this doesn't work.  Even though the region is entered, going in to foreground mode doesn't get you any more aggressive scanning, its still just Scheduled Jobs, so you might get a few minutes of the scan, then you're done for 10+ minutes.
So what I want to do is setEnableScheduledJobs(false) and then bind to the actual BeaconService to get the old behavior. Then once the region exits, I would go back to just using the passively job provided by the bootstrap..
BTW - I am running my local RangingService as a foreground service now, so it will not get killed.

### Actual behavior
The issue I am having with the above approach is that once the region is exited, I am not able to set the EnableScheduledJobs flag back to true.  There are some checks there that do not allow you to set the flag if you are already bound to the service.  My RangingService is no longer bound to the BeaconService but the problem is that the BeaconManager is still tracking the BeaconConsumer from the initial RegionBootstrap, and it believes that BeaconConsumer is still bound.

I have been trying to create a PR that will support this, but I keep running into a fundamental design problem that I'm hesitant to change. When any BeaconConsumer binds to the BeaconService, then all BeaconConsumers are treated as bound.  So if only one consumer was ever bound, and that unbinds, any other services are always assumed bound as well until all consumers are acutally removed.
I've tried to do slight modifications to get the behavior I'm looking for but its seems this will require some larger design changes.  Effectively I think we need to track the BeaconConsumers that are using ScheduledJobs versus those that are bound to the service?

Looking for your opinion on this use case.  I want to support cases where we can truly toggle between Jobs and the BeaconService, which will be useful to apps that have decided to run as a foreground service to work around Android's terrible new restrictions to Background services.

Thanks!

### Mobile device model and OS version
Nexus 5X, Android 8

### Android Beacon Library version
2.12.3-beta2

hashbrown, 2017-10-05:
BTW - I was trying to support this by fixing https://github.com/AltBeacon/android-beacon-library/pull/600, but it didn't work, which is why I haven't put up an alternate PR for that yet.

davidgyoung, 2017-10-05:
Not sure I understand the issue.

The way this is designed to work in the foreground is that once a ScanJob
completes, the next one will be scheduled to start after the
foregroundBetweenScanPeriod.  With default settings, the
foregroundBetweenScanPeriod is 0 so the next ScanJob will start immediately
after the last one ends.

This works because the minimum job interval is not enforced by Android when
the app is in the foreground.

This is how the reference app ranges constantly in the foreground.  Do you
see different behavior?

strooooke, 2017-10-05:
I think what @hashbrown  might be missing (I've been there) is that the library does not automatically detect that your app is actually in the foreground (due to the foreground service); you have to tell it that `BeaconManager.setBackgroundMode`). A `BackgroundPowerSaver` instance does it for activities; I rolled up my own replacement that also takes my ForegroundService into account - works fine.

hashbrown, 2017-10-05:
Thanks @strooooke, I am toggling that state.

@davidgyoung I do see that behavior in the code, I'll go back to do more testing.  It may be impacted  by Doze.  However what started me down the path was that even though my app was actually resumed and in the foreground, it still was not running the jobs.

hashbrown, 2017-10-06:
@davidgyoung I'm not seeing the behavior you're describing.
Once a beacon enters the region, or my app enters the foreground, I get an ImmediateScan executed.  This runs for ```mScanState.getScanJobRuntimeMillis()```, (5 min by default) and then completes (```Scan job runtime expired```).  There is nothing that reschedules another Immediate Scan.  It does schedule a passive scan if needed. However if I already have a beacon in the region this is is skipped.  So I find myself in a situation where I have my app open with one beacon detected, and any subsequent beacons are not detected. If my app is in the foreground I would expect to be aggressive about detection of all beacons.
Either way, after the initial 5 minute scan, I have to wait another ~10 minutes for the period scan to come around again, regardless if my app is in the foreground.

It seems like the behavior I'm really looking for, is if the app has set the library to run in foreground mode, then immediate scans should always just reschedule themselves.  Its fine if it uses the foregroundBetweenScanPeriod value, but I don't see where that's happening in the code, and I also don't see that behavior.
I'm currently testing against 2.12.

davidgyoung, 2017-10-06:
The way the design is implemented is in the `ScanJobScheduler#schedule` method.  A periodic job is always set up even if an immediate one is active.  The period is set to every 5 minutes (the maximum job run time).  So in theory a new job will start right after the old one ends.  In the **foreground**, the OS should allow running a job every 5 min, and I see this with the reference app on Android 8 yielding continuous scanning (with a between job dropout of a few 10s of millis).

The code that commands that is here:

https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/java/org/altbeacon/beacon/service/ScanJobScheduler.java#L174

Apparently this is not working for you, and I'd like to understand why.   I'd like to try and find a solution within the existing design before looking for other designs.

Can you see if you have any system level debug messaging when the periodic job is scheduled in the foreground that indicates it will not be run every 5 min as requested?

hashbrown, 2017-10-07:
Hi @davidgyoung, to try and reduce the variables, I actually reproduced this same behavior using the reference app.
The app is targeting/compiling against sdk 25, and is using 2.12.2 of the library.
I'm running this on a Nexus 5X running 8.0.0.

What I am seeing is that the period job runs for 5 minutes, but does not seem to schedule itself until at least 5 minutes after the previous job finished.
I believe what you intended is a periodic job that has an interval of 5 minutes, but also runs for 5 minutes, so that you get somewhat a continuous scan in the foreground?

Here's a partial log example:
```
10-06 19:30:32.530  7437  7437 D ScanJobScheduler: Applying scan job settings with background mode false
10-06 19:30:32.530  7437  7437 D ScanState: ScanState says background mode for ScanJob is false
10-06 19:30:32.530  7437  7437 D ScanJobScheduler: Scheduling immediate ScanJob to run in 50 millis
10-06 19:30:32.532  7437  7437 D ScanJobScheduler: Scheduling ScanJob to run every 300000 millis
10-06 19:30:32.627  7437  7437 I ScanJob : Running immdiate scan job: instance is org.altbeacon.beacon.service.ScanJob@adaabd6
10-06 19:30:32.627  7437  7437 I ScanJob : scanJob version 2.12.2 is starting up on the main process
10-06 19:30:32.666  7437  7437 D ScanState: ScanState says background mode for ScanJob is false
10-06 19:30:32.666  7437  7437 I ScanJob : Scan job running for 300000 millis
10-06 19:30:32.666  7437  7437 D ScanState: ScanState says background mode for ScanJob is false

#### SCAN FOR 5 MINUTES

10-06 19:35:32.666  7437  7437 I ScanJob : Scan job runtime expired
10-06 19:35:32.681  7437  7437 D ScanJob : Scanning stopped
10-06 19:35:32.741  7437  7437 D ScanState: Temp file is /data/user/0/org.altbeacon.beaconreference/files/android-beacon-library-scan-state-temp
10-06 19:35:32.741  7437  7437 D ScanState: Perm file is /data/user/0/org.altbeacon.beaconreference/files/android-beacon-library-scan-state
10-06 19:35:32.747  7437  7437 D ScanJob : Checking to see if we need to start a passive scan
10-06 19:35:32.747  7437  7437 I ScanJob : We are inside a beacon region.  We will not scan between cycles.

#### NEXT SCAN 5 MINUTES LATER

10-06 19:40:32.556  7437  7437 I ScanJob : Running periodic scan job: instance is org.altbeacon.beacon.service.ScanJob@6c9bdb8
10-06 19:40:32.557  7437  7437 I ScanJob : scanJob version 2.12.2 is starting up on the main process
10-06 19:40:32.685  7437  7437 D ScanState: ScanState says background mode for ScanJob is false
10-06 19:40:32.685  7437  7437 I ScanJob : Scan job running for 300000 millis
10-06 19:40:32.685  7437  7437 D ScanState: ScanState says background mode for ScanJob is false

#### SCAN FOR 5 MINUTES

10-06 19:45:32.685  7437  7437 I ScanJob : Scan job runtime expired
10-06 19:45:32.698  7437  7437 D ScanJob : Scanning stopped
10-06 19:45:32.735  7437  7437 D ScanState: Temp file is /data/user/0/org.altbeacon.beaconreference/files/android-beacon-library-scan-state-temp
10-06 19:45:32.735  7437  7437 D ScanState: Perm file is /data/user/0/org.altbeacon.beaconreference/files/android-beacon-library-scan-state
10-06 19:45:32.738  7437  7437 D ScanJob : Checking to see if we need to start a passive scan
10-06 19:45:32.738  7437  7437 I ScanJob : We are inside a beacon region.  We will not scan between cycles.

#### NEXT SCAN >10 MINUTES LATER

10-06 19:56:12.595  7437  7437 I ScanJob : Running periodic scan job: instance is org.altbeacon.beacon.service.ScanJob@b0c2d7f
10-06 19:56:12.597  7437  7437 D ScanJob : Processing 0 queued scan resuilts
10-06 19:56:12.597  7437  7437 D ScanJob : Done processing queued scan resuilts
10-06 19:56:12.597  7437  7437 I ScanJob : scanJob version 2.12.2 is starting up on the main process
10-06 19:56:12.678  7437  7437 D ScanState: Scan state restore regions: monitored=1 ranged=1
10-06 19:56:12.690  7437  7437 D ScanState: ScanState says background mode for ScanJob is false
10-06 19:56:12.690  7437  7437 I ScanJob : Scan job running for 300000 millis
10-06 19:56:12.690  7437  7437 D ScanState: ScanState says background mode for ScanJob is false

#### SCAN FOR 5 MINUTES

10-06 20:01:12.691  7437  7437 I ScanJob : Scan job runtime expired
10-06 20:01:12.708  7437  7437 D ScanJob : Scanning stopped
```

Is there any specific tests you would like me to run or logs you want captured?

davidgyoung, 2017-10-10:
Interesting.  In my tests I did see consistent foreground scan job scheduling every 5 minutes.  But it may be that relying on this consistency is not possible.  

An simple alternative would be to check to see if the app is in the foreground when the scan job ends, and if so, immediately reschedule a scan job for now.  This should be possible with just one or two lines of code.

hashbrown, 2017-10-13:
@davidgyoung Created a PR based on your comments.  https://github.com/AltBeacon/android-beacon-library/pull/607

davidgyoung, 2017-10-26:
Resolved in #607

closed 2017-10-26 by davidgyoung