AltBeacon/android-beacon-library #606: ALT beacon Eddystone detection FAILS. No way to get it work on Android 6.0

allnash, 2017-10-09:
### Expected behavior

### Actual behavior

### Steps to reproduce this behavior

### Mobile device model and OS version

### Android Beacon Library version

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2017-10-10:
@allnash, I have certainly successfully tested detection of Eddystone beacons using the reference app on Android 6.0.  You'll need to post your code so we can find out why it isn't working for you.  Please post this as a new question on StackOverflow.com, as this is the proper forum for project-specific support requests.

closed 2017-10-10 by davidgyoung

allnash, 2017-10-10:
@davidgyoung I will give it a try one more time, I can send you my sample app. I am using the BLU R1 HD phone.

allnash, 2017-10-10:
@davidgyoung Here is the code of your reference App. The only line I added was to be able to see Eddystone beacon URL.

```package org.altbeacon.beaconreference;

import java.util.Collection;

import android.app.Activity;

import android.os.Bundle;
import android.os.RemoteException;
import android.util.Log;
import android.widget.EditText;

import org.altbeacon.beacon.AltBeacon;
import org.altbeacon.beacon.Beacon;
import org.altbeacon.beacon.BeaconConsumer;
import org.altbeacon.beacon.BeaconManager;
import org.altbeacon.beacon.BeaconParser;
import org.altbeacon.beacon.RangeNotifier;
import org.altbeacon.beacon.Region;

public class RangingActivity extends Activity implements BeaconConsumer {
    protected static final String TAG = "RangingActivity";
    private BeaconManager beaconManager = BeaconManager.getInstanceForApplication(this);

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_ranging);
        // Detect the URL frame:
        beaconManager.getBeaconParsers().add(new BeaconParser().
                setBeaconLayout(BeaconParser.EDDYSTONE_URL_LAYOUT));
        beaconManager.bind(this);
    }

    @Override 
    protected void onDestroy() {
        super.onDestroy();
        beaconManager.unbind(this);
    }

    @Override 
    protected void onPause() {
        super.onPause();
        if (beaconManager.isBound(this)) beaconManager.setBackgroundMode(true);
    }

    @Override 
    protected void onResume() {
        super.onResume();
        if (beaconManager.isBound(this)) beaconManager.setBackgroundMode(false);
    }

    @Override
    public void onBeaconServiceConnect() {
        beaconManager.addRangeNotifier(new RangeNotifier() {
           @Override
           public void didRangeBeaconsInRegion(Collection<Beacon> beacons, Region region) {
              if (beacons.size() > 0) {
                 EditText editText = (EditText)RangingActivity.this.findViewById(R.id.rangingText);
                 Beacon firstBeacon = beacons.iterator().next();
                 logToDisplay("The first beacon " + firstBeacon.toString() + " is about " + firstBeacon.getDistance() + " meters away.");
              }
           }

        });

        try {
            beaconManager.startRangingBeaconsInRegion(new Region("myRangingUniqueId", null, null, null));
        } catch (RemoteException e) {   }
    }

    private void logToDisplay(final String line) {
        runOnUiThread(new Runnable() {
            public void run() {
                EditText editText = (EditText)RangingActivity.this.findViewById(R.id.rangingText);
                editText.append(line+"\n");
            }
        });
    }
}
```

allnash, 2017-10-10:
@danialgoodwin please advice ^

davidgyoung, 2017-10-10:
**Please post this as a new question on StackOverflow.com, as this is the proper forum for project-specific support requests.**  Please also note in that question how you have verified your beacon is actually transmitting a valid Eddystone-URL packet, perhaps by testing with the off-the-shelf Locate Beacon app [here.](https://play.google.com/store/apps/details?id=com.radiusnetworks.locate&hl=en)  Again, **Please post this additional info as a new question on StackOverflow.com**.