AltBeacon/android-beacon-library #609: False beacon region entries/exits on Android 7

tpkonrad, 2017-10-20:
Hi David, how are you?

The company I work for uses AltBeacon for a while now and we greatly appreciate your work and effort on it. Our application relies on beacon region entries and exits events and we need to make sure that false entries/exits don't happen. Unfortunately, we've been experiencing some issues on Samsung Galaxy S8 (Android 7), as described below:

### Expected behavior

No false exits when device is close to beacon.

### Actual behavior

Even after setting **ExitRegionPeriod** to 60000ms, **BackgroundScanPeriod** to 1100 and **BackgroundBetweenScanPeriod** to 0,  false exits still happen.

### Steps to reproduce this behavior

Run AltBeacon test app in background for 1 hour or more and log entries/exits.

### Mobile device model and OS version

Samsung Galaxy S8 - Android 7

### Android Beacon Library version

2.12.2

We have been testing Android native BLE scanning API on Android 7 and we found that Batch scan seems to give better results (in terms of frequency of scan results) compared to regular scan. Have you tried this scan mode already? If yes, what are the results of your tests?

Thank you very much,

Thiago Konrad

davidgyoung, 2017-10-20:
@tpkonrad, do you know how frequently your beacon is advertising?  This problem can often be solved by making this more frequent or extending the scanPeriod.

Even with an infrequent advertising interval, it does not sound right to me that you would detect no beacon packets in 60 seconds, so I suspect something is wrong.  A log capture showing system level logs for the 60 secs prior to region exit might help track it down.

I have no data on the relative reliability of batch scans, and offhand I cannot think of an explanation as to why this would make a difference.  I'd be curious to see any data you have.

davidgyoung, 2018-03-21:
Closing due to inactivity.

closed 2018-03-21 by davidgyoung