AltBeacon/android-beacon-library #610: Detect Estimote Beacon in Android beacon library reference

Artestico, 2017-10-23:
Hello , 

I'm having some trouble with the Android library to detect my Estimote beacons . 
We already use Estimote SDK in our Platform situate.io  and we decide recently to migrate our system to use an openSource Library and we start with the AltBeacon library. 

We start testing the AltBecon Android library and we are having trouble with the demo app to detect any Estimote beacons.

We  make some research about using Estimote Beacon and we found a suggestion in estimote forum : https://forums.estimote.com/t/estimote-beacons-not-detected-with-altbeacon-library/3129/4

We apply the code in RangingActivity.class :

```
    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_ranging);
        beaconManager = BeaconManager.getInstanceForApplication(this);
        beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));
        beaconManager.bind(this);
        beaconManager.setDebug(true);
    }
```

No Beacon are detected and this is the Log : 

```
0-23 12:26:32.402 10663-10675/org.altbeacon.beaconreference D/ScanRecord: first manudata for manu ID
10-23 12:26:32.402 10663-10663/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: got record
10-23 12:26:32.402 10663-10663/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: with service uuid: 0000180f-0000-1000-8000-00805f9b34fb
10-23 12:26:32.403 10663-11025/org.altbeacon.beaconreference D/BeaconParser: Ignoring pdu type 01
10-23 12:26:32.404 10663-11025/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 0201061aff4c000215df96db9f46cb4c1dab92ef169f30ea183f4f14f9b6040945535403030f180e160a184f3ff9140fd3b64f3ff9140000000000000000 with startIndex: 5, endIndex: 29
10-23 12:26:32.404 10663-11025/org.altbeacon.beaconreference D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 0201061aff4c000215df96db9f46cb4c1dab92ef169f30ea183f4f14f9b6040945535403030f180e160a184f3ff9140fd3b64f3ff9140000000000000000
10-23 12:26:32.427 10663-10676/org.altbeacon.beaconreference D/ScanRecord: parseFromBytes
10-23 12:26:32.427 10663-10676/org.altbeacon.beaconreference D/ScanRecord: first manudata for manu ID
10-23 12:26:32.428 10663-10663/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: got record
10-23 12:26:32.428 10663-10663/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: with service uuid: 0000180f-0000-1000-8000-00805f9b34fb
10-23 12:26:32.430 10663-11029/org.altbeacon.beaconreference D/BeaconParser: Ignoring pdu type 01
10-23 12:26:32.431 10663-11029/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 0201061aff4c000215df96db9f46cb4c1dab92ef169f30ea186adeb864b6040945535403030f180e160a18de6a64b84cfdb6de6a64b80000000000000000 with startIndex: 5, endIndex: 29
10-23 12:26:32.432 10663-11029/org.altbeacon.beaconreference D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 0201061aff4c000215df96db9f46cb4c1dab92ef169f30ea186adeb864b6040945535403030f180e160a18de6a64b84cfdb6de6a64b80000000000000000
10-23 12:26:32.483 10663-10675/org.altbeacon.beaconreference D/ScanRecord: parseFromBytes
10-23 12:26:32.483 10663-10675/org.altbeacon.beaconreference D/ScanRecord: first manudata for manu ID
10-23 12:26:32.484 10663-10663/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: got record
10-23 12:26:32.484 10663-10663/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: with service uuid: 0000180f-0000-1000-8000-00805f9b34fb
10-23 12:26:32.485 10663-11039/org.altbeacon.beaconreference D/BeaconParser: Ignoring pdu type 01
10-23 12:26:32.486 10663-11039/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 0201061aff4c000215df96db9f46cb4c1dab92ef169f30ea18ec57b309b6040945535403030f180e160a1857ec09b3d4dbb657ec09b30000000000000000 with startIndex: 5, endIndex: 29
10-23 12:26:32.487 10663-11039/org.altbeacon.beaconreference D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 0201061aff4c000215df96db9f46cb4c1dab92ef169f30ea18ec57b309b6040945535403030f180e160a1857ec09b3d4dbb657ec09b30000000000000000
10-23 12:26:32.534 10663-10663/org.altbeacon.beaconreference D/CycledLeScanner: Waiting to stop scan cycle for another 99 milliseconds
10-23 12:26:32.634 10663-10663/org.altbeacon.beaconreference D/CycledLeScanner: Done with scan cycle
10-23 12:26:32.635 10663-10663/org.altbeacon.beaconreference D/BeaconService: Calling ranging callback
10-23 12:26:32.635 10663-10663/org.altbeacon.beaconreference D/Callback: attempting callback via intent: ComponentInfo{org.altbeacon.beaconreference/org.altbeacon.beacon.BeaconIntentProcessor}
10-23 12:26:32.645 10663-10663/org.altbeacon.beaconreference D/CycledLeScanner: Not stopping scan because this is Android N and we keep scanning for a minimum of 6 seconds at a time. We will stop in 2625 millisconds.
10-23 12:26:32.645 10663-10663/org.altbeacon.beaconreference D/CycledLeScanner: starting a new scan cycle
10-23 12:26:32.645 10663-10663/org.altbeacon.beaconreference D/CycledLeScanner: We are already scanning and have been for 3376 millis
10-23 12:26:32.646 10663-10663/org.altbeacon.beaconreference D/CycledLeScanner: Waiting to stop scan cycle for another 1100 milliseconds
10-23 12:26:32.646 10663-10663/org.altbeacon.beaconreference D/CycledLeScanner: Scan started
10-23 12:26:32.649 10663-11517/org.altbeacon.beaconreference D/BeaconIntentProcessor: got an intent to process
10-23 12:26:32.650 10663-11517/org.altbeacon.beaconreference D/BeaconIntentProcessor: got ranging data
10-23 12:26:32.805 10663-10675/org.altbeacon.beaconreference D/ScanRecord: parseFromBytes
10-23 12:26:32.805 10663-10675/org.altbeacon.beaconreference D/ScanRecord: first manudata for manu ID
10-23 12:26:32.806 10663-10663/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: got record

```

### Expected behavior
Detect Estimote Beacons 
### Actual behavior
No Beacon shown in the EditText 
### Steps to reproduce this behavior
First Test :
1- run the android Demo app 
2- allow permission 
3- Click Start Ringing 
Second Test : 
1- add setBeaconLayout code 
2- run the android Demo app 
3- allow permission 
4- Click Start Ringing 

### Mobile device model and OS version
Samsung A5 2016 Android 7.0 
Samsung S5 - SM- G900F Android 6.0.1
Nexus 5X Android 8.0.0 
### Android Beacon Library version
'org.altbeacon:android-beacon-library:2.11'

davidgyoung, 2017-10-23:
The log looks like it may be detecting properly.  Can you show the code that sets up monitoring and ranging and processes the results in the callback?  Do you know if these callbacks are getting called or not?

Artestico, 2017-10-23:
When i start the StartRinging.class i get a callback on the didRangeBeaconsInRegion()
The beacons collections is always empty .

This is the full Activity code source : 
```

public class RangingActivity extends Activity implements BeaconConsumer {
    protected static final String TAG = "RangingActivity";
    private BeaconManager beaconManager;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_ranging);
        beaconManager = BeaconManager.getInstanceForApplication(this);
        beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("m:0-3=4c000215,i:4-19,i:20-21,i:22-23,p:24-24"));
        beaconManager.bind(this);
        beaconManager.setDebug(true);
    }

    @Override
    protected void onDestroy() {
        super.onDestroy();
        beaconManager.unbind(this);
    }

    @Override
    protected void onPause() {
        super.onPause();
        if (beaconManager.isBound(this)) beaconManager.setBackgroundMode(true);
    }

    @Override
    protected void onResume() {
        super.onResume();
        if (beaconManager.isBound(this)) beaconManager.setBackgroundMode(false);
    }

    @Override
    public void onBeaconServiceConnect() {
        beaconManager.setRangeNotifier(new RangeNotifier() {
            @Override
            public void didRangeBeaconsInRegion(Collection<Beacon> beacons, Region region) {
                if (beacons.size() > 0) {
                    //EditText editText = (EditText)RangingActivity.this.findViewById(R.id.rangingText);
                    Beacon firstBeacon = beacons.iterator().next();
                    logToDisplay("The first beacon " + firstBeacon.toString() + " is about " + firstBeacon.getDistance() + " meters away.");
                }
            }

        });

        try {
            beaconManager.startRangingBeaconsInRegion(new Region("myRangingUniqueId", null, null, null));
        } catch (RemoteException e) {
        }
    }

    private void logToDisplay(final String line) {
        runOnUiThread(new Runnable() {
            public void run() {
                EditText editText = (EditText) RangingActivity.this.findViewById(R.id.rangingText);
                editText.append(line + "\n");
            }
        });
    }
}

```
This is the Debug Screenshot : 

[image]

For other class i didn't make any change to the code on the https://github.com/AltBeacon/android-beacon-library-reference

Artestico, 2017-10-23:
This is the list of iBeacon nearby shown by Beacon Scanner : 
[image]

Artestico, 2017-10-23:
yes i' m pretty sure that this code is the one running and i just change the beacon layout early when i found this beacon parser expression on the https://github.com/Bridouille/android-beacon-scanner project . 
 const val IBEACON_LAYOUT = "m:0-3=4c000215,i:4-19,i:20-21,i:22-23,p:24-24"

Artestico, 2017-10-24:
We make some testing with iBeacons and AltBeacon and we still having some trouble : 

1 - Estimote iBeacon are not showing at all on the demo 
2 - Xicato iBeacon are not showing at all on the demo 
3 - Xicato altBeacon **Are Showing Correctly on the demo** 

(Should we wait for your next release or maybe we can help by adding some logs on the library , just if you can guide me where to do it )

[image]
[image]
[image]

davidgyoung, 2017-10-24:
This further supports the theory that the iBeacon parser is not registered with the library, although the code you shared looks correct for doing so.

You might try adding a debug line like this:

Lod.d(TAG, "Registered parser count: "+beacon manager.getBeaconParsers().size());

I would put that in two places:

1. Right after you register the new beacon parser.
2. Inside the didRangeBeaconsInRegion method.

If all is working correctly it should output a value of 2, one for the AltBeacon parser and one for the iBeaconParser..

Artestico, 2017-10-24:
The log is showing 2 but no beacon detected .

[image]
[image]
[image]

davidgyoung, 2017-10-24:
sorry you have having trouble with this, @Artestico.   This really should be dead simple -- there has to be something small wrong that we are overlooking.

The BeaconParser really should start with m:2-3=0215, *not* m:0-3=4c000215, but I doubt that is what is causing your problem, especially on Estimote.  Still, I'd change it to be sure.

And let's change that log line inside didRangeBeaconsInRegion to look like this:

```
        for (BeaconParser parser: mBeaconManager.getBeaconParsers()) {
            Log.d(TAG, "One of the registered beacon parsers is: "+parser.getLayout());
        }
```

That way we can see the two registered beacon parsers to make sure they are the ones we expect to be registered.  I would expect to see this show up in the log:

One of the registered beacon parsers is: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24

Artestico, 2017-10-24:
Thank you for your support . 
we keep having trouble even after changing the parser to  "m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"

[image]
[image]
[image]

davidgyoung, 2017-10-24:
Thanks for your patience, @Artestico.  Let's try something else, and clear the AltBeacon parser and see what happens.  Please change the code to do this:

        beaconManager = BeaconManager.getInstanceForApplication(this);
        beaconManager.getBeaconParsers().clear();
        beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));
        beaconManager.setDebug(true);

Then capture a log output when your iBeacon is transmitting nearby, and try to find the lines that appear after "D/BeaconParser: Processing pdu type FF: "

I'd like to see what that shows.

Artestico, 2017-10-24:
```
10-24 15:19:35.911 15231-15243/org.altbeacon.beaconreference D/ScanRecord: first manudata for manu ID
10-24 15:19:35.912 15231-15231/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: got record
10-24 15:19:35.912 15231-15231/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: with service uuid: 0000180f-0000-1000-8000-00805f9b34fb
10-24 15:19:35.915 15231-15682/org.altbeacon.beaconreference D/BeaconParser: Ignoring pdu type 01
10-24 15:19:35.916 15231-15682/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 0201061aff4c000215df96db9f46cb4c1dab92ef169f30ea186adeb864c4040945535403030f180e160a18de6a64b84cfdc4de6a64b80000000000000000 with startIndex: 5, endIndex: 29
10-24 15:19:35.917 15231-15682/org.altbeacon.beaconreference D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 0201061aff4c000215df96db9f46cb4c1dab92ef169f30ea186adeb864c4040945535403030f180e160a18de6a64b84cfdc4de6a64b80000000000000000
10-24 15:19:36.038 15231-15244/org.altbeacon.beaconreference D/ScanRecord: parseFromBytes
10-24 15:19:36.038 15231-15244/org.altbeacon.beaconreference D/ScanRecord: first manudata for manu ID
10-24 15:19:36.039 15231-15231/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: got record
10-24 15:19:36.039 15231-15231/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: with service uuid: 0000180f-0000-1000-8000-00805f9b34fb
10-24 15:19:36.042 15231-15642/org.altbeacon.beaconreference D/BeaconParser: Ignoring pdu type 01
10-24 15:19:36.043 15231-15642/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 0201061aff4c000215df96db9f46cb4c1dab92ef169f30ea183f4f14f9b6040945535403030f180e160a184f3ff9140fd3b64f3ff9140000000000000000 with startIndex: 5, endIndex: 29
10-24 15:19:36.045 15231-15642/org.altbeacon.beaconreference D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 0201061aff4c000215df96db9f46cb4c1dab92ef169f30ea183f4f14f9b6040945535403030f180e160a184f3ff9140fd3b64f3ff9140000000000000000
10-24 15:19:36.153 15231-15243/org.altbeacon.beaconreference D/ScanRecord: parseFromBytes
10-24 15:19:36.153 15231-15243/org.altbeacon.beaconreference D/ScanRecord: first manudata for manu ID
10-24 15:19:36.153 15231-15231/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: got record
10-24 15:19:36.154 15231-15231/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: with service uuid: 0000180f-0000-1000-8000-00805f9b34fb
10-24 15:19:36.156 15231-15645/org.altbeacon.beaconreference D/BeaconParser: Ignoring pdu type 01
10-24 15:19:36.157 15231-15645/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 0201061aff4c000215df96db9f46cb4c1dab92ef169f30ea18ec57b309b6040945535403030f180e160a1857ec09b3d4dbb657ec09b30000000000000000 with startIndex: 5, endIndex: 29
10-24 15:19:36.159 15231-15645/org.altbeacon.beaconreference D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 0201061aff4c000215df96db9f46cb4c1dab92ef169f30ea18ec57b309b6040945535403030f180e160a1857ec09b3d4dbb657ec09b30000000000000000
10-24 15:19:36.321 15231-15244/org.altbeacon.beaconreference D/ScanRecord: parseFromBytes
10-24 15:19:36.322 15231-15244/org.altbeacon.beaconreference D/ScanRecord: first manudata for manu ID
10-24 15:19:36.322 15231-15231/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: got record
10-24 15:19:36.322 15231-15231/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: with service uuid: 0000180f-0000-1000-8000-00805f9b34fb
10-24 15:19:36.325 15231-15646/org.altbeacon.beaconreference D/BeaconParser: Ignoring pdu type 01
10-24 15:19:36.326 15231-15646/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 0201061aff4c000215df96db9f46cb4c1dab92ef169f30ea186adeb864c4040945535403030f180e160a18de6a64b84cfdc4de6a64b80000000000000000 with startIndex: 5, endIndex: 29
10-24 15:19:36.328 15231-15646/org.altbeacon.beaconreference D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 0201061aff4c000215df96db9f46cb4c1dab92ef169f30ea186adeb864c4040945535403030f180e160a18de6a64b84cfdc4de6a64b80000000000000000
10-24 15:19:36.445 15231-15244/org.altbeacon.beaconreference D/ScanRecord: parseFromBytes
```

[image]

davidgyoung, 2017-10-24:
OK, this proves that the beacon parser change is not being seen by the beacon scanner process.  It still reports:

"10-24 15:19:36.159 15231-15645/org.altbeacon.beaconreference D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 0201061aff4c000215df96db9f46cb4c1dab92ef169f30ea18ec57b309b6040945535403030f180e160a1857ec09b3d4dbb657ec09b30000000000000000"

The fact that it says "Was expecting be ac" means it is still looking for the AltBeacon parser even though we cleared it.  This is not normal.   Perhaps the change to the BeaconParser is not getting applied to the BeaconScanner process.

I think I know what is going on.  If you are using the reference app as a starting point, you need to move the registration of the beacon parsers to the first line of code before the scanning service starts up.  The reference app also includes a class called BeaconReferenceApplication.java.

You need to move the code that registers the new BeaconParser into the onCreate method of this class, right after these lines:

```
    public void onCreate() {
        super.onCreate();
        BeaconManager beaconManager = org.altbeacon.beacon.BeaconManager.getInstanceForApplication(this);
        // *** ADD THE NEW PARSER HERE
```

Artestico, 2017-10-24:
Ok, now we are detecting iBeacons 👍 👍 .
 just i have a question , Should we define BeaconParser layout on  BeaconReferenceApplication.java    and RangingActivity.java to make the ringing working?
when i just comment the BeaconParser layout code on  BeaconReferenceApplication.java  i can see 1 beacon on monitoring but no ringing at all , the ringing is only working when the   
BeaconReferenceApplication.java  have this code : 

```
 beaconManager.getBeaconParsers().clear();
 beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));

```

[image]
[image]

davidgyoung, 2017-10-26:
As described above, you need to set the BeaconLayout before the first call to bind() or the first construction of RegionBootstrap().  So the BeaconReferenceApplication is the right place to do this.

Glad it is working for you now!

closed 2017-10-26 by davidgyoung