AltBeacon/android-beacon-library #612: Identifier parsing creates to many new objects

ekryvoshey, 2017-10-25:
Hello!

I have a problem: my BeaconLayout is like this "m:0-1=7502,p:3-3,i:0-1,i:4-14,i:15-24". Identifiers summary length is pretty long, and it causing Indetifier object to create too many bew Object during toHexString() execution. And that causing to many garbage collection in the app, UI thread is constantly interrupting with GC so I have UI lag every time I'm receiving beacon colection.

https://drive.google.com/open?id=0B_546mjMnsJ9d2QyazJyc1FraU0

As far as I remember, data fields could be an alternative, but aren't they limited to 4 bytes? Perhaps you can suggest different solution to this?

davidgyoung, 2017-10-25:
Yes, data fields are limited to four bytes.

It sounds like the proper answer is to simply optimize the toHexString() method of the Identifier class so toString() isn't so expensive for identifiers that aren't two bytes or 16 bytes.

At first glance simply replacing `String.format("%02x", item)` with code that doesn't use the heavy-weight String.format method might improve things quite a bit.  Here's one such proposed implementation: https://stackoverflow.com/a/32123766/1461050

@ekryvoshey, do you want to try benchmarking a replacement method implementation to see if you can come up with something faster?  If so, I'll put it into a test release for you to try.

ekryvoshey, 2017-10-25:
Yes, I would like to try it.

ekryvoshey, 2017-10-25:
Yes, I would like to try and figure out the better solution. If you can assist, it would be great!