AltBeacon/android-beacon-library #613: beaconManager.isBound() does not work in Android O

mgsteinkamp, 2017-10-26:
### Expected behavior
After onBeaconServiceConnect(), beaconManager.isBound() will evaluate as true
### Actual behavior
After onBeaconServiceConnect(), beaconManager.isBound() evaluates as false
### Steps to reproduce this behavior
```
beaconManager.bind(BeaconConsumer b);
...
onBeaconServiceConnect();
...
Log.i(TAG, "Bound: " + beaconManager.isBound());
```

### Mobile device model and OS version
Nexus 5X

### Android Beacon Library version
2.12.3

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

Possibly related: RegionBootstrap doesn't seem to function any more either, but sorry, I don't have time to file a separate report

davidgyoung, 2017-10-26:
Thanks for this report.  A proposed solution is described in #600

davidgyoung, 2018-04-27:
Fixed in #655

closed 2018-04-27 by davidgyoung