AltBeacon/android-beacon-library #614: Android O (8.0.0) doesn't stop scanning on unbind

samanthaeschaffer, 2017-10-27:
I'm pretty sure this is a known issue based on [this TODO](https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/java/org/altbeacon/beacon/BeaconManager.java#L461), but being able to turn off beacon scanning is a pretty big feature to be missing (unless I'm missing some work around?)

### Expected behavior
A call to unbind should stop the app from scanning for beacons.

### Actual behavior
App scans for beacons indefinitely after calling unbind.

### Steps to reproduce this behavior
Call unbind on a device running Android O and observe that didRangeBeaconsInRegion continues to be called.

### Mobile device model and OS version
Pixel, Android O (8.0.0)

### Android Beacon Library version
2.12.3

davidgyoung, 2017-10-27:
Yes, I agree it should stop scanning after a call to unBind() for the sake of backward compatibility.

In the interim, you might consider calling stopRangingBeaconsInRegion

samanthaeschaffer, 2017-10-31:
@davidgyoung Ah yup, there's the workaround I was missing. Thanks for that.~

Even with that workaround though, wouldn't the indefinite scanning (especially if combined with longScanForcingEnabled) adversely affect battery life?

davidgyoung, 2017-10-31:
I think the workaround is pretty safe when it comes to battery.  If no regions are monitored or ranged, then this `ScanJob` code will kick in:

```
        if (mScanHelper.getRangedRegionState().size() > 0 || mScanHelper.getMonitoringStatus().regions().size() > 0) {
            mScanHelper.getCycledScanner().start();
            return true;
        }
        else {
            mScanHelper.getCycledScanner().stop();
            return false;
        }
```

So basically there won't be any scanning active at all during normal scan cycles.

However, when between scan cycles, I see that the code in `startPassiveScanIfNeeded();` does not check to see if any monitoring regions are active, and will needlessly start a passive scan even if there are none.  While the power usage of this is relatively small, there is no reason to do this, so this should be fixed as well as part of the fix for this issue.

stephenruda, 2020-07-23:
Hi @davidgyoung , sorry to bump this issue but I am just looking for some clarification and I think this issue seems close to my own.  If you need me to open a new issue let me know...

In my app, I allow users to press a button and will range for a Beacon for a short time (or until the Beacon is found).

When the user pushes the button to start ranging, I will call `beaconManager.bind(this);` which will trigger
```
@Override
public void onBeaconServiceConnect() {
    try {
       beaconManager.startRangingBeaconsInRegion(...);
       beaconManager.addRangeNotifier(this);
    }
    catch(RemoteException e) {
        e.printStackTrace();
    }
}
```

Then after the Beacon has been found through ranging or if the Beacon has not been found in a small amount of time (30 seconds or so) I will stop the process by calling...
```
try {
    beaconManager.stopRangingBeaconsInRegion(...);
}
catch(RemoteException e) {
    e.printStackTrace();
}
beaconManager.removeAllRangeNotifiers();
beaconManager.unbind(this);
```

So at this point I have stop ranging for the beacon, removed the notifiers, and called unbind.  I would expect there to be nothing 'registered' when it comes to the Beacon Library.  However, I still see some console logs that indicates something is still 'registered'.  About every 10 seconds I see logs that look like this...
```
I/ScanJob: Using immediateScanJobId from manifest: 208352939
I/ScanJob: Using periodicScanJobId from manifest: 208352940
```
Even after I have closed the app, those keep showing up in my logs about every 10 seconds.

On top of this, when I turn on debug mode with `BeaconManager.setDebug(true)` I see more logs that indicate that something is still 'registered'.
These logs are seen every few seconds, repeating very frequently...
```
D/StartupBroadcastReceiver: onReceive called in startup broadcast receiver
D/StartupBroadcastReceiver: Passive background scan callback type: 1
D/StartupBroadcastReceiver: got Android O background scan via intent
D/ScanJobScheduler: Not scheduling an immediate scan job because we just did recently.
```

Again, seeing this about every 10 seconds...
```
D/ScanJobScheduler: scheduling an immediate scan job because last did 10022millis ago.
D/ScanState: Scan state restore regions: monitored=0 ranged=0
D/BeaconLocalBroadcastProcessor: Register calls: global=19 instance=19
D/ScanState: ScanState says background mode for ScanJob is false
D/ScanJobScheduler: We just woke up in the background based on a new scan result or first run of the app. Start scan job immediately.
D/ScanJobScheduler: We are not monitoring or ranging any regions.  We are going to cancel all scan jobs.
I/ScanJob: Using immediateScanJobId from manifest: 208352939
I/ScanJob: Using periodicScanJobId from manifest: 208352940
```

And I keep getting these logs even after I have closed the app.

Then I added `beaconManager.setEnableScheduledScanJobs(false);` right before the `beaconManager.bind(this)` call, so that we disable the jobs before we bind.  This solved all of my problems and I am no longer seeing anything 'registered' after the unbind.  All of those logs I mentioned above effectively stop now when I call `unbind()`.

**My ultimate question(s)**:
- Is something from the JobScheduler not getting unregistered properly?  It seems like some kind of job is still scheduled even after I have cleared everything out and called `unbind()`.
- Unrelated - If I were to apply custom scan periods or add a BeaconParser, would it be smarter / more efficient to do this before or after calling `bind()`?

I understand that disabling scan jobs may disable delivery of beacon callbacks in the background on Android 8+, but I am only worried about ranging in the foreground for a short period of time for this case.

I am running on a Pixel 3 XL on Android 10 using library version 2.17.1.

Thanks!

davidgyoung, 2020-07-23:
Indeed, this is a bug:

1. Scan jobs should be stopped if no ranging/monitoring regions are active
2. Between scan period low power scans should not be started if no ranging/monitoring regions are active

This has simply not been a priority as the battery consequences are usually not significant.  The primary consequence is log noise.

If this is more important for your particular use case and you want to work on a fix, I can provide tips to do so.

stephenruda, 2020-07-23:
Hi David,

Thank you for the reply.

I have added the `setEnableScheduledScanJobs(false)` to my code before I bind and everything is working well.  I don't see any extra logs after I unbind.  So I don't need the bug fix immediately because I have found a way around the problem.  I'd be happy to look into opening up a PR to fix the problem if you want to leave any tips for me here.  I can look into it on one of my days off.

My guess is that when I `unbind()`, I should check if any `isAnyConsumerBound()`and if there are none to also cancel all scheduled jobs using `ScanJobScheduler.getInstance().cancelSchedule();`.  How would I cancel those between scan period low power scans?  Finally, once those things are 'unregistered' will I have to do extra work on the next `bind()` to 're-register' these two things or would that just happen automatically?

Any advice you have to help me get started would be great.  Again, I won't be looking into this right away but I will when I have some time.

Thanks!

davidgyoung, 2020-07-23:
Looking in the BeaconManager.java unbind method, there is already code to cancel the scan jobs. It is unclear why that is not working:

```
                if (consumers.size() == 0) {
                    // If this is the last consumer to disconnect, the service will exit
                    // release the serviceMessenger.
                    serviceMessenger = null;
                    // If we are using scan jobs, we cancel the active scan job
                    if (mScheduledScanJobsEnabled) {
                        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.LOLLIPOP) {
                            LogManager.i(TAG, "Cancelling scheduled jobs after unbind of last consumer.");
                            ScanJobScheduler.getInstance().cancelSchedule(mContext);
                        }
                    }
                }
```

This needs investigation.  There should probably be a new method ScanJobScheduler.getInstance().shutDown(mContext) or something like that, that not only cancels the jobs but also stops the background scanning like this:

```
new ScanHelper(context).stopAndroidOBackgroundScan()
```

davidgyoung, 2021-08-06:
Closing in favor of #1046

closed 2021-08-06 by davidgyoung