AltBeacon/android-beacon-library #616: NPE Background Crash in ScanHelper#processBeaconFromScan

hustoj2, 2017-11-05:
### Actual behavior
* App crashed in background due to NPE
* The phone showed a popup saying the app which uses altbeacon has crashed, but this alert dismissed itself after less than a second without interaction on my part.

### Steps to reproduce this behavior
I don't know reproducible steps. At the time this happened, the following things were happening:
* The phone was connected to my car's bluetooth which uses HFP, A2DP, and AVRCP protocols
* I was playing music with Spotify over the Bluetooth
* Two Tile beacons would have been in range
* I hadn't run the app that uses AltBeacon for several days

### Mobile device model and OS version
Pixel 2 (walleye), Android 8.0

### Android Beacon Library version
2.12.2

### Logs
> 11-02 08:33:06.597 10156 25957 26151 D BluetoothAdapter: isLeEnabled(): ON
> 11-02 08:33:06.598  1002  1401  2142 E bt_att  : GATT_Register: can't Register GATT client, MAX client 32 reached!
> 11-02 08:33:06.598  1002  1401  2142 E bt_btif : Register with GATT stack failed.
> 11-02 08:33:06.598  1002  1401  2142 E bt_att  : GATT_Register: can't Register GATT client, MAX client 32 reached!
> 11-02 08:33:06.598  1002  1401  2142 E bt_btif : Register with GATT stack failed.
> 11-02 08:33:06.598 10156 25957 25971 D BluetoothLeScanner: onScannerRegistered() - status=133 scannerId=0 mScannerId=0
> 11-02 08:33:07.838 10198  7094  7094 I ScanJob : Scan job runtime expired
> 11-02 08:33:07.845 10198  7094 14338 D BluetoothAdapter: isLeEnabled(): ON
> 11-02 08:33:07.898 10198  7094  7094 D BluetoothAdapter: isLeEnabled(): ON
> 11-02 08:33:08.127 10198  7094  7094 W JobInfo : Specified interval for 1 is +5m10s0ms. Clamped to +15m0s0ms
> 11-02 08:33:08.127 10198  7094  7094 W JobInfo : Specified flex for 1 is 0. Clamped to +5m0s0ms
> 11-02 08:33:08.128 10198  7094  7094 W JobInfo : Specified interval for 1 is +5m10s0ms. Clamped to +15m0s0ms
> 11-02 08:33:08.128 10198  7094  7094 W JobInfo : Specified flex for 1 is 0. Clamped to +5m0s0ms
> 11-02 08:33:08.150 10198  7094  7094 I ScanJob : Running immdiate scan job: instance is org.altbeacon.beacon.service.ScanJob@e378df3
> 11-02 08:33:08.151 10198  7094  7094 I ScanJob : scanJob version 2.12.2 is starting up on the main process
> 11-02 08:33:08.155 10198  7094  7094 W ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
> 11-02 08:33:08.155 10198  7094  7094 W ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
> 11-02 08:33:08.166 10198  7094  7094 I CycledLeScanner: Using Android O scanner
> 11-02 08:33:08.170 10198  7094  7094 D BluetoothAdapter: isLeEnabled(): ON
> 11-02 08:33:08.176 10198  7094 14360 D BluetoothAdapter: isLeEnabled(): ON
> 11-02 08:33:08.176 10198  7094  7094 I ScanJob : Scan job running for 10000 millis
> 11-02 08:33:08.178 10198  7094  7120 D BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=15 mScannerId=0
> 11-02 08:33:08.477 10198  7094 14362 I ScanHelper: Non-distinct packets detected in a single scan.  Restarting scans unecessary.
> 11-02 08:33:09.429 10041 11725 11725 I CarrierServices: [2] n.a: CellServiceUtils.getRegisteredCellInfo, no coarse location permission.
> 11-02 08:33:09.451 10041 11725 11725 D ConnectivityManager: getMobileDataEnabled()+ subId=1
> 11-02 08:33:09.453 10041 11725 11725 D ConnectivityManager: getMobileDataEnabled()- subId=1 retVal=true
> 11-02 08:33:09.677  1000   786   873 I CHRE    : @ 1167657.750: [AR_CHRE] detected in-vehicle, confidence: 84, motionConfidence: 0
> 11-02 08:33:09.679  1000   786   873 I CHRE    : @ 1167657.750: [AR_CHRE] send to host, reason=1, sampleSize=1, transitionSize=0
> 11-02 08:33:09.695  1000   786   873 I VSC     : @ 1167657.875: [SMD] Publishing SMD with confidence 100
> 11-02 08:33:11.007 10192 11397 11397 I ScanJob : Scan job runtime expired
> 11-02 08:33:11.012 10192 11397 14356 D BluetoothAdapter: isLeEnabled(): ON
> 11-02 08:33:11.040 10192 11397 11397 D BluetoothAdapter: isLeEnabled(): ON
> 11-02 08:33:11.097 10192 11397 11397 W JobInfo : Specified interval for 1 is +5m10s0ms. Clamped to +15m0s0ms
> 11-02 08:33:11.097 10192 11397 11397 W JobInfo : Specified flex for 1 is 0. Clamped to +5m0s0ms
> 11-02 08:33:11.097 10192 11397 11397 W JobInfo : Specified interval for 1 is +5m10s0ms. Clamped to +15m0s0ms
> 11-02 08:33:11.097 10192 11397 11397 W JobInfo : Specified flex for 1 is 0. Clamped to +5m0s0ms
> 11-02 08:33:11.104 10192 11397 11397 I ScanJob : Running immdiate scan job: instance is org.altbeacon.beacon.service.ScanJob@c0aaf2c
> 11-02 08:33:11.116 10192 11397 11397 I ScanJob : scanJob version 2.12.2 is starting up on the main process
> 11-02 08:33:11.117 10192 11397 11397 W ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
> 11-02 08:33:11.117 10192 11397 11397 W ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
> 11-02 08:33:11.129 10192 11397 14375 E AndroidRuntime: FATAL EXCEPTION: pool-66-thread-7
> 11-02 08:33:11.129 10192 11397 14375 E AndroidRuntime: Process: com.hyvee.android, PID: 11397
> 11-02 08:33:11.129 10192 11397 14375 E AndroidRuntime: java.lang.RuntimeException: An error occurred while executing doInBackground()
> 11-02 08:33:11.129 10192 11397 14375 E AndroidRuntime: 	at android.os.AsyncTask$3.done(AsyncTask.java:353)
> 11-02 08:33:11.129 10192 11397 14375 E AndroidRuntime: 	at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
> 11-02 08:33:11.129 10192 11397 14375 E AndroidRuntime: 	at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
> 11-02 08:33:11.129 10192 11397 14375 E AndroidRuntime: 	at java.util.concurrent.FutureTask.run(FutureTask.java:271)
> 11-02 08:33:11.129 10192 11397 14375 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
> 11-02 08:33:11.129 10192 11397 14375 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
> 11-02 08:33:11.129 10192 11397 14375 E AndroidRuntime: 	at java.lang.Thread.run(Thread.java:764)
> 11-02 08:33:11.129 10192 11397 14375 E AndroidRuntime: Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'org.altbeacon.beacon.Beacon org.altbeacon.beacon.service.ExtraDataBeaconTracker.track(org.altbeacon.beacon.Beacon)' on a null object reference
> 11-02 08:33:11.129 10192 11397 14375 E AndroidRuntime: 	at org.altbeacon.beacon.service.ScanHelper.processBeaconFromScan(ScanHelper.java:287)
[... 3 lines cut ...]
> 11-02 08:33:11.129 10192 11397 14375 E AndroidRuntime: 	at android.os.AsyncTask$2.call(AsyncTask.java:333)
> 11-02 08:33:11.129 10192 11397 14375 E AndroidRuntime: 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
> 11-02 08:33:11.129 10192 11397 14375 E AndroidRuntime: 	... 3 more
> 11-02 08:33:11.129 10192 11397 11397 I CycledLeScanner: Using Android O scanner
> 11-02 08:33:11.132 10192 11397 14376 E AndroidRuntime: FATAL EXCEPTION: pool-66-thread-8
> 11-02 08:33:11.132 10192 11397 14376 E AndroidRuntime: Process: com.hyvee.android, PID: 11397
> 11-02 08:33:11.132 10192 11397 14376 E AndroidRuntime: java.lang.RuntimeException: An error occurred while executing doInBackground()
> 11-02 08:33:11.132 10192 11397 14376 E AndroidRuntime: 	at android.os.AsyncTask$3.done(AsyncTask.java:353)
> 11-02 08:33:11.132 10192 11397 14376 E AndroidRuntime: 	at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
> 11-02 08:33:11.132 10192 11397 14376 E AndroidRuntime: 	at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
> 11-02 08:33:11.132 10192 11397 14376 E AndroidRuntime: 	at java.util.concurrent.FutureTask.run(FutureTask.java:271)
> 11-02 08:33:11.132 10192 11397 14376 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
> 11-02 08:33:11.132 10192 11397 14376 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
> 11-02 08:33:11.132 10192 11397 14376 E AndroidRuntime: 	at java.lang.Thread.run(Thread.java:764)
> 11-02 08:33:11.132 10192 11397 14376 E AndroidRuntime: Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'org.altbeacon.beacon.Beacon org.altbeacon.beacon.service.ExtraDataBeaconTracker.track(org.altbeacon.beacon.Beacon)' on a null object reference
[... cut ...]
> 11-02 08:33:11.132 10192 11397 14376 E AndroidRuntime: 	at android.os.AsyncTask$2.call(AsyncTask.java:333)
> 11-02 08:33:11.132 10192 11397 14376 E AndroidRuntime: 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
> 11-02 08:33:11.132 10192 11397 14376 E AndroidRuntime: 	... 3 more
> 11-02 08:33:11.137 10192 11397 14374 I ScanHelper: Non-distinct packets detected in a single scan.  Restarting scans unecessary.
> 11-02 08:33:11.138 10192 11397 11397 D BluetoothAdapter: isLeEnabled(): ON
> 11-02 08:33:11.148 10192 11397 14379 D BluetoothAdapter: isLeEnabled(): ON

davidgyoung, 2017-11-05:
Thanks for this report.   This appears to be caused by a race condition that will only happen (sometimes) on Android 8 if beacons are newly detected in the background after not being detected for awhile.

The problem is in this block of code on ScanJob.java :

```
        List<ScanResult> queuedScanResults = ScanJobScheduler.getInstance().dumpBackgroundScanResultQueue();
        LogManager.d(TAG, "Processing %d queued scan resuilts", queuedScanResults.size());
        for (ScanResult result : queuedScanResults) {
            ScanRecord scanRecord = result.getScanRecord();
            if (scanRecord != null) {
                mScanHelper.processScanResult(result.getDevice(), result.getRssi(), scanRecord.getBytes());
            }
        }
        LogManager.d(TAG, "Done processing queued scan resuilts");

        boolean startedScan;
        if (mInitialized) {
            LogManager.d(TAG, "Scanning already started.  Resetting for current parameters");
            startedScan = restartScanning();
        }
        else {
            startedScan = startScanning();
        }
```

The `mScanHelper` object is not fully initialized until the  `restartScanning()` or `startScanning()` methods are called at the end of the code excerpt, and `mExtraDataBeaconTracker` might be left null until this initialization happens.  If a  new background detection happened it will cause a call to `mScanHelper.processScanResult(...)`.  If this call executes before this initialization completes later down in the method, then a crash will occur.  This only happens *sometimes* because the call happens on a separate thread, and often times it is delayed until the initialization completes anyway.  This is why the crash does not *always* happen.

The fix is to move the initialization of `mScanHelper` to happen earlier.

ncapdevi, 2017-12-06:
Also get this
```
Fatal Exception: java.lang.RuntimeException: An error occurred while executing doInBackground()
       at android.os.AsyncTask$3.done(AsyncTask.java:353)
       at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
       at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
       at java.util.concurrent.FutureTask.run(FutureTask.java:271)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
       at java.lang.Thread.run(Thread.java:764)
Caused by java.lang.NullPointerException: Attempt to invoke virtual method 'org.altbeacon.beacon.Beacon org.altbeacon.beacon.service.ExtraDataBeaconTracker.track(org.altbeacon.beacon.Beacon)' on a null object reference
       at org.altbeacon.beacon.service.ScanHelper.processBeaconFromScan(ScanHelper.java:291)
       at org.altbeacon.beacon.service.ScanHelper.access$600(ScanHelper.java:60)
       at org.altbeacon.beacon.service.ScanHelper$ScanProcessor.doInBackground(ScanHelper.java:375)
       at org.altbeacon.beacon.service.ScanHelper$ScanProcessor.doInBackground(ScanHelper.java:340)
       at android.os.AsyncTask$2.call(AsyncTask.java:333)
       at java.util.concurrent.FutureTask.run(FutureTask.java:266)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
       at java.lang.Thread.run(Thread.java:764)
```

davidgyoung, 2017-12-07:
Fixed in #626

closed 2017-12-07 by davidgyoung