AltBeacon/android-beacon-library #618: Region registering (monitoring, ranging) takes way too long (Android O)

tomacco, 2017-11-09:
### Expected behavior
In Android 7 or lower, the execution time of `startMonitoringBeaconsInRegion` and `startRangingBeaconRegion` is very fast. 
Behavior on a Galaxy S5 running 6.0.1. Time to register 168 regions:
- `startMonitoringBeaconsInRegion`: 384ms
- `startRangingBeaconRegion`: 382ms

### Actual behavior

In a Nexus 5X running Android 8.0.0. Time to register 168 regions:
- `startMonitoringBeaconRegion`: 3606 ms (10 x times more)
- `startRangingBeaconRegion`:  17558ms (45 x times more)

### Steps to reproduce this behavior

- Register several regions using the library on Android 8.0

### Mobile device model and OS version

Nexus 5X - Android 8.0.0

### Android Beacon Library version

2.12.3

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2017-11-09:
The reasons this happens is because on Android 8, new regions must be serialized to persistent storage so that a ScheduledJob may be used to look for them at a later time.  

The library is not really designed to be used with such a large number of regions, and using more than 50 disables remembering region state across app restarts and may lead to spurious enter/exit events.

It might be possible to improve this performance by disabling the auto serialization of regions until an explicit commit method is called.

tomacco, 2017-11-09:
Thank you @davidgyoung 

I will move all this work off the main thread by using an `AsyncTask`.
Is it safe to do so for the AltBeacon Library?

davidgyoung, 2017-11-09:
Yes, it is safe to register regions off the main thread.