AltBeacon/android-beacon-library #619: Attempt to invoke virtual method 'java.lang.String org.altbeacon.beacon.Region.getUniqueId()' on a null object reference

Hubertzzz, 2017-11-13:
### Expected behavior
After monitoring a beacon then get its uniqueId.
### Actual behavior
The app crash and here is the information.
Attempt to invoke virtual method 'java.lang.String org.altbeacon.beacon.Region.getUniqueId()' on a null object reference
### Steps to reproduce this behavior
Open the app  then wait for a while it crashes.
### Mobile device model and OS version
One plus 3T. android 7.1.1
### Android Beacon Library version
2.9.2
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

Hubertzzz, 2017-11-13:
Here is the code.
`  public void didDetermineStateForRegion(int status, Region region) {
        final SharedPreferences pref = getSharedPreferences(Preferences.SharedPreferencesTag,Preferences.SharedPreferences_ModeTag);

        String isLogin = pref.getString("isLogin","false");

        if (isLogin.equals("true") && region != null){
            //in the beacon region
            if(status == 1){
                try{
                    final String auCode = pref.getString("authorizationCode",null);
                    final String lecturerId = pref.getString("lecturer_id",null);

                    String[] studentId_lessonDateId = region.getUniqueId().split(";");

                    if(studentId_lessonDateId.length > 0){
                        //Get data of detected student
                        String detectedStudentId = studentId_lessonDateId[0];
                        String lessonDateId = studentId_lessonDateId[1];`

davidgyoung, 2017-11-13:
Hi @Hubertzzz.

I have tried reproducing this problem with the reference app and version
2.9.2 with no success.  Can you please show me the Region definition you
are using that causes this crash?   I'm especially interested in knowing
the uniqueId value you are using, as well as the beacon identifiers.

If you prefer not to share your specific beacon identifier, please feel
free to alter a few of the digits.

Hubertzzz, 2017-11-13:
This is the region
`region = new Region("Late student", Identifier.parse(monitors.get(0).getUuid());,null,null);`
for the Uuid is from the database I established, and it saved from the server. A specific UUID.
Beacon parser:
`mBeaconmanager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));`

davidgyoung, 2017-11-13:
Hmmmm.  Still can't reproduce.  

Can you try turning on beconManager.setDebug(true); then capturing a logcat sequence for about 10 seconds before the crash happens?

georgeRCluj, 2017-12-15:
@davidgyoung, I just got the same error encountered by @Hubertzzz. We use intensively this library in more applications and this is the first time this happens. It happens from time to time but especially when I updated from 'org.altbeacon:android-beacon-library:2.9.2' to 'org.altbeacon:android-beacon-library:2.12.3' two days ago. Should I go back to 2.9.2?

In "stopScanning" method crashes, but not fatally. It just does not go into any of the callbacks, because of the crash in BeaconHandler.

public void onEvent(final PublishReportEvent event) {
        beaconHandler.stopScanning(new BeaconHandler.ClosestBeaconCallback() {
            @Override
            public void closestBeaconFound(Beacon beacon) {
                EventBus.getDefault().post(new ClosestBeaconFoundEvent(beacon));
                com.transreportuk.storage.LocalStorage.getInstance(getApplicationContext()).store("UUID", beacon.getIdentifier(0));
                com.transreportuk.storage.LocalStorage.getInstance(getApplicationContext()).store("MAJOR", beacon.getIdentifier(1));
                com.transreportuk.storage.LocalStorage.getInstance(getApplicationContext()).store("MINOR", beacon.getIdentifier(2));
            }

            @Override
            public void closestBeaconNotFound() {
                EventBus.getDefault().post(new ClosestBeaconFoundEvent(null));
            }
        });
        beaconHandler.unbindBeaconManager();
    }

davidgyoung, 2017-12-15:
Any chance I can get a stack trace of the crash?  Without one,  finding he cause is basically guesswork.