AltBeacon/android-beacon-library #620: Device RSSI measurements Moto G4, Nexus 5X, Huawei P9 Lite

davidgyoung, 2017-11-18:
These data were collected using a RadBeacon USB transmitting iBeacon at +0dBm (max power) at 20Hz, with an iPhone 5 measured power at 1m of -66 dBm.

RSSI was the average of the last 30 seconds of packets received discarding the top 10% and bottom 10% of RSSI measurements.

Transmitter was placed on a cardboard box elevated approx. 12 inches above ground level in an open field.  The beacon side opposite the USB connector was aimed in the direction of measurements.  Measurements were taken with mobile phones at recorded distances with the back of the phones facing the beacon transmitter.

Device Name: HUAWEI P9 Lite
Model number: HUAWEI VNS-L21
Build number: VNS-L21C432B380
Android version: 7.0
Manufacturer: HUAWEI

Device Name:  Moto G4 Plus
Model Number:  Moto G4 Plus
Build number: NPJ25.93-14.7
Android version: 7.0
Manufacturer: motorola

Device Name: Nexus 5X
Model Number: Nexus 5X
Build Number: MDB08L 
Android version: 6.0
Manufacturer: LGE

Distance | Nexus 5X | Huawei 9Lite | Moto G4
-- | -- | -- | --
0.25 | -45 | -75.5 | -60.9
0.5 | -51 | -76 | -59
1 | -55.1 | -75.7 | -66.4
2 | -66.3 | -78.7 | -75.1
3 | -70.8 | -85.2 | -79.2
4 | -76.2 | -89.9 | -84.9
5 | -77.9 | -92.3 | -86.1
6 | -83.2 | -95.3 | -90.6
7 | -83.4 | -92.1 | -90.8
8 | -82.4 | -97 | -93.5
9 | -88 | -95.8 | -94
10 | -92.8 |   | -96.2
12 | -90.1 | -99.3 | -95.3
14 | -89.7 |   | -97.8
16 | -92.1 |   | -98.8
18 | -90.6 |   | -95.3
20 | -89.9 |   | -98.4
25 | -91 |   | -97.9
30 | -92.5 |   | -99.3
35 | -95 |   | -99
40 | -92.9 |   | -99.3

Notes:

* The Huawei P9 Lite could not detect the beacon at all at distances of 10 meters or 14-40 meters.  For data recorded in the spreadsheet for regression purposes, a dBm of -100 is recorded for these cells as it is slightly higher than the weakest signal ever detected for this device of -99.3 dBm.

Analysis:

* Clearly the Huawei device has a very weak receiver, detecting a very low signal level even at close distances and unable to detect beacons at all at distances of 10m or more.

* Across all devices measured, signal level at 10m - 40m did not appear to be correlated at all with distance.  Signal levels at these distances appeared to vary randomly at the minimum detectable signal level for the device.  There may be better correlation for other device models, but not for these.

Conclusions:

* For regression purposes, I think it may be useful to discard data points between 1-10 meters, as above 10 meters the measured values are not correlated with distance.  I suspect doing so may

davidgyoung, 2017-11-18:
Recorded these numbers in Google Docs here:

https://docs.google.com/spreadsheets/d/1iEusNfB0HgyzwPM25_oAsfNe8uxCUHix5fk1RPIeBGs/edit#gid=420885588

closed 2017-11-18 by davidgyoung

pierre-oord, 2017-11-19:
Thanks David!

I'll get back to you soon. I'm currently analyzing the variance in signal measurements between devices. So far, it seems devices are pretty similar in that respect, which is great as it allows me to use the same Kalman filter for every device.

I'm writing some conclusions myself and will post them here once ready!

In the meantime, if you have more data, let me know!

pierre-oord, 2018-04-17:
@sharker92 yes I did. I found the difference between a moving average filter was not very large. Actually, the moving average filter was performing slightly better when I did some virtual tests with moving objects in Excel.

We are currently switching our approach. In my experience, phones of different models were showing very different results, even after calibration. The speed of getting new readings differs, the amount of outliers differs, and even worse, on iPhone's it pretty impossible to keep measuring Bluetooth in the background.

We decided to do the reading from the device the phones are connecting to. For that, we had to change to a different processor, one that supports multiple concurrent connections. Now, readings are much more accurate, fast, and also much more comparable even without doing any kind of calibration.