AltBeacon/android-beacon-library #622: IllegalArgumentException in BeaconManager.startRangingBeaconsInRegion on Android O 8.0.0

dhabensky, 2017-11-20:
I am getting IllegalArgumentException during call to `startRangingBeaconsInRegion` on Android 8.0.0

Seems like some data is passed incorrectly and thus producing an exception.

Stacktrace:
`android.os.Parcel.readException (Parcel.java:1946)
android.os.Parcel.readException (Parcel.java:1888)
android.app.job.IJobScheduler$Stub$Proxy.schedule (IJobScheduler.java:180)
android.app.JobSchedulerImpl.schedule (JobSchedulerImpl.java:44)
org.altbeacon.beacon.service.ScanJobScheduler.schedule (ScanJobScheduler.java:156)
org.altbeacon.beacon.service.ScanJobScheduler.applySettingsToScheduledJob (ScanJobScheduler.java:85)
org.altbeacon.beacon.service.ScanJobScheduler.applySettingsToScheduledJob (ScanJobScheduler.java:92)
org.altbeacon.beacon.BeaconManager.applyChangesToServices (BeaconManager.java:962)
org.altbeacon.beacon.BeaconManager.startRangingBeaconsInRegion (BeaconManager.java:814)
com.a1s.navimuseum.location.BeaconLocationService.onBeaconServiceConnect (BeaconLocationService.java:142)
org.altbeacon.beacon.BeaconManager.bind (BeaconManager.java:418)
com.a1s.navimuseum.location.BeaconLocationService$1.doInBackground (BeaconLocationService.java:86)
com.a1s.navimuseum.location.BeaconLocationService$1.doInBackground (BeaconLocationService.java:72)
android.os.AsyncTask$2.call (AsyncTask.java:333)
java.util.concurrent.FutureTask.run (FutureTask.java:266)
android.os.AsyncTask$SerialExecutor$1.run (AsyncTask.java:245)
java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1162)
java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:636)
java.lang.Thread.run (Thread.java:764)`

### Expected behavior
Ranging gets started

### Actual behavior
Application crashes

### Steps to reproduce this behavior
bind to `BeaconManager`
call `startRangingBeaconsInRegion(region)`

if it is crucial:
 - `startRangingBeaconsInRegion` is called not in main thread
 - `region` argument has all three ids set to null

### Mobile device model and OS version
LG Nexus 5x, Android 8.0.0

### Android Beacon Library version
2.12.3

davidgyoung, 2017-11-20:
Thanks for this report.

I do not understand how this is possible.  The exception happens after these two lines.  The first line creates an empty PersistableBundle.  The exception in the second line appears to be caused by unmarshalling this empty bundle, which should not be a problem..  But perhaps it is some internal error in the Android class.

```
JobInfo immediateJob = new JobInfo.Builder(ScanJob.IMMEDIATE_SCAN_JOB_ID, new ComponentName(context, ScanJob.class))
                        .setPersisted(true) // This makes it restart after reboot
                        .setExtras(new PersistableBundle())
                        .setMinimumLatency(millisToNextJobStart)
                        .setOverrideDeadline(millisToNextJobStart).build();
int error = jobScheduler.schedule(immediateJob);
```     

You say the calling class must not be on the main thread.  This should be ok, too.  Are you suggesting you do not see the problem if the method is called on the main thread?

dhabensky, 2017-11-20:
Unfortunatly, I have no access to this device, it is just a crash report from the user. I have no info about how it will behave if called from the main thread.
But I should say that binding to BeaconManager right after turning bluetooth on causes a noticable lag (up to several seconds) on the main thread, that is the reason why I call bind on a background thread.

dhabensky, 2018-01-10:
This issue got reproduced on two more devices running Android O 8.0.0.
I have just read that persistable jobs require RECEIVE_BOOT_COMPLETED permission.
Earlier, I intentionally removed this permission because my application starts scanning only on one of its activities and it does not require boot completed event.
May this crash be related to lacking RECEIVE_BOOT_COMPLETED in manifest?

My AndroidManifest fragment:
  `  <uses-permission android:name="android.permission.RECEIVE_BOOT_COMPLETED" tools:node="remove" />`

davidgyoung, 2018-01-10:
Good find!  I think this may be it.  You should be s le to test my simply rebooting a device with your app then connecting to ADB shortly after boot to look for a log line with the crash.  Then repeat the tests with the RECEIVE_BOOT_COMPLETED permission.

dhabensky, 2018-01-11:
It seems that my guess was right: **RECEIVE_BOOT_COMPLETED is required for scans to work on Android O.**
It would be great to remove this limitation in future releases. 

Personally I don't understand the need to persist _ALL_ jobs across device reboots.
Probably it should be enough to remove `.setPersisted(true)` or call it only if the permission is granted.
But this is just a guess, I don't have full understanding of code, neither had tested it.

Thanks for your fast replies @davidgyoung, I'm closing this issue.

closed 2018-01-11 by dhabensky

davidgyoung, 2018-01-11:
I think the idea of only persisting the job if  the permission is granted is a good one.  I would welcome a pull request to add this feature to the library.