AltBeacon/android-beacon-library #632: RegionBootstrap enter region triggered for any beacon even when specific regions are set

arickatz, 2017-12-14:
### Expected behavior
Enter event only for specific regions when set
### Actual behavior
RegionBootstrap enter region triggered for any beacon even when specific regions are set
### Steps to reproduce this behavior
  regionBootstrap = new RegionBootstrap(this, new Region("region" + String.valueOf(16), Identifier.parse("2F234454-CF6D-4A0F-ADF2-F4911BA9F161"), null, null));

enters region for a beacon set to "2F234454-CF6D-4A0F-ADF2-F4911BA9F081"

### Mobile device model and OS version
Nexus 6 - Motorola
### Android Beacon Library version

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2017-12-14:
I suspect this problem is caused by persistence of old monitored regions.  If you have previously run the same app on the same device with a different region set up for monitoring, it will still be there unless you have explicitly called stopMonitoringBeaconsInRegion.

You can test to see if that is true in your case by either:

beaconManager.setRegionStatePersistenceEnabled(false); // put in code right after creating beaconManager

Or call:

beaconManager.getMonitoredRegions();

And loop through the results to stop monitoring

Either of the above should fix it.

arickatz, 2017-12-14:
You are correct it was some kind of memory issue after removing the app and reinstalling it
it returned to function as expected :)