AltBeacon/android-beacon-library #633: Android 8.1 and Ranging

breon, 2017-12-15:
Let me start by saying that I'm not a developer. I know enough to be dangerous. 

Since the release of Android 8.1, my phone no longer ranges beacons when the screen is off. Turn the screen back on (not even unlocking the phone) and ranging kicks in again. I've observed this behavior in the logs for our app.

### Expected behavior
* If the phone is locked, the screen is off and there are beacons are nearby, they should continue to be ranged

### Actual behavior
* As soon as the screen turns off, ranging stops

### Steps to reproduce this behavior
* Fire up scanner
  * See beacons in the logs
* Lock phone/turn screen off
  * Beacons no longer in logs

### Mobile device model and OS version
* Pixel 2 XL, Android 8.1
* Nexus 5X, Android 8.1

### Android Beacon Library version
* `android-beacon-library-2.12.3-beta2.aar`

Here is a sample from our logs. You can clearly see where the phone was put to sleep.
```
12-15 14:54:02.398 I/RangingService: HEARTBEAT...Entered Beacons 10, Ranged Beacons 7
12-15 14:54:02.398 I/RangingService: Ranged Beacons IDs found: 0x000025ff623e
12-15 14:54:02.399 I/RangingService: Ranged Beacons IDs found: 0x00a5a42fc203
12-15 14:54:02.399 I/RangingService: Ranged Beacons IDs found: 0x0000da4b7e03
12-15 14:54:02.399 I/RangingService: Ranged Beacons IDs found: 0x000072f3e3c2
12-15 14:54:02.399 I/RangingService: Ranged Beacons IDs found: 0x000012005988
12-15 14:54:02.399 I/RangingService: Ranged Beacons IDs found: 0x11223344aabb
12-15 14:54:02.401 I/RangingService: Ranged Beacons IDs found: 0x00002a67792c
12-15 14:54:02.403 D/RangingService: Found 3 removal candidates
12-15 14:54:03.499 I/RangingService: HEARTBEAT...Entered Beacons 10, Ranged Beacons 7
12-15 14:54:03.500 I/RangingService: Ranged Beacons IDs found: 0x00007f79c4ec
12-15 14:54:03.500 I/RangingService: Ranged Beacons IDs found: 0x000025ff623e
12-15 14:54:03.500 I/RangingService: Ranged Beacons IDs found: 0x00a5a42fc203
12-15 14:54:03.500 I/RangingService: Ranged Beacons IDs found: 0x0000da4b7e03
12-15 14:54:03.500 I/RangingService: Ranged Beacons IDs found: 0x000072f3e3c2
12-15 14:54:03.500 I/RangingService: Ranged Beacons IDs found: 0x000012005988
12-15 14:54:03.500 I/RangingService: Ranged Beacons IDs found: 0x00002a67792c
12-15 14:54:03.501 D/RangingService: Found 3 removal candidates
12-15 14:54:04.602 I/RangingService: HEARTBEAT...Entered Beacons 10, Ranged Beacons 9
12-15 14:54:04.603 I/RangingService: Ranged Beacons IDs found: 0x00007f79c4ec
12-15 14:54:04.603 I/RangingService: Ranged Beacons IDs found: 0x000054ddcbd2
12-15 14:54:04.603 I/RangingService: Ranged Beacons IDs found: 0x000025ff623e
12-15 14:54:04.603 I/RangingService: Ranged Beacons IDs found: 0x00a5a42fc203
12-15 14:54:04.603 I/RangingService: Ranged Beacons IDs found: 0x0000da4b7e03
12-15 14:54:04.604 I/RangingService: Ranged Beacons IDs found: 0x000012005988
12-15 14:54:04.604 I/RangingService: Ranged Beacons IDs found: 0x11223344aabb
12-15 14:54:04.604 I/RangingService: Ranged Beacons IDs found: 0x00002a67792c
12-15 14:54:04.604 I/RangingService: Ranged Beacons IDs found: 0x000054ddcbd1
12-15 14:54:04.606 D/RangingService: Found 1 removal candidates
12-15 14:54:05.706 I/RangingService: HEARTBEAT...Entered Beacons 10, Ranged Beacons 5
12-15 14:54:05.708 I/RangingService: Ranged Beacons IDs found: 0x000054ddcbd2
12-15 14:54:05.709 I/RangingService: Ranged Beacons IDs found: 0x00a5a42fc203
12-15 14:54:05.709 I/RangingService: Ranged Beacons IDs found: 0x0000da4b7e03
12-15 14:54:05.709 I/RangingService: Ranged Beacons IDs found: 0x00002a67792c
12-15 14:54:05.709 I/RangingService: Ranged Beacons IDs found: 0x000054ddcbd1
12-15 14:54:05.710 D/RangingService: Found 5 removal candidates
12-15 14:54:06.808 I/RangingService: HEARTBEAT...Entered Beacons 10, Ranged Beacons 0
12-15 14:54:06.808 D/RangingService: Found 10 removal candidates
12-15 14:54:07.915 I/RangingService: HEARTBEAT...Entered Beacons 10, Ranged Beacons 0
12-15 14:54:07.916 D/RangingService: Found 10 removal candidates
12-15 14:54:09.027 I/RangingService: HEARTBEAT...Entered Beacons 10, Ranged Beacons 0
12-15 14:54:09.029 D/RangingService: Found 10 removal candidates
12-15 14:54:10.134 I/RangingService: HEARTBEAT...Entered Beacons 10, Ranged Beacons 0
12-15 14:54:10.135 D/RangingService: Found 10 removal candidates
```

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2017-12-16:
Thanks for this report.    Two requests if you would be so kind:

1.  Can you please capture a system level LogCat capture (one without a filter for your application) for about 5 seconds before and after the time the screen turns off?

2. Can you try the reference app to verify you see the same behavior there:  https://github.com/altbeacon/android-beacon-library-reference

breon, 2017-12-18:
Thanks for the reply. Attached are the full device logs from app launch, to phone being locked, plus some time after lock.

[org.awareservices.vitals.debug.FULL.txt](https://github.com/AltBeacon/android-beacon-library/files/1568556/org.awareservices.vitals.debug.FULL.txt)

I also tried the reference app, after updating it to find Eddystone and writing a line to the log. Attached are app only logs, as well as full device logs.

[appOnlyLogs.txt](https://github.com/AltBeacon/android-beacon-library/files/1568561/appOnlyLogs.txt)
[fullLogs.txt](https://github.com/AltBeacon/android-beacon-library/files/1568562/fullLogs.txt)

Our Android developer (@hashbrown) is also taking a look at the logs.

Let me know if you need anything else.

hashbrown, 2017-12-19:
This looks to be a telling line:
```E/BtGatt.ScanManager: Cannot start unfiltered scan in screen-off. This scan will be resumed later: 9```

hashbrown, 2017-12-19:
https://android.googlesource.com/platform/packages/apps/Bluetooth/+/319aeae6f4ebd13678b4f77375d1804978c4a1e1

davidgyoung, 2017-12-19:
@hashbrown, thanks for digging n to find this.  What an unwelcome surprise change in a minor release!

Looking at the code, there may be some wiggle room with filtered scans.  I will run some tests.

davidgyoung, 2018-01-04:
@breon and @hashbrown, I have created a test release to see if this resolves the issue.  I don't have an Android 8.1 device available right now, so if one of you could try it and confirm if it allows scans with the screen off, I would appreciate it.

Here's a link to the test release with instructions on configuring it with your app and a link to download the binary:

https://github.com/AltBeacon/android-beacon-library/releases/tag/2.12.4-android8_1

breon, 2018-01-04:
@davidgyoung I updated the library in our project and it appears to be working fine. I let it run for over 30 minutes and ranging never stopped.

Is there anything else specific you'd like me to test?

davidgyoung, 2018-01-04:
Good news, @breon.  Thanks for the test!  I can't think of any other tests we need on 8.1 right now, so this should cover it.

davidgyoung, 2018-01-04:
Fixed in #637

closed 2018-01-04 by davidgyoung

breon, 2018-01-19:
This has been working great! When do you expect it to be officially released?
https://github.com/AltBeacon/android-beacon-library/releases/tag/2.12.4-android8_1

davidgyoung, 2018-01-20:
Hi, @breon.  I had hoping to get a second confirmation that this technique works after I received this report that the solution didn't work:  https://stackoverflow.com/questions/48114602/ble-scan-stop-when-mobile-screen-off-in-android-8-1-0.  I tend to treat that report with some skepticism, but I'd still love to get a second confirmation.  I've been using a Nexus Player for my Android 8 testing, but Google has not released 8.1.0 for that platform.

breon, 2018-01-22:
Totally understand! We'll continue to build with the pre-release for now.  Thanks for the reply.

ufukoe, 2018-02-19:
I tried to use this library to start a filtered scan on my pixel 2 with Android 8.1. But still I don't receive an scan results whenever I turn off my display. Following output still exists: 
E/BtGatt.ScanManager: Cannot start unfiltered scan in screen-off. This scan will be resumed later
I thought the wildcard Scan Filter allows to detect any beacon advertisement even if the screen is off.

davidgyoung, 2018-02-19:
Thanks for the test, @ufukoe.  Can you please add the following lines to your app:

```
beaconManager.setDebug(true);
Log.d(TAG, "I am using android beacon library version "+org.altbeacon.beacon.BuildConfig.VERSION_NAME);
```

Then run it and attach a log that shows two things:  

(1) The version of the library actually being used:

```
02-19 08:57:35.146 29096-29096/org.altbeacon.beaconreference D/BeaconReferenceApp: I am using android beacon library version 2.12.4-android8_1-1-g6cf77fe-SNAPSHOT
```

(2) What the system says when the library tries to start a scan in SCAN_MODE_LOW_LATENCY.  Please show the system messages you see including the one that you mention `Cannot start unfiltered scan in screen-off.`

```
02-19 09:01:18.511 29096-29096/org.altbeacon.beaconreference D/CycledLeScanner: starting a new scan cycle
02-19 09:01:18.512 29096-29096/org.altbeacon.beaconreference D/CycledLeScanner: starting a new bluetooth le scan
02-19 09:01:18.513 29096-29096/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: starting non-filtered scan in SCAN_MODE_LOW_LATENCY
02-19 09:01:18.514 29096-29096/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Making new Android L scanner
02-19 09:01:18.514 29096-29096/org.altbeacon.beaconreference D/CycledLeScanner: Waiting to stop scan cycle for another 10000 milliseconds
02-19 09:01:18.515 29096-29814/org.altbeacon.beaconreference D/BluetoothAdapter: STATE_ON
```

Please ignore the 'non-filtered scan' verbiage in the sample debug output I show above.  The library is starting a scan with an empty filter, despite what the debug message says.

davidgyoung, 2018-03-05:
@ufukoe  and @breon,   I have obtained an 8.1 device for testing and verified that the change I put in place to fix this issue does cause scans to work with the screen off.  You can see my test here that shows the failure with library version 2.12.4 and working with the pre-release version  2.12.4-2-gca0c23c.aar that has the proposed fix:

https://github.com/AltBeacon/android-beacon-library/pull/637#issuecomment-370308798

wodnd23, 2018-11-08:
Hello. @davidgyoung :)

I try to develop an service using altBeacon library version 2.15.2 at android 8.1.
Looking at the comments above, I expected that ranging would work well even with the screen off.
However, size of the beacon list is always 0 at 'didRangeBeaconsInRegion', even though there are active beacons around with the screen off.

Could you please help me to fix this issue? Thank you.