AltBeacon/android-beacon-library #636: NullPointerException - ScanHelper class

himanshu-android, 2018-01-02:
### Expected behavior
Start background scanning in Android Oreo

### Actual behavior
Application crashed due to NullPointerException

### Steps to reproduce this behavior
Unavailability of Android O device but crash report should help.

### Mobile device model and OS version
OS Version: 8.0.0
Device: ONEPLUS A3003

### Android Beacon Library version
Version - 2.12.4

[le_scan_exception.txt](https://github.com/AltBeacon/android-beacon-library/files/1597417/le_scan_exception.txt)

ncapdevi, 2018-01-08:
I am getting a similar crash

```
Fatal Exception: java.lang.RuntimeException: java.lang.NullPointerException: Attempt to invoke virtual method 'java.util.Set com.android.bluetooth.gatt.ScanManager.getBatchScanQueue()' on a null object reference
       at android.app.job.JobServiceEngine$JobHandler.handleMessage(JobServiceEngine.java:112)
       at android.os.Handler.dispatchMessage(Handler.java:105)
       at android.os.Looper.loop(Looper.java:164)
       at android.app.ActivityThread.main(ActivityThread.java:6809)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)
Caused by java.lang.NullPointerException: Attempt to invoke virtual method 'java.util.Set com.android.bluetooth.gatt.ScanManager.getBatchScanQueue()' on a null object reference
       at android.os.Parcel.readException(Parcel.java:1948)
       at android.os.Parcel.readException(Parcel.java:1888)
       at android.bluetooth.IBluetoothGatt$Stub$Proxy.stopScanForIntent(IBluetoothGatt.java:970)
       at android.bluetooth.le.BluetoothLeScanner.stopScan(BluetoothLeScanner.java:290)
       at org.altbeacon.beacon.service.ScanHelper.stopAndroidOBackgroundScan(ScanHelper.java:194)
       at org.altbeacon.beacon.service.ScanJob.restartScanning(ScanJob.java:183)
       at org.altbeacon.beacon.service.ScanJob.startScanning(ScanJob.java:221)
       at org.altbeacon.beacon.service.ScanJob.onStartJob(ScanJob.java:77)
       at android.app.job.JobService$1.onStartJob(JobService.java:71)
       at android.app.job.JobServiceEngine$JobHandler.handleMessage(JobServiceEngine.java:108)
       at android.os.Handler.dispatchMessage(Handler.java:105)
       at android.os.Looper.loop(Looper.java:164)
       at android.app.ActivityThread.main(ActivityThread.java:6809)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)
```

mikhailmelnik, 2018-04-17:
Getting similar crash from production but in different place:

```
java.lang.NullPointerException: Attempt to invoke interface method 'void android.bluetooth.IBluetoothGatt.stopScanForIntent(android.app.PendingIntent, java.lang.String)' on a null object reference
	at android.bluetooth.le.BluetoothLeScanner.stopScan()(BluetoothLeScanner.java:278)
	at org.altbeacon.beacon.service.ScanHelper.stopAndroidOBackgroundScan()(ScanHelper.java:194)
	at org.altbeacon.beacon.service.ScanJob.restartScanning()(ScanJob.java:187)
	at org.altbeacon.beacon.service.ScanJob.startScanning()(ScanJob.java:225)
	at org.altbeacon.beacon.service.ScanJob.onStartJob()(ScanJob.java:81)
	at android.app.job.JobService$1.onStartJob()(JobService.java:71)
	at android.app.job.JobServiceEngine$JobHandler.handleMessage()(JobServiceEngine.java:108)
	at android.os.Handler.dispatchMessage()(Handler.java:105)
	at android.os.Looper.loop()(Looper.java:164)
	at android.app.ActivityThread.main()(ActivityThread.java:6541)
	at java.lang.reflect.Method.invoke()(Method.java:-2)
	at com.android.internal.os.Zygote$MethodAndArgsCaller.run()(Zygote.java:240)
	at com.android.internal.os.ZygoteInit.main()(ZygoteInit.java:767)
```
Version: 2.13.1
Devices: Pixel, ONEPLUS A3003, Nexus 6P

mannodermaus, 2018-05-15:
Same here, and very similar trace to above:

```
java.lang.NullPointerException: Attempt to invoke virtual method 'java.util.Set com.android.bluetooth.gatt.ScanManager.getBatchScanQueue()' on a null object reference
	at android.os.Parcel.readException (Parcel.java:1968)
	at android.os.Parcel.readException (Parcel.java:1898)
	at android.bluetooth.IBluetoothGatt$Stub$Proxy.stopScanForIntent (IBluetoothGatt.java:970)
	at android.bluetooth.le.BluetoothLeScanner.stopScan (BluetoothLeScanner.java:278)
	at org.altbeacon.beacon.service.ScanHelper.stopAndroidOBackgroundScan (ScanHelper.java:194)
	at org.altbeacon.beacon.service.ScanJob.restartScanning (ScanJob.java:187)
	at org.altbeacon.beacon.service.ScanJob.startScanning (ScanJob.java:225)
	at org.altbeacon.beacon.service.ScanJob.onStartJob (ScanJob.java:81)
	at android.app.job.JobService$1.onStartJob (JobService.java:71)
	at android.app.job.JobServiceEngine$JobHandler.handleMessage (JobServiceEngine.java:108)
	at android.os.Handler.dispatchMessage (Handler.java:105)
	at android.os.Looper.loop (Looper.java:251)
	at android.app.ActivityThread.main (ActivityThread.java:6572)
	at java.lang.reflect.Method.invoke (Method.java:-2)
	at com.android.internal.os.Zygote$MethodAndArgsCaller.run (Zygote.java:240)
	at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:767)
```

Version: 2.13.1
Devices: Xperia XZ Premium, Xperia XZ
Android: 8.0.0

EmmettWilson, 2018-06-08:
We are also seeing the same 

```android.os.Parcel.readException (Parcel.java:2010)
android.os.Parcel.readException (Parcel.java:1950)
android.bluetooth.IBluetoothGatt$Stub$Proxy.stopScanForIntent (IBluetoothGatt.java:972)
android.bluetooth.le.BluetoothLeScanner.stopScan (BluetoothLeScanner.java:283)
arrow_right
org.altbeacon.beacon.service.ScanHelper.stopAndroidOBackgroundScan (ScanHelper.java:194)
org.altbeacon.beacon.service.ScanJob.restartScanning (ScanJob.java:187)
org.altbeacon.beacon.service.ScanJob.startScanning (ScanJob.java:225)
org.altbeacon.beacon.service.ScanJob.onStartJob (ScanJob.java:81)
android.app.job.JobService$1.onStartJob (JobService.java:71)
android.app.job.JobServiceEngine$JobHandler.handleMessage (JobServiceEngine.java:108)
android.os.Handler.dispatchMessage (Handler.java:106)
android.os.Looper.loop (Looper.java:164)
android.app.ActivityThread.main (ActivityThread.java:6494)
java.lang.reflect.Method.invoke (Method.java)
com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:438)
com.android.internal.os.ZygoteInit.main (ZygoteInit.java:807)```

Version 2.14
Devices: Pixel 2XL
Android: 8.1.0

davidgyoung, 2018-06-08:
Anybody know how to reproduce this?

All the reports I have heard of come from remote  crash log reports which give no indication of what conditions cause this, how frequently it happens etc.

Without more info to reproduce it will be very hard to solve.

siralam, 2018-08-07:
Just to provide more information, don't know if it helps or not though:

In an old app that I have published **which does not use this library**, I also get similar NullPointerException:

```
Caused by java.lang.NullPointerException: Attempt to invoke virtual method 'void com.android.bluetooth.gatt.ScanManager.registerScanner(java.util.UUID)' on a null object reference
       at android.os.Parcel.readException(Parcel.java:1969)
       at android.os.Parcel.readException(Parcel.java:1899)
       at android.bluetooth.IBluetoothGatt$Stub$Proxy.registerScanner(IBluetoothGatt.java:878)
       at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.startRegistration(BluetoothLeScanner.java:369)
       at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:235)
       at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:132)
       at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:113)
```

Also happens only in Android 8, and occurs when I call
`BluetoothLeScanner.startScan(scanCallback);`

Which is different from the above crash logs which indicates crash happens in `stopScan`; but the similarity is both are `NullPointerException`s due to a null `ScanManager` which is a class of android framework.

davidgyoung, 2018-08-07:
Since this NPE is thrown during unknown conditions internally by Android, on this call:

`scanner.stopScan(getScanCallbackIntent());`

There is unfortunately no choice but to explicitly catch that exception to prevent a crash.

kababu007, 2019-02-07:
Where do we catch this exception?

davidgyoung, 2019-02-07:
@kababu007, it looks like a fix for this was never implemented.  We need to catch a NullPointerException internally to the library and roll a new library release to fix this.

I believe a change would simply require modifying ScanHelper.java to catch a NullPointerException on each of these two calls: `scanner.stopScan(getScanCallbackIntent());` and   `int result = scanner.startScan(filters, settings, getScanCallbackIntent());`

I'll put this on my list, but for a faster fix, I'd welcome a pull request and any testing you could do if you'd like to volunteer.  That would certainly speed things along.

kababu007, 2019-02-07:
Any tentative timeline to release the new library with this fix? I see these 2 functions stopScan and startScan throwing NPE very frequently.

kababu007, 2019-02-07:
Pull Request submitted for merge -->  "kababu007-patch-nullpointexception-in-start-and-stop-scan-in-background" The fix is straightforward and we tested it for 45 mins idle time leaving the app in background.

suzukieng, 2019-03-06:
Just chiming in on this issue, I'm seeing this crash as  well, with a slightly different stack trace. It's occurring very infrequently.  This particular one is from an LG phone (LG-H930) running Android 8.0.0.

```
java.lang.NullPointerException: Attempt to invoke virtual method 'void com.android.bluetooth.gatt.ScanManager.registerScanner(java.util.UUID)' on a null object reference
	at android.os.Parcel.readException(Parcel.java:1949)
	at android.os.Parcel.readException(Parcel.java:1889)
	at android.bluetooth.IBluetoothGatt$Stub$Proxy.registerScanner(IBluetoothGatt.java:936)
	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.startRegistration(BluetoothLeScanner.java:389)
	at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:249)
	at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:133)
```

Does anyone know if there is a ticket in the Android bugtracker for this?

davidgyoung, 2019-03-06:
This issue has been fixes in this library by catching the exception per the above PR.  The change is in this release: https://github.com/AltBeacon/android-beacon-library/releases/tag/2.16.1

Closing this issue per that fix.  If anybody sees this again in 2.16.1+ please comment here.

VanHoangTran, 2019-06-05:
Got that crash 
```
Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'void com.android.bluetooth.gatt.ScanManager.registerScanner(java.util.UUID)' on a null object reference
       at android.os.Parcel.readException(Parcel.java:1957)
       at android.os.Parcel.readException(Parcel.java:1897)
       at android.bluetooth.IBluetoothGatt$Stub$Proxy.registerScanner(IBluetoothGatt.java:936)
       at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.startRegistration(BluetoothLeScanner.java:389)
       at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:249)
       at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:133)
       at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:114)
       at org.altbeacon.bluetooth.BluetoothMedic.runScanTest(BluetoothMedic.java:263)
       at org.altbeacon.bluetooth.BluetoothTestJob$1.run(BluetoothTestJob.java:104)
       at android.os.Handler.handleCallback(Handler.java:789)
       at android.os.Handler.dispatchMessage(Handler.java:98)
       at android.os.Looper.loop(Looper.java:164)
       at android.os.HandlerThread.run(HandlerThread.java:65)
```
Version: 2.16.1
Devices: LG G6
Android: 8.0.0

davidgyoung, 2019-06-05:
@VanHoangTran, thanks for this report.  This crash comes from the BluetoothMedic instead of the primary ScanHelper, do the same gux above needs to be applied there, too.