AltBeacon/android-beacon-library #638: Attempt to invoke virtual method 'boolean org.altbeacon.beacon.service.RangedBeacon.isTracked()' on a null object reference

Eider3a, 2018-01-12:
Hi, I have the next issue using the library AltBeacon, can you help me to fix it, sometimes I have this trouble (java.lang.NullPointerException: Attempt to invoke virtual method 'boolean org.altbeacon.beacon.service.RangedBeacon.isTracked()' on a null object reference), appers to be in the file RangeState.class in the line 59, I tried to modify the library but it is lock.

Thanks for all.
This is my code in MainActivity:

`
@Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);

beaconManager=BeaconManager.getInstanceForApplication(context);
        beaconManager.getBeaconParsers().add(new BeaconParser().
                setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25"));
        beaconManager.bind(this);

    }

    @Override
    public void onBeaconServiceConnect() {
        final Region region=new Region("MyBeacon",null,null,null);

        beaconManager.addRangeNotifier(new RangeNotifier() {
            //Se ejecuta cada segundo para verificar si existen beacons alrededor.
            @Override
            public void didRangeBeaconsInRegion(Collection<Beacon> beacons, Region region) {
                beaconList.clear();//Limpio la lista.

                for(Beacon oneBeacon:beacons){

                    final String dirMAC=oneBeacon.getBluetoothAddress();
                    final String uuid=oneBeacon.getId1().toString();
                    final String minor=oneBeacon.getId2().toString();
                    final String major=oneBeacon.getId3().toString();
                    final String distancia=String.valueOf(oneBeacon.getDistance());

                    //Crear la lista.
                    Beacons beaconElement=new Beacons(uuid,minor,major,dirMAC,distancia);
                    beaconList.add(beaconElement);

                    oneBeacon.setHardwareEqualityEnforced(true);

                    Log.i("MainActivity", "dist:"+oneBeacon.getDistance()+" adr:"+oneBeacon.getBluetoothAddress()+" txP"+oneBeacon.getRssi());

                }
                //shows the list in a recyclerview.
                runOnUiThread(new Runnable() {
                    public void run() {
                        beaconAdapter.notifyDataSetChanged();
                    }
                });

            }
        });

        try {
            beaconManager.startRangingBeaconsInRegion(region);
        } catch (RemoteException e) {
            e.printStackTrace();
        }
    }

    @Override
    protected void onDestroy() {
        super.onDestroy();
        beaconManager.unbind(this);
    }`

This is the whole error tha appers in androidStudio:
`java.lang.NullPointerException: Attempt to invoke virtual method 'boolean org.altbeacon.beacon.service.RangedBeacon.isTracked()' on a null object reference
                      at org.altbeacon.beacon.service.RangeState.finalizeBeacons(RangeState.java:74)
                      at org.altbeacon.beacon.service.ScanHelper.processRangeData(ScanHelper.java:272)
                      at org.altbeacon.beacon.service.ScanHelper.access$200(ScanHelper.java:60)
                      at org.altbeacon.beacon.service.ScanHelper$1.onCycleEnd(ScanHelper.java:223)
                      at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:388)
                      at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:378)
                      at org.altbeacon.beacon.service.scanner.CycledLeScanner$2.run(CycledLeScanner.java:374)
                      at android.os.Handler.handleCallback(Handler.java:739)
                      at android.os.Handler.dispatchMessage(Handler.java:95)
                      at android.os.Looper.loop(Looper.java:135)
                      at android.app.ActivityThread.main(ActivityThread.java:5343)
                      at java.lang.reflect.Method.invoke(Native Method)
                      at java.lang.reflect.Method.invoke(Method.java:372)
                      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:905)
                      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:700)`

MGasztold, 2018-03-20:
Hello,

I experience the same issue. It happens only on my Android 8 devices. Thank you.

kanikarakheja, 2018-07-18:
Hi,

I am having the same issue and it is on Android 7. I am using latest Alt Beacon lib.  I would appreciate the help.

Thanks 
Kanika

davidgyoung, 2018-07-18:
I am not sure how it is possible for this to be null, but the crash can be prevented by adding a block after this line:

                RangedBeacon rangedBeacon = mRangedBeacons.get(beacon);
                if (rangedBeacon != null) {
                    ...
                }

thebertozz, 2018-08-02:
Hi @davidgyoung , were can I put the code above? Is it something in the library? I already have the null checks for the Collection<Beacon> collection and on the single beacon item when in the didRangeBeaconsInRegion, but the issue is still there on Android > 8.

Thanks for your time,

Federico

davidgyoung, 2018-08-02:
Hi, @thebertozz, I believe I have a fix for this in #716.  Please take a look and let me know if you agree this will solve it.

closed 2018-08-02 by davidgyoung

thebertozz, 2018-08-02:
Thanks! I've built the library from the fix-ranged-beacon-null-pointer branch and i will keep it in our demo app to check what happens. I'll keep you posted.