AltBeacon/android-beacon-library #641: Android O Ranging problem

FongMi, 2018-01-26:
### Expected behavior
Android O (8.0) can search beacon in my regions

### Actual behavior
The log says
> BeaconManager started up on pid 7266 named 'com.direk.android.newest' for application package 'com.direk.android.newest'.  isMainProcess=true
> This consumer is not bound.  Binding now: com.direk.android.clock.ClockBeaconActivity@fbefc
> Not starting beacon scanning service. Using scheduled jobs
> consumer count is now: 1

### Steps to reproduce this behavior
Same as Ranging Example Code
http://altbeacon.github.io/android-beacon-library/samples.html

### Mobile device model and OS version
HTC_M10
Android 8.0

### Android Beacon Library version
android-beacon-library-2.12.4

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-02-28:
@FongMi, the log lines shown look normal.   Can you please describe the problem in greater detail?

Do you ever get a ranging callback on Android 8?
Does your code work on Android 4.3-7.x devices?
How long to do you wait for a ranging callback on Android 8?
Does it make a difference in the foreground vs. the background?
Can you detect your beacon using an off the shelf beacon detector?
Have you added the proper beacon layout for your beacon (if it is iBeacon or Eddystone)?

FongMi, 2018-03-01:
Do you ever get a ranging callback on Android 8?
->no

Does your code work on Android 4.3-7.x devices?
->yes

How long to do you wait for a ranging callback on Android 8?
->I have not get the callback

Does it make a difference in the foreground vs. the background?
->I only use foreground scan

Can you detect your beacon using an off the shelf beacon detector?
->Sure

Have you added the proper beacon layout for your beacon (if it is iBeacon or Eddystone)?
->Yep, I add a iBeacon layout by myself.
->m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24

davidgyoung, 2018-03-01:
Thanks for the answers, @FongMi .  This certainly is not expected behavior and is not consistent with tests I have done with the reference app on Android 8.  Would you mind running the reference app on your device and see if you see the same?

https://github.com/AltBeacon/android-beacon-library-reference

I wonder if there is some difference with the way your app is using the library that makes it incompatible with Android 8.

FongMi, 2018-03-01:
ok, I will try again, thanks.

MahmoudElFerekh, 2018-03-02:
Any update on this issue? Since I'm facing the same scenario with the same device and same Android version. I can't reach other device with Android 8.0 installed on to test, but on any other devices with versions 4.3-7.x it's working fine. Thanks in advance.

FongMi, 2018-03-02:
@MahmoudElFerekh I don't have phone to test now, have you try the sample?
https://github.com/AltBeacon/android-beacon-library-reference

MahmoudElFerekh, 2018-03-02:
Now I tried my app on android 8.0 Huawei Mate 10 pro, it didn't work, but the sample worked fine and was able to locate beacons. I will try to figure out what's the difference between my code and that in the sample and will give update if I come up with a solution.

MahmoudElFerekh, 2018-03-06:
Hello, the issue that I was facing is related to issue #613, where I was depending on isBound() method in order to start scanning, that's why it wasn't detecting any beacon. I hope this can help you. Thanks

closed 2018-03-09 by FongMi