AltBeacon/android-beacon-library #646: Virtual beacon transmitter

Rkr999, 2018-02-23:
Hi Team,

How to transmit  and detect eddy stone url, major id and minor id using virtual beacon.

davidgyoung, 2018-02-23:
@Rkr999, if you need help in using this library please open a question on StackOverflow.com as this forum is reserved for feature requests and bug reports.  

You may wish to review the documentation on the [Samples page](http://altbeacon.github.io/android-beacon-library/samples.html) which shows you how to set up a beacon transmitter.  The ranging sample on the same page will work for detection.

closed 2018-02-23 by davidgyoung

Rkr999, 2018-02-27:
Hi Team,

I am able to transmit virtual beacon in android device but it is not detecting in IOS and Android device. Please reply me ASAP.

Beacon beacon = new Beacon.Builder()
                .setId1(getResources().getString(R.string.BeaconUUID))
                .setId2("133")
                .setId3("144")
                .setManufacturer(0x0118)
                .setTxPower(-59)
                .setDataFields(Arrays.asList(new Long[]{0l}))
                .build();
        beaconTransmitter.startAdvertising(beacon, new AdvertiseCallback() {
            @Override
            public void onStartSuccess(AdvertiseSettings settingsInEffect) {
                super.onStartSuccess(settingsInEffect);
                Log.d(TAG, "Transmission started");
                
                    }
                });
            }

            @Override
            public void onStartFailure(int errorCode) {
                super.onStartFailure(errorCode);
                Log.e(TAG, "Transmission failed with error code : " + errorCode);
            }
        });

davidgyoung, 2018-02-27:
Please use the Apple manufacturer code 0x004c for detection on iOS.  If you need more help, please open a new question on StackOverflow.com.

Rkr999, 2018-02-27:
Thank you david

Rkr999, 2018-02-28:
in IOS by using CoreLocation library we are unable to detect android virtual iBeacon.

davidgyoung, 2018-02-28:
@Rkr999, please see the note above about the manufacturer code.  If you need more help, please open a question on Stackoverflow.com.   I am happy to answer there, but this forum is reserved for feature requests and bug reports with the library itself.

Rkr999, 2018-03-01:
Still we are facing the issue please find the question in stack overflow using below link.
https://stackoverflow.com/questions/49024321/android-device-as-virtual-ibeacon