AltBeacon/android-beacon-library #648: Huawei P8 Lite ble scan not stopped

ohdonpiano, 2018-02-28:
### Expected behavior
stopMonitoringBeaconsInRegion(region) causes ble scan to stop after some time (immediately, or after a specific amount of time depending on the configuration) 

### Actual behavior
scan results continue to be present in LogCat

### Steps to reproduce this behavior
first start the monitoring or a region, the stop it

### Mobile device model and OS version
Huawei P8 Lite (ALE-L21), EMUI 4.0.3, Android 6.0

### Android Beacon Library version
implementation 'org.altbeacon:android-beacon-library:2.12.3'

I unterstand that the problem is specific to this mobile device (maybe also other Huawei devices like P9 Lite, or other devices that use EMUI)
My issue is that BluetoothDevice connectGatt() is not working (gatt error 0x85 (133)) on this mobile device while ble scan is active, so I wanted to report that this is also caused by starting the beacon region monitoring with AltBeacon

Hope that someone has found a solution or managed to talk with Huawei about this

davidgyoung, 2018-02-28:
I'm not sure I understand this issue, as it sounds like two different problems are reported:

1. Stopping scanning fails using this library
2. BluetoothDevice connectGatt() returns a 133 error

Is this correct?

This kind of problem may indicate an issue bluetooth stack on the Android device getting into a bad state.  Sometimes this can be recovered by cycling power to bluetooth. There is a new library feature that can automatically do this described here: AltBeacon/android-beacon-library#644, but it is not in a formal release yet. If you have comments on this, please discuss in that thread.

davidgyoung, 2018-04-22:
closing due to inactivity

closed 2018-04-22 by davidgyoung