AltBeacon/android-beacon-library #650: BootstrapRegion takes a long time for ENTERED evend, and EXIT events are not called.

alin-rosu, 2018-03-05:
I had a code where I would create a region, and start monitoring it.

  Region region = new Region("all-beacons-region", Identifier.parse(beacon.getUuid()), null, null);
    beaconManager.startMonitoringBeaconsInRegion(region);

This works PERFECT. I get the ENTERED event at the moment I plug the beacon USB in my PC, and the exited not long after I unplug it.
But I need to monitor 2 regions at the same time. I tried making a for Loop for my regions and call: 
 beaconManager.startMonitoringBeaconsInRegion(region);
But this would only monitor the last region.

So I looked over the code, and I saw I can send a Collection of Regions to the RegionBoostrap.
This is what I do now: 

       regionBootstrap = new RegionBootstrap(bootstrapNotifier, regions);

And my ApplicationClass implements BootstrapNotifier. so I have now this on my Application class:

      @Override
    public void didDetermineStateForRegion(int arg0, Region arg1) {
        // Don't care
    }

    @Override
    public void didEnterRegion(final Region region) {
        createNotification("ENTERED");
       PSBluetoothService.getInstance(PSApplicationClass.this).didEnterCalled(region);
    }

    @Override
    public void didExitRegion(Region arg0) {
        createNotification("EXITED");
        PSBluetoothService.getInstance(PSApplicationClass.this).beaconToUseForCheckin = null;
        Utils.appendLog("addRegion app EXITED a beacon region", "I", Constants.OTHER);
    } 

This works horrible. It takes more than 4-5 minutes to get the "didEnterRegion" event, after I plug in the Beacon. And I don't even get the "didExitRegion" called. Why is this happening? Why is it so sluggish compared to when I set on the beaconManager with "startMonitoringBeaconsInRegion"

davidgyoung, 2018-03-05:
A few points:

* Make sure when constructing regions that you have a  *unique* string for the first parameter. `Region region = new Region("backgroundRegion", null, null, null);`  If the same string is reused across multiple regions, they will get overwritten and only the last region will be monitored.

* Understand that if you are using BackgroundPowerSaver and the app is in the background, beacon scans will only happen every 5 minutes if any beacons are visible.  This is necessary to save battery.  You will only get fast detections the first time any beacon is detected in the vicinity after previously no beacons had been detected.

If you need more help on these issues, please open a question on StackOverflow.com and I can assist there.  This forum is reserved for feature request and reproducible bug reports.  If we find the issues you are seeing are a bug in the library.  I am happy to reopen.

closed 2018-03-05 by davidgyoung

alin-rosu, 2018-03-05:
What I do not get is why if I set only 1, I get the notification instant. but if I set more, it takes about 5 minutes? I have no power saver on. So if it works for 1, cause I have no power saver, shouldn't it be the same for the second one?