AltBeacon/android-beacon-library #652: Android 8.0 crash

quzhipeng, 2018-03-06:
Caused by: java.lang.RuntimeException: Cannot get job id from manifest.  Make sure that the immediateScanJobId is configured in the manifest for the ScanJob.
at org.altbeacon.beacon.service.ScanJob.getJobIdFromManifest(ScanJob.java:312)
at org.altbeacon.beacon.service.ScanJob.getImmediateScanJobId(ScanJob.java:282)

modify:51c3fff

immediateScanJobId=>immmediateScanJobId

davidgyoung, 2018-03-06:
Thanks for this report.  I'm working on a quick fix in #653.  Will push an update as soon as I fully test it.

closed 2018-03-06 by davidgyoung

davidgyoung, 2018-03-06:
This is in a 2.13.1 release on JCenter and MavenCentral.  If you rebuild with that release you should be good.

quzhipeng, 2018-03-07:
Thank you for your modification.

PR #653 (2018-03-06): Fix crash from incorrect metadata name 
This fixes #652